SatDump/SatDump #266: Add support for BladeRF?

jamesmac10, 2023-03-25:
Hi there,

There seems to be quite a few people out there with blades that would fit in your demographic. Any plans to release support? I personally have the bladerf 2.0 micro xa4 and its a great tool.

Thanks,

Jamesmac10

AlexandreRouma, 2023-03-25:
It's already supported

pluggerlockett, 2023-03-30:
Can confirm, I've got that exact bladeRF and it works in SatDump. You have to install libbladerf-dev (I think that's the package name) to get it to work though. For some reason that's not included in the Install instructions.

jamesmac10, 2023-03-30:
Ok, I will install the package and give it a shot. Thank you!

Aang23, 2023-04-15:
Yes, indeed BladeRF is already supported, You will need to do what @pluggerlockett said to get it working.

closed 2023-04-15 by Aang23