SatDump/SatDump #270: Unable to spawn SatDump in another program.

bclswl0827, 2023-04-08:
**Description of the issue**

I was spawning a SatDump as a subprocess in a Go program, but it always throws error.

**Hardware (SDR/PC/OS)**

SDR: SDRplay 1A
PC: x86_64 architecture
OS: Bullseye Debian GNU/Linux

**Version (Eg, 1.0.0, CI Build #171)**

SatDump v1.1.0-a35e30c

**Logs after the cras (satdump.logs)**

I was using this command
```
/usr/local/bin/satdump live gk2a_lrit /files --source sdrplay --samplerate 3000000 --frequency 1.69214e9 --gain 40 --http_server 127.0.110.22:1234  --agc  --bias --dc_block true
```

satdump logs
```
[11:30:28 - 08/03/2023] (I)    _____       __  ____                      
[11:30:28 - 08/03/2023] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ 
[11:30:28 - 08/03/2023] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[11:30:28 - 08/03/2023] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[11:30:28 - 08/03/2023] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ 
[11:30:28 - 08/03/2023] (I)                                     /_/      
[11:30:28 - 08/03/2023] (I) Starting SatDump v1.1.0-a35e30c
[11:30:28 - 08/03/2023] (I) 
[11:30:28 - 08/03/2023] (I) Loading config /usr/local/share/satdump//satdump_cfg.json
[11:30:28 - 08/03/2023] (W) No user configuration found! Keeping defaults.
[11:30:28 - 08/03/2023] (I) Loading plugins from /usr/local/share/satdump//plugins
[11:30:28 - 08/03/2023] (T) Loading plugin /usr/local/share/satdump//plugins/libsdrpp_server_support.so...
[11:30:28 - 08/03/2023] (T) Plugin spyserver_support loaded!
[11:30:28 - 08/03/2023] (T) Loading plugin /usr/local/share/satdump//plugins/libjpss_support.so...
[11:30:28 - 08/03/2023] (T) Plugin jpss_support loaded!
[11:30:28 - 08/03/2023] (T) Loading plugin /usr/local/share/satdump//plugins/libanalog_support.so...
[11:30:28 - 08/03/2023] (T) Plugin analog_support loaded!
[11:30:28 - 08/03/2023] (T) Loading plugin /usr/local/share/satdump//plugins/libcubesat_support.so...
[11:30:28 - 08/03/2023] (T) Plugin cubesat_support loaded!
[11:30:28 - 08/03/2023] (T) Loading plugin /usr/local/share/satdump//plugins/libjason3_support.so...
[11:30:28 - 08/03/2023] (T) Plugin jason3_support loaded!
[11:30:28 - 08/03/2023] (T) Loading plugin /usr/local/share/satdump//plugins/librtltcp_support.so...
[11:30:28 - 08/03/2023] (T) Plugin rtltcp_support loaded!
[11:30:28 - 08/03/2023] (T) Loading plugin /usr/local/share/satdump//plugins/libmeteor_support.so...
[11:30:28 - 08/03/2023] (T) Plugin meteor_support loaded!
[11:30:28 - 08/03/2023] (T) Loading plugin /usr/local/share/satdump//plugins/libfengyun4_support.so...
[11:30:28 - 08/03/2023] (T) Plugin fenyun4_support loaded!
[11:30:28 - 08/03/2023] (T) Loading plugin /usr/local/share/satdump//plugins/libfengyun3_support.so...
[11:30:28 - 08/03/2023] (T) Plugin fengyun3_support loaded!
[11:30:28 - 08/03/2023] (T) Loading plugin /usr/local/share/satdump//plugins/libeos_support.so...
[11:30:28 - 08/03/2023] (T) Plugin eos_support loaded!
[11:30:28 - 08/03/2023] (T) Loading plugin /usr/local/share/satdump//plugins/libusrp_sdr_support.so...
[11:30:28 - 08/03/2023] (T) Plugin usrp_sdr_support loaded!
[11:30:28 - 08/03/2023] (T) Loading plugin /usr/local/share/satdump//plugins/liboceansat_support.so...
[11:30:28 - 08/03/2023] (T) Plugin oceansat_support loaded!
[11:30:28 - 08/03/2023] (T) Loading plugin /usr/local/share/satdump//plugins/libfengyun2_support.so...
[11:30:28 - 08/03/2023] (T) Plugin fengyun2_support loaded!
[11:30:28 - 08/03/2023] (T) Loading plugin /usr/local/share/satdump//plugins/libhimawari_support.so...
[11:30:28 - 08/03/2023] (T) Plugin himawari_support loaded!
[11:30:28 - 08/03/2023] (T) Loading plugin /usr/local/share/satdump//plugins/libaim_support.so...
[11:30:28 - 08/03/2023] (T) Plugin aim_support loaded!
[11:30:28 - 08/03/2023] (T) Loading plugin /usr/local/share/satdump//plugins/libdvb_support.so...
[11:30:28 - 08/03/2023] (T) Plugin dvb_support loaded!
[11:30:28 - 08/03/2023] (T) Loading plugin /usr/local/share/satdump//plugins/libbluewalker3_support.so...
[11:30:28 - 08/03/2023] (T) Plugin bluewalker3_support loaded!
[11:30:28 - 08/03/2023] (T) Loading plugin /usr/local/share/satdump//plugins/libothers_support.so...
[11:30:28 - 08/03/2023] (T) Plugin others_support loaded!
[... 370 lines cut ...]
[11:30:28 - 08/03/2023] (T) Add satellite PROBA-V
[11:30:28 - 08/03/2023] (T) Add satellite CLOUDSAT
[11:30:28 - 08/03/2023] (T) Add satellite ISS (ZARYA)
[11:30:28 - 08/03/2023] (T) Add satellite GCOM-W1 (SHIZUKU)
[11:30:28 - 08/03/2023] (T) Add satellite GCOM-C (SHIKISAI)
&{true 0 <nil>}
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::substr: __pos (which is 16) > this->size() (which is 0)
```
**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

I also tried to spawning SatDump in supervisor, but I got the same result.

My Go code snippet:

```go
cmd = exec.Command("/usr/local/bin/satdump", "live", "gk2a_lrit", "/files", "--source", "sdrplay", "--samplerate", "3000000", "--frequency", "1.69214e9", "--gain", "40", "--http_server", "127.0.110.22:1234", "--agc", "--bias", "--dc_block", "true")

cmd.Stderr = os.Stderr
cmd.Stdout = os.Stdout

if err := cmd.Start(); err != nil {
    log.Fatalln("Failed to Start SatDump!")
}
```

Aang23, 2023-04-15:
Interesting, and does it work if you run it manually, not through your Go program?

bclswl0827, 2023-04-15:
SatDump works well if I run it manually

Aang23, 2023-04-20:
Hmmmm... With the exact same command?

bclswl0827, 2023-04-21:
Exactly the same. BTW, I checked many version and got a working one (But it's not the last available version)
https://github.com/SatDump/SatDump/tree/8903c566d9a62f655f47beccfcdbf380849abd4f

Aang23, 2023-04-23:
Do you have a minimal example go code I can debug with?

bclswl0827, 2023-07-19:
Thank you! This problem seems to come from environment variable settings, I switched to another account and it worked

closed 2023-07-19 by bclswl0827