SatDump/SatDump #272: NOAA 18 AVHRR Channel 3a Calibration

JVital2013, 2023-04-27:
**Description of the issue**
When the GAC/HRPT signal is weak from NOAA satellites, SatDump puts sporadic lines across channel 3A - which is otherwise black, since 3a is off on the NOAA POES sats.

There is an inconsistency in how **calibrated** channel 3a works (cch3a). On NOAA 19, cch3a is simply black - no matter what errors are in ch3a. On NOAA 18, the erroneous lines are passed along to cch3a instead of removed. I believe the desired behavior is for NOAA 18's cch3a to be black, like NOAA 19's.

Passing along the bad data to the calibrated channel causes problems with some of the RGB composites: for example, the "Natural Color" RGB composite gets red lines across it unless cch3a is removed from the formula:

[image]

**I'm not sure which of the following needs done:**
- I noticed when there is an erroneous line in ch3a, there is a correlated missing line from ch3b. Maybe the light lines in 3a should really be in 3b, and "passing along bad data to the calibrated version" is a non-issue.
- Make NOAA 18's cch3a work like NOAA 19's

**Hardware (SDR/PC/OS)**
Generic Msi2500/Msi001 SDR, running on multiple PCs. Tested Windows 10

**Version (Eg, 1.0.0, CI Build 171)**
CI Build f753584

**Logs after the crash (satdump.logs)**
No crash, and the calibration log always looks the same:

```
[02:23:11 - 27/03/2023] (I) Generating RGB Composite
[02:23:11 - 27/03/2023] (D) Composite needs channel ch3a
[02:23:11 - 27/03/2023] (T) Generating calibrated image channel 3. Range 0.000000 1.000000. Type 0
[02:23:13 - 27/03/2023] (D) Composite needs calibrated channel ch3a
[02:23:13 - 27/03/2023] (T) Max Total Width 2048
[02:23:13 - 27/03/2023] (T) Max Total Used  2048
[02:23:13 - 27/03/2023] (T) Offset for cch3a is 0
[02:23:13 - 27/03/2023] (T) Offset for ch3a is 0
[02:23:14 - 27/03/2023] (I) Done
```

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

_GAC FRM Examples - the same thing happens with HRPT, but it happens less for me since I can get a stronger HRPT signal_
- [NOAA-18](https://www.dropbox.com/s/segmjrler9lzibt/2023-04-04%200321%20UTC%20-%20NOAA%2018.frm?dl=1)
- [NOAA-19](https://www.dropbox.com/s/xsiip03ua7mo3kf/2023-03-23%200132%20UTC%20-%20NOAA%2019.frm?dl=1)

JVital2013, 2023-04-27:
Gotcha - it makes sense it's a single bit differentiating 3a from 3b. I agree with not "fudging" the data to make it work, and I'm fine with keeping the channel differentiation code as-is for the reasons you gave. It's a bad idea to coerce data into what you think it should be.

The thing is, noisy NOAA18 and 19 signals are _not_ treated the same when it comes to calibration. You can try this yourself with the GAC FRMs I provided - both of which have roughly the same amount of noise.

- **NOAA 19:** Calibrated Channel 3a **removes** erroneous lines from noisy data visible in Channel 3a
- **NOAA 18:** Calibrated Channel 3a **keeps** erroneous lines from noisy data visible in Channel 3a

This is repeatable with every sample I try. I think it's because of the satellite-specific calibration data under the "resources/calibration" folder, but I don't know enough about how the SatDump calibration system works to fix it.

**NOAA 19 Screenshot**
[image]

**NOAA 18 Screenshot**
[image]

Of course, maybe NOAA-19's processing is the broken one and it should be passing along the bad data in 3a

ZbychuButItWasTaken, 2023-04-27:
Everything is right. if you look closely, NOAA-19 data still is calibrated (though I don't think we can really talk about calibrating an empty channel). It's just the fact that the slope and intercept values are massively different (mainly due to the lack of recent data regarding NOAA-18), so the output albedo is much lower.
I'll close the issue, since there is no problem, but thanks for reporting your concerns!

closed 2023-04-27 by ZbychuButItWasTaken

JVital2013, 2023-04-27:
@ZbychuButItWasTaken You're not kidding - I should have looked A LOT closer. On NOAA-19, the output albedo for cch3a goes all the way down to 1! Indistinguishable from the actual blank parts, which are 0 - unless you know to look for it. Shows what I know 😄 .

Thank you so much for your attention to this (non)issue, and taking the time to explain it to me. I really appreciate it.

I'll just work around the negative effects it has on NOAA-18's "Natural Color" composite by removing cch3a from the composite equation. With Channel 3a off, it's effectively unused in the equation anyway.

Thanks again!