SatDump/SatDump #273: GOES HRIT Ch13 images disappearing, possibly ruining false color disks.

JB-Micro, 2023-04-30:
**Description of the issue**
When viewing the false color disks for GOES HRIT, many of them have an orange band where it appears the ch13 data was not present. When I went to check the ch13 images to see what went wrong, the ch13 images for every messed up false color image was gone. I'm not sure when they are deleted, but they are definitely present at some point since most of the false color image looks fine. My SNR hovers around 5dB, which should be plenty for a perfect image.

None of the mesoscale images show this issue, and all of the channel 2 images are present and perfect. This only seems to affect the ch13 and false color disks. This has occurred on both my computer and a friend's Pi.

**Hardware (SDR/PC/OS)**
Mine: RTL-SDR v3, Dell Latitude E7470, Linux Mint 20.3
Friend's: RTL-SDR v3, Raspberry Pi 4, Raspbian

**Version (Eg, 1.0.0, CI Build #171)**
I've checked V1.0.3 and V0.39, both have this issue.

**Logs after the crash (satdump.logs)**
I'll only post a small portion of the logs. This shows the image data being read in from channel 13, then the program just moving on without saving the final image. 
`[04/25/23 - 21:55:58] (D) This is image data. Size 5424x339
[04/25/23 - 21:55:58] (D) This is RICE-compressed! Decompressing...
[04/25/23 - 21:55:58] (D) Rice header is present!
[04/25/23 - 21:55:58] (D) Pixels per block : 32
[04/25/23 - 21:55:58] (D) Scan lines per packet : 1
[04/25/23 - 21:55:59] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/Z_QATA00KWBC260445_C_KWIN_20230426044544_294554-3-RADSTHPL.GIF...
[04/25/23 - 21:56:00] (I) Progress -nan%, SNR : 4.189666dB, Peak SNR: 6.187331dB
[04/25/23 - 21:56:00] (I) Progress -nan%, Viterbi : SYNCED BER : 0.014648, Deframer : SYNCED
[04/25/23 - 21:56:00] (I) Progress inf%
[04/25/23 - 21:56:03] (I) New LRIT file : pH-23116045551-A.lrit
[04/25/23 - 21:56:03] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/DCS/pH-23116045548-A.lrit...
[04/25/23 - 21:56:04] (I) New LRIT file : A_SAMH31PKMJ260500_C_KWIN_20230426045558_294763-2-SAHOURLY.lrit
[04/25/23 - 21:56:04] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/A_SATO31NFTF260500_C_KWIN_20230426045544_294761-2-TAF031TG.txt...
[04/25/23 - 21:56:04] (I) New LRIT file : A_SAUS70KWBC260500_C_KWIN_20230426045600_294764-2-SAHOURLY.lrit
[04/25/23 - 21:56:04] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/A_SAMH31PKMJ260500_C_KWIN_20230426045558_294763-2-SAHOURLY.txt...
[04/25/23 - 21:56:06] (I) New LRIT file : A_SAUS80KWBC260500_C_KWIN_20230426045601_294766-2-SAHOURLY.lrit
[04/25/23 - 21:56:06] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/A_SAUS70KWBC260500_C_KWIN_20230426045600_294764-2-SAHOURLY.txt...
[04/25/23 - 21:56:06] (I) New LRIT file : pH-23116045558-A.lrit
[04/25/23 - 21:56:06] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/DCS/pH-23116045551-A.lrit...
[04/25/23 - 21:56:06] (I) New LRIT file : A_SAUS70KWBC260500_C_KWIN_20230426045600_294765-2-SAHOURLY.lrit
[04/25/23 - 21:56:06] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/A_SAUS80KWBC260500_C_KWIN_20230426045601_294766-2-SAHOURLY.txt...
[04/25/23 - 21:56:07] (I) New LRIT file : A_SAUS70KWBC260500_C_KWIN_20230426045601_294767-2-SAHOURLY.lrit
[04/25/23 - 21:56:07] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/A_SAUS70KWBC260500_C_KWIN_20230426045600_294765-2-SAHOURLY.txt...
[04/25/23 - 21:56:08] (I) New LRIT file : A_SAUS70KWBC260500_C_KWIN_20230426045601_294768-2-SAHOURLY.lrit
[04/25/23 - 21:56:08] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/A_SAUS70KWBC260500_C_KWIN_20230426045601_294767-2-SAHOURLY.txt...
[04/25/23 - 21:56:08] (I) New LRIT file : A_SAUS70KWBC260500_C_KWIN_20230426045601_294769-2-SAHOURLY.lrit
[04/25/23 - 21:56:08] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/A_SAUS70KWBC260500_C_KWIN_20230426045601_294768-2-SAHOURLY.txt...
[04/25/23 - 21:56:09] (I) New LRIT file : A_SAUS70KWBC260500_C_KWIN_20230426045601_294770-2-SAHOURLY.lrit
[04/25/23 - 21:56:09] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/A_SAUS70KWBC260500_C_KWIN_20230426045601_294769-2-SAHOURLY.txt...
[04/25/23 - 21:56:10] (I) Progress -nan%, SNR : 4.805422dB, Peak SNR: 6.187331dB
[04/25/23 - 21:56:10] (I) Progress -nan%, Viterbi : SYNCED BER : 0.028076, Deframer : SYNCED
[04/25/23 - 21:56:10] (I) Progress inf%
[04/25/23 - 21:56:10] (I) New LRIT file : A_SAUS80KWBC260500_C_KWIN_20230426045601_294771-2-SAHOURLY.lrit
[04/25/23 - 21:56:10] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/A_SAUS70KWBC260500_C_KWIN_20230426045601_294770-2-SAHOURLY.txt...
[04/25/23 - 21:56:10] (I) New LRIT file : A_SAUS70KWBC260500_C_KWIN_20230426045601_294772-2-SAHOURLY.lrit
[04/25/23 - 21:56:10] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/A_SAUS80KWBC260500_C_KWIN_20230426045601_294771-2-SAHOURLY.txt...
[04/25/23 - 21:56:11] (I) New LRIT file : A_SAUS70KWBC260500_C_KWIN_20230426045601_294773-2-SAHOURLY.lrit
[04/25/23 - 21:56:11] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/A_SAUS70KWBC260500_C_KWIN_20230426045601_294772-2-SAHOURLY.txt...
[04/25/23 - 21:56:11] (I) New LRIT file : OR_ABI-L2-CMIPF-M6C15_G16_s20231160430204_e20231160439518_c20231160440007.lrit
[04/25/23 - 21:56:11] (D) This is image data. Size 5424x339
[04/25/23 - 21:56:11] (D) This is RICE-compressed! Decompressing...
[04/25/23 - 21:56:11] (D) Rice header is present!
[04/25/23 - 21:56:11] (D) Pixels per block : 32
[04/25/23 - 21:56:11] (D) Scan lines per packet : 1
[04/25/23 - 21:56:11] (I) New LRIT file : A_SAUS70KWBC260500_C_KWIN_20230426045601_294774-2-SAHOURLY.lrit
[04/25/23 - 21:56:11] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/A_SAUS70KWBC260500_C_KWIN_20230426045601_294773-2-SAHOURLY.txt...
[04/25/23 - 21:56:12] (I) New LRIT file : A_SAUS70KWBC260500_C_KWIN_20230426045601_294776-2-SAHOURLY.lrit
[04/25/23 - 21:56:12] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/A_SAUS70KWBC260500_C_KWIN_20230426045601_294774-2-SAHOURLY.txt...
[04/25/23 - 21:56:13] (I) New LRIT file : A_SAUS70KWBC260500_C_KWIN_20230426045601_294775-2-SAHOURLY.lrit
[04/25/23 - 21:56:13] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/A_SAUS70KWBC260500_C_KWIN_20230426045601_294776-2-SAHOURLY.txt...
[04/25/23 - 21:56:14] (I) New LRIT file : pH-23116045601-A.lrit
[04/25/23 - 21:56:14] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/DCS/pH-23116045558-A.lrit...
[04/25/23 - 21:56:14] (I) New LRIT file : A_SAUS80KWBC260500_C_KWIN_20230426045601_294778-2-SAHOURLY.lrit
[04/25/23 - 21:56:14] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/A_SAUS70KWBC260500_C_KWIN_20230426045601_294775-2-SAHOURLY.txt...
[04/25/23 - 21:56:14] (I) New LRIT file : A_SAUS70KWBC260500_C_KWIN_20230426045601_294777-2-SAHOURLY.lrit
[04/25/23 - 21:56:14] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/A_SAUS80KWBC260500_C_KWIN_20230426045601_294778-2-SAHOURLY.txt...
[... 17 lines cut ...]
[04/25/23 - 21:56:18] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/A_SAUS70KWBC260500_C_KWIN_20230426045602_294786-2-SAHOURLY.txt...
[04/25/23 - 21:56:19] (I) New LRIT file : A_FXUS66KPQR260456_C_KWIN_20230426045611_294791-2-AFDPQROR.lrit
[04/25/23 - 21:56:19] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/A_SAUS70KWBC260500_C_KWIN_20230426045601_294785-2-SAHOURLY.txt...
[04/25/23 - 21:56:20] (I) Progress -nan%, SNR : 4.608709dB, Peak SNR: 6.187331dB
[04/25/23 - 21:56:20] (I) Progress -nan%, Viterbi : SYNCED BER : 0.025635, Deframer : SYNCED
[04/25/23 - 21:56:20] (I) Progress inf%
[04/25/23 - 21:56:20] (I) New LRIT file : A_FPUS66KPDT260455_C_KWIN_20230426045536_294759-3-SFTPDTOR.lrit
[04/25/23 - 21:56:20] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/A_SRUS46KPQR260455_C_KWIN_20230426045513_294753-3-RRMPDXOR.txt...
[04/25/23 - 21:56:21] (I) New LRIT file : A_FOUS56KPDT260456_C_KWIN_20230426045610_294790-3-PFMPDTOR.lrit
[04/25/23 - 21:56:21] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/EMWIN/A_FPUS66KPDT260455_C_KWIN_20230426045536_294759-3-SFTPDTOR.txt...
[... cut ...]
[04/25/23 - 21:56:25] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/DCS/pH-23116045609-A.lrit...
[04/25/23 - 21:56:26] (I) New LRIT file : pH-23116045619-A.lrit
[04/25/23 - 21:56:26] (I) Writing file ./live_output/2023-04-26_02-36_goes_hrit_1694Mhz/DCS/pH-23116045612-A.lrit...
[04/25/23 - 21:56:30] (I) Progress -nan%, SNR : 4.949914dB, Peak SNR: 6.187331dB
[04/25/23 - 21:56:30] (I) Progress -nan%, Viterbi : SYNCED BER : 0.031738, Deframer : SYNCED
[04/25/23 - 21:56:30] (I) Progress inf%`
**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

thomasgolladay, 2023-04-30:
I have been getting the bug as well for GOES 18.

JB-Micro, 2023-04-30:
I've got some new weirdness. In a recent recording session, 5 ch2 images were saved and 3 ch13 images were saved.
[image]
From the screenshot you can see they're complete disks.

Satdump created 5 false color disks based on these images. All but 1 have the brown bars covering the image. I used my own false color code in python to use the existing ch2 and 13 images to make false color disks. The screenshot below shows my 3 resulting false color disks compared to the 5 false color disks that satdump output.
[image]
Since 2 of the ch13 images were missing, I couldn't make false color disks from those time periods.

Finally, here's an enlarged comparison of one of the disks:
[image]
The data to create the false color disk is clearly there, but something is malfunctioning.

JB-Micro, 2023-05-04:
Another issue has popped up now and then, I'm not sure how related it is to the missing images issue. Sometimes channel 2 or 13 will have weird strips that look like the band was pulled from a different time frame. The first two screenshots below are the lines on the same false color image, right side vs left side. The last screenshot is the same false color disk, but near the bottom. As you can see, the terminator does not line up in the strips, and the brightness is different. The clouds dont quite line up at the lines between these weird strips either.
[image]
[image]
[image]

These weird artifacts are seen on the ch2 image that produced the false color, but its a lot easier to see what's happening in the false color image. The ch13 is perfect for this time frame.

thomasgolladay, 2023-05-04:
Its only in my ch2, where there are lines in the image. It looks like a pervious image or something. 
[image]

Aang23, 2023-05-07:
I would suggest you all try a CI build / current master branch. There were other changes/fixes that perhaps could be related. If the issues persists in those versions, I'll investigate further.... But it sounds very similar to a  bug introduced by mistake a long while ago.

Aang23, 2023-05-07:
Also, on Rasbperrys, do ensure Volk is using NEON. If it is not performance will be atrocious and could cause drops & missing segments.

JB-Micro, 2023-05-07:
To troubleshoot, I moved back to my laptop to ensure the Pis aren't adding extra issues. Also, I was wrong earlier about the version my Pi was running, it had been running the master branch. 

On my laptop, I deleted Satdump and used the command line to get the master branch installed. I ran that for about 2 hours to see what I got. I ended up getting the exact same issues my Pi had been showing.

Below is a screenshot of the CH2, CJ13, and FC disks I recorded. I started recording during the T2000 time period, and I either missed the ch2 disk or it wasn't saved properly by satdump. I stopped recording after the T2130 time period. I am missing the T2100 CH13 disk that should've been there.
[image]

You can also see the false colors are messed up still.
[image]
[image]
The first disk, T2030 FC, has the wrong time period's strips. The ch2 channel was the culprit but it's harder to see in the greyscale image. The second disk, T2130 FC, is mostly brown. The ch2 and ch13 images for that time period were properly saved and look perfect. The T2100 FC shows the same issues but less severely.

Aang23, 2023-05-09:
I can't reproduce this with any older data. Could you modify the pipeline so it records CADU, let it run for perhaps 2h and then send it over?

JB-Micro, 2023-05-09:
Sure. What do I change to make it record CADU?

JB-Micro, 2023-05-10:
https://drive.google.com/file/d/1gO0u-YSBf0tDHxjnZe-g016vVp8RO7vN/view?usp=sharing
This one is ~515 MB, it shows the brown bars in the false color disks. It's a bit hard to see since it was late at night for GOES 16, but it's there. The CH2 and CH13 images for that time frame are also present and look fine.

https://drive.google.com/file/d/1XBWRFvk_mqKaQSZ3gMkwPvsysArawaOU/view?usp=sharing
This one is ~470 MB, the false color disks don't have brown bars but one of them has the wrong time period's strip, and is missing the disk that strip is from.

https://drive.google.com/file/d/1aJzTEH0vINeOUs-Pcl8OSMwNf9smPNzA/view?usp=sharing
This one is ~270 MB, it shows the wrong time period strip as well but it was on the first false color disk so I didn't get disk images from the previous time frame.

JVital2013, 2023-07-20:
I figured this out - I'll submit a PR later this afternoon once I verify some things!

Aang23, 2023-07-21:
Thanks to @JVital2013 for figuring this one one. I was always unable to reproduce on my end! Good thing this one is done.

closed 2023-07-21 by Aang23

JB-Micro, 2023-07-22:
Thanks y'all, I ran satdump overnight and everything looks fixed.