SatDump/SatDump #275: Unable to start satdump

DroidKali, 2023-05-01:
**Description of the issue**
Unable to start satdump, double-click the satdump-ui.exe but nothing to happen.Run the satdump.exe in cmd.exe as the same result.
**Hardware (SDR/PC/OS)**
RTL-SDR Blog V3 dongle
**Version (Eg, 1.0.0, CI Build #171)**
1.0.3
**Logs after the cras (satdump.logs)**
Not create the klog
**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
[image]

JVital2013, 2023-05-04:
Do you have Microsoft Visual C++ Redistributable installed? You can get it from https://learn.microsoft.com/en-US/cpp/windows/latest-supported-vc-redist?view=msvc-170

Aang23, 2023-05-07:
I'd guess this is the most likely issue here, probably a missing C++ Runtime.

cfd238, 2023-05-15:
I am having same issue.

cfd238, 2023-05-15:
I am having same issue.  Have updated C++ runtime,  issue persists.
Windows 10 Pro (unlicensed) - 10.0.19045 Build 19045
Old HP 2000 Notebook AMD e-300 APU /radeon HD 6310 graphics

Aang23, 2023-05-16:
Are you launching satdump-ui.exe?

DroidKali, 2023-05-17:
> Are you launching satdump-ui.exe?
> 
> Edit : Can you try the latest CI build just in case?

I have tryed the latest CI build and it has been fixed this issue.
Thanks for your help!
[image]

closed 2023-05-18 by DroidKali