SatDump/SatDump #276: SatDump waterfall on rtl-sdr not working on 1.0.3

bradparrott, 2023-05-02:
The Waterfall/FFT does not appear to be working using rtl-sdr SmartTee XTR in v1.0.3. Works as expected in v1.0.2 with same hardware and antenna tune. Radio does appear to start in the GUI, but waterfall remains blank.

Nooelec SmartTee XTR HF on Windows 11

V1.0.3

No files received, or crash logs created.

Feuerwerko, 2023-05-02:
You need to adjust the min and max values in the FFT settings. For an rtl-sdr Id recommend -30 FFT max and -80 FFT min. The values default to zero, so you dont see anything.

bradparrott, 2023-05-02:
Thanks, I tried again with your suggestions and it works. I thought I had tried those settings already but it appears as though I did not try hard enough. But now I'm testing with AirSpy and was able to see the waterfall with these settings. When I tried originally side-by-side v1.0.2 and v1.0.3 with the dongle connected directly I could see the waterfall on 1.0.2, but not in 1.0.3 with the same FFT settings.

closed 2023-05-02 by bradparrott