SatDump/SatDump #279: Install errors MacOS

VincentVolmer, 2023-05-10:
Hello,

Install stops at 32%. Mac OS Catalina 10.15.7 - XCode 12.4

A lot of warnings generated.
[ 32%] Linking CXX shared library ../libsatdump_core.dylib
ld: library not found for -lvolk
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libsatdump_core.dylib] Error 1
make[1]: *** [src-core/CMakeFiles/satdump_core.dir/all] Error 2
make: *** [all] Error 2

What can I do next? Thanks for any help!
Vy73 Vincent PE2V

Aang23, 2023-05-12:
How did you install volk?

VincentVolmer, 2023-05-15:
Hello,
Thank you for your answer. I followed all steps at the main page. I installed volk this way:

# macOS: Install dependencies
brew install cmake volk libpng glew glfw nng pkg-config llvm libomp luajit

VincentVolmer, 2023-05-15:
Tried it again and I see: Error: volk: SHA256 mismatch so I'll try to install with Port (if present)

VincentVolmer, 2023-05-15:
Installed volk

When I run the line: 
`cmake -DCMAKE_BUILD_TYPE=Release ..`

I see this about volk:
```
-- Checking for module 'volk'
--   Found volk, version 2.5
-- Performing Test VOLK_HAS_ALLOC
-- Performing Test VOLK_HAS_ALLOC - Failed
-- VOLK does not have volk_alloc.hh, including volk_includes
```

When running build (I have to do this first, because nproc is not a command om Mac):
`alias nproc="sysctl -n hw.logicalcpu"`

```
build % make -j`nproc` 
```
I still see this:
```
[  1%] Linking CXX shared library ../libsatdump_core.dylib
ld: library not found for -lvolk
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libsatdump_core.dylib] Error 1
make[1]: *** [src-core/CMakeFiles/satdump_core.dir/all] Error 2
make: *** [all] Error 2

```

Aang23, 2023-05-16:
Where is the .dylib installed? It sounds like Volk isn't in a standard directory so SatDump isn't able to find it.

VincentVolmer, 2023-05-16:
I don't know how to find... could you please help? Secondly I can try to install volk again with brew but it has still the SHA256 mismatch error...

AlexandreRouma, 2023-05-16:
try running `brew update` and then `brew cleanup`. Then try installing volk with it again.

VincentVolmer, 2023-05-17:
Thank... did that but still the "Error: volk: SHA256 mismatch"... so not possible to install volk with brew for now. So I reported this to https://github.com/google/cpu_features.git

Aang23, 2023-05-20:
Perhaps build and install Volk manually?

VincentVolmer, 2023-05-21:
Ok, install volk is OK now. But an other error at 95% and install stops...:

[ 95%] Built target bladerf_sdr_support
make: *** [all] Error 2

Aang23, 2023-05-21:
Did you clean it all up from scratch? Can you show the full logs? here I can't see the error.

VincentVolmer, 2023-05-22:
Hello,
I just started from "# Finally, SatDump", and before that I did 'mv satdump satdumptemp'.

VincentVolmer, 2023-05-22:
I started from scratch but it ended the same way at 95% ... maybe it is not running on my MacOS Catalina 10.15.7

VincentVolmer, 2023-09-05:
Good morning, I'll give it a try today..... first some brew updates .... :)

JVital2013, 2023-09-05:
That's the best part - brew is no longer required!

VincentVolmer, 2023-09-05:
> That's the best part - brew is no longer required!

Okay.... after brew is ready (still updating) I can start at this point?: 
# Finally, SatDump 
git clone https://github.com/altillimity/satdump.git
etc....

JVital2013, 2023-09-05:
What I meant is compiling is no longer necessary - just download the dmg from https://github.com/SatDump/SatDump/releases/tag/nightly

VincentVolmer, 2023-09-05:
Aah ohw... thats great. Goin' to try now!

VincentVolmer, 2023-09-05:
That was easy! It's running.... and now I need to find out how it is working.... Thank you for your help!!

JVital2013, 2023-09-05:
Heh, that's the trick to SatDump. I'm still learning about all it has to offer :). 

Feel free to close this issue if you feel the pre-compiled build resolves your problem!

closed 2023-09-05 by VincentVolmer