SatDump/SatDump #287: Deframer sync issues with Elektro-L LRIT on Satdump 1.x (Raspberry Pi)

pluggerlockett, 2023-06-23:
Been meaning to raise this for some time but just haven't had the time. I've been attempting to migrate my Elektro-L3 LRIT station from Satdump v0.39 to Satdump v1.x but there appears to be a Deframer sync issue in the 1.x code. I've tried both 1.0.3 release as well as cloning the repo and building from source (two versions) over the past few weeks without success. 

Compiling from source completes without issue. SNR values are fine, all I'm doing is moving the station USB cable from my old Pi to the new one. SNR is in the mid 5's. Viterbi syncs without issue, but the Deframer never syncs on the 1.x branch.

I'm running 2x Raspberry Pis in the same IP-65 box for testing. My working station is running on a Pi 3A+ and the newer build is either a 3B or 3B+. I doubt this would be a hardware issue.

Let me know if you need any screenshots or console output. Happy to test as well if you believe you've identified the root cause.

pluggerlockett, 2023-06-24:
Here's some console output highlighting what I'm seeing.
```
$ ./satdump live elektro_lrit /home/fy/software/satdump/build/L3_output --source airspy -
-samplerate 3e6 --frequency 1691e6 --general_gain 18 --gain_type 0 --bias                                                        
[09:40:39 - 17/06/2023] (I)    _____       __  ____                                                                              
[09:40:39 - 17/06/2023] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____                                                         
[09:40:39 - 17/06/2023] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \                                                        
[09:40:39 - 17/06/2023] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /                                                        
[09:40:39 - 17/06/2023] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/                                                         
[09:40:39 - 17/06/2023] (I)                                     /_/                                                              
[09:40:39 - 17/06/2023] (I) Starting SatDump v1.1.0-25fdf148                                                                     
[09:40:39 - 17/06/2023] (I)                                                                                                      
[09:40:39 - 17/06/2023] (I) Loading config satdump_cfg.json                                                                      
[09:40:39 - 17/06/2023] (I) Loading user settings /home/fy/.config/satdump/settings.json                                         
[09:40:39 - 17/06/2023] (I) Loading plugins from ./plugins                                        
...
[09:40:40 - 17/06/2023] (T) Airspy device has samplerate 3000000 SPS
[09:40:40 - 17/06/2023] (T) Airspy device has samplerate 6000000 SPS
[09:40:40 - 17/06/2023] (D) Set Airspy samplerate to 3000000
[09:40:40 - 17/06/2023] (D) Set Airspy frequency to 1691000000
[09:40:40 - 17/06/2023] (D) Set Airspy gain (sensitive) to 18
[09:40:40 - 17/06/2023] (D) Set Airspy bias to 1
[09:40:40 - 17/06/2023] (D) Set Airspy LNA AGC to 0
[09:40:40 - 17/06/2023] (D) Set Airspy Mixer AGC to 0
[09:40:40 - 17/06/2023] (D) Parameters :
[09:40:40 - 17/06/2023] (D)    - baseband_format : "f32"
[09:40:40 - 17/06/2023] (D)    - bias : true
[09:40:40 - 17/06/2023] (D)    - buffer_size : 1000000
[09:40:40 - 17/06/2023] (D)    - constellation : "bpsk"
[09:40:40 - 17/06/2023] (D)    - frequency : 1691000000
[09:40:40 - 17/06/2023] (D)    - gain_type : 0
[09:40:40 - 17/06/2023] (D)    - general_gain : 18
[09:40:40 - 17/06/2023] (D)    - pll_bw : 0.02
[09:40:40 - 17/06/2023] (D)    - rrc_alpha : 0.5
[09:40:40 - 17/06/2023] (D)    - samplerate : 3000000
[09:40:40 - 17/06/2023] (D)    - source : "airspy"
[09:40:40 - 17/06/2023] (D)    - start_timestamp : 1686994840.0
[09:40:40 - 17/06/2023] (D)    - symbolrate : 294000
[09:40:40 - 17/06/2023] (I) Module psk_demod
[09:40:40 - 17/06/2023] (D) Parameters :
[09:40:40 - 17/06/2023] (D)    - baseband_format : "f32"
[09:40:40 - 17/06/2023] (D)    - bias : true
[09:40:40 - 17/06/2023] (D)    - buffer_size : 1000000
[09:40:40 - 17/06/2023] (D)    - constellation : "bpsk"
[09:40:40 - 17/06/2023] (D)    - frequency : 1691000000
[09:40:40 - 17/06/2023] (D)    - gain_type : 0
[09:40:40 - 17/06/2023] (D)    - general_gain : 18
[09:40:40 - 17/06/2023] (D)    - pll_bw : 0.02
[09:40:40 - 17/06/2023] (D)    - rrc_alpha : 0.5
[09:40:40 - 17/06/2023] (D)    - samplerate : 3000000
[09:40:40 - 17/06/2023] (D)    - source : "airspy"
[09:40:40 - 17/06/2023] (D)    - start_timestamp : 1686994840.0
[09:40:40 - 17/06/2023] (D)    - symbolrate : 294000
[09:40:40 - 17/06/2023] (D) Parameters :
[09:40:40 - 17/06/2023] (D)    - baseband_format : "f32"
[09:40:40 - 17/06/2023] (D)    - bias : true
[09:40:40 - 17/06/2023] (D)    - buffer_size : 1000000
[09:40:40 - 17/06/2023] (D)    - cadu_size : 8192
[09:40:40 - 17/06/2023] (D)    - constellation : "bpsk"
[09:40:40 - 17/06/2023] (D)    - derandomize : true
[09:40:40 - 17/06/2023] (D)    - frequency : 1691000000
[09:40:40 - 17/06/2023] (D)    - gain_type : 0
[09:40:40 - 17/06/2023] (D)    - general_gain : 18
[09:40:40 - 17/06/2023] (D)    - rs_i : 4
[09:40:40 - 17/06/2023] (D)    - rs_type : "rs223"
[09:40:40 - 17/06/2023] (D)    - samplerate : 3000000
[09:40:40 - 17/06/2023] (D)    - source : "airspy"
[09:40:40 - 17/06/2023] (D)    - start_timestamp : 1686994840.0
[09:40:40 - 17/06/2023] (D)    - viterbi_ber_thresold : 0.3
[09:40:40 - 17/06/2023] (D)    - viterbi_outsync_after : 20                                                                      
[... 78 lines cut ...]
[09:42:20 - 17/06/2023] (I) Progress nan%, SNR : 5.951147dB, Peak SNR: 7.187170dB
[09:42:20 - 17/06/2023] (I) Progress inf
[09:42:30 - 17/06/2023] (I) Progress nan%, SNR : 5.882762dB, Peak SNR: 7.187170dB
[09:42:30 - 17/06/2023] (I) Progress nan%, Viterbi : SYNCED BER : 0.119629, Deframer : NOSYNC
[09:42:30 - 17/06/2023] (I) Progress inf
[09:42:40 - 17/06/2023] (I) Progress nan%, SNR : 5.908037dB, Peak SNR: 7.187170dB
[09:42:40 - 17/06/2023] (I) Progress nan%, Viterbi : SYNCED BER : 0.014648, Deframer : NOSYNC
[09:42:40 - 17/06/2023] (E) LRIT CRC is invalid... Skipping.
[09:42:40 - 17/06/2023] (I) Progress inf
[09:42:50 - 17/06/2023] (I) Progress nan%, SNR : 5.794511dB, Peak SNR: 7.187170dB
[09:42:50 - 17/06/2023] (I) Progress nan%, Viterbi : SYNCED BER : 0.026855, Deframer : NOSYNC
[09:42:50 - 17/06/2023] (I) Progress inf
[... cut ...]
[09:42:56 - 17/06/2023] (E) LRIT CRC is invalid... Skipping.
[09:42:58 - 17/06/2023] (E) LRIT CRC is invalid... Skipping.
[09:43:00 - 17/06/2023] (I) Progress nan%, SNR : 6.069541dB, Peak SNR: 7.187170dB
[09:43:00 - 17/06/2023] (I) Progress nan%, Viterbi : SYNCED BER : 0.039062, Deframer : NOSYNC
[09:43:00 - 17/06/2023] (I) Progress inf
^C[09:43:03 - 17/06/2023] (W) SIGINT Received. Stopping.
[09:43:04 - 17/06/2023] (I) Stop processing
[09:43:04 - 17/06/2023] (I) Demodulation finished
[09:43:04 - 17/06/2023] (I) Done! Goodbye
```

[image]

RAD750, 2023-06-24:
Apologies if I sound pedantic, but this is just to clarify:
If you run 0.39 it works fine (no changes to the RF side of things), right?
Did you specify `--DCMAKE_BUILD_TYPE=Release` when running cmake?
Did you run `volk_profile` before compiling?
Do you notice an increased CPU load when running the latest Satdump when compared to 0.39?

Thanks for your time :smile:

pluggerlockett, 2023-06-24:
>  If you run 0.39 it works fine (no changes to the RF side of things), right?

That's correct. I've had a station running On Raspbian 10 (buster) with Satdump v0.39 for over a year. Raspberry Pi 3 Model A Plus Rev 1.0 (BCM2835 @ 1.4GHz and 512MB RAM). I move the USB extension from that to my new build running Raspbian 11 (bullseye) and Satdump v1.x. Raspberry Pi 3 Model B Rev 1.2 (BCM2835 @ 1.2GHz and 1GB RAM) to test.

>  Did you specify `--DCMAKE_BUILD_TYPE=Release` when running cmake?

Yes.

>  Did you run `volk_profile` before compiling?

Yes.

>  Do you notice an increased CPU load when running the latest Satdump when compared to 0.39?

I've not checked this specifically no. But anecdotally I can say I've had issues with `satdump live` on 1.x with FY-2G as well.

Aang23, 2023-06-26:
So, I'm seeing on issue already, but it also should not be changing this much. Could you record a full dump (with some spare no-signal before/after) of LRIT so I can reproduce your setup 1:1 and test this way?

You could also try updating Volk. I made a PR a while ago to improve performance on ARM drastically. 

```
[09:40:40 - 17/06/2023] (T) Volk does not have the spiral/neonspiral kernel, will default to bundled generic.
[09:40:40 - 17/06/2023] (T) Volk does not have the spiral/neonspiral kernel, will default to bundled generic.
```

pluggerlockett, 2023-06-27:
I can definitely do a full dump record. If you don't mind, what command do you want me to run (or what config would you like me to use)? I ask because given the deframer sync issue I don't get any output in my nominated folder during or after a transmission. I'm assuming you'd want me to run `satdump record` ?

Thanks for the response, I really appreciate it.

Aang23, 2023-06-27:
Yes, `satdump record` if possible. Thanks!

pluggerlockett, 2023-06-28:
I'll have a look and get back to you in a few days. Thank you!

pluggerlockett, 2023-07-12:
Over the past week I've done more testing, unfortunately I'm having issues with getting `satdump record` to work full stop. I also rebuilt from source and still couldn't get record to work properly.
Given that I decided to deploy another dish and point to GK-2A LRIT, as tbh I'm keen on getting a station up and running for that as well, especially given I've got everything in hand hardware wise to deploy. FYI the same issue occurs as with Elektro-L3 LRIT where Viterbi syncs but the Deframer never does. Furthermore (and unlike L3) my SNR has taken quite a significant hit and instead of sitting around 8dB for GK-2A on x64 (a 12 year old i7 when aligning the dish) it's roughly 5ish dB with the Pi 3B+.
Given all this, and taking into account how much time I've sunk into this, I'm most likely going to start migrating to x86/x64 thin clients or small PCs. I've always had good success with those architectures combined with satdump and beyond my Elektro-L3 station running on a Pi 3A with Debian 10 I've never been able to get satdump to work reliably on arm. Not trying to give you any stick here btw, just trying to give you a bit of an indication of where I'm heading. I'll keep you updated.

pluggerlockett, 2023-08-07:
Hey Alan, quick FYI, all is working down here with GK-2A LRIT. I started by testing with a HP t520 thin client; you were correct, it didn't have enough grunt to get Deframer Sync either. On Friday my nuc arrived. I grabbed a wo-we Mini PC with AMD Excavator A9-9400 For $100USD and now my GK-2A station has been running for days. Happy times.

pluggerlockett, 2023-11-03:
@JVital2013 nope, I'm happy for you to close this one.

closed 2023-11-03 by JVital2013