SatDump/SatDump #289: Wrongly processing an audio .wav file for APT as baseband causes satdump to stop processing any file

RAD750, 2023-06-25:
**Description of the issue**
Wrongly processing an audio .wav file for APT as baseband causes satdump to stop processing any file until it is restarted

**Hardware (SDR/PC/OS)**
Not hardware specific, but it's an HP Elitebook 820 G2.

OS: Linux Mint 21.1

**Version (Eg, 1.0.0, CI Build #171)**
1.1.0-25fdf148

**Logs after the cras (satdump.logs)**
[satdump.log](https://github.com/SatDump/SatDump/files/11859217/satdump.log)

**Steps to reproduce**
1. Obtain an audio recording of a NOAA APT satellite, such as NOAA 19, from another program such as SDR++ or GQRX.
2. Configure the satdump window as follows
[image]

3. Push start

[image]
4. Of course, it will give lots of errors (which is expected)
[image]

5. The start button no longer works, even if you change pipeline or reselect the file.

5.

MrQbasic, 2023-06-25:
[image]

I get the same error + this is my processing screen. Also if you switch to other tabs while processing you only see a blank window. Same version / build as you can see in the titlebar of the window.

RAD750, 2023-06-25:
@MrQbasic yours is not related to the bug report, and I can see a composite. Try using the mouse to enlarge the product viewer.if you still can't, open another bug report

RAD750, 2023-07-01:
The same happens if you press abort

Aang23, 2023-07-02:
Well I can reproduce, what a weird one.

commit b5989c89 (2023-07-02): Fix offline hanging on fail. Fixes #289
closed 2023-07-02 by Aang23