SatDump/SatDump #296: Crash when demodulating

KentuckyFriedData, 2023-07-02:
**Description of the issue**
App crashes anytime when demodulating at any progress percentage, with any pipeline and attached are logs from the "Report to Apple" dialog and lldb

**Hardware (SDR/PC/OS)**
SDR: Using a baseband level file
PC: MacBook Pro M2
OS: macOS Ventura 13.4

**Version (Eg, 1.0.0, CI Build #171)**
Built locally using commit 94c9453d25856ce2861e9f0e0d34185054961d78

**Logs after the crash (satdump.logs)**
No useful information in logs

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
[Report from apple dialog.txt](https://github.com/SatDump/SatDump/files/11927766/Report.from.apple.dialog.txt)
[lldb.txt](https://github.com/SatDump/SatDump/files/11927767/lldb.txt)

Aang23, 2023-07-02:
Is this running on Rosetta or ARM directly? I'll admit it's a bit hard to debug like as I have no access to any M1/M2 hardware...

Can you redo lldb in -DCMAKE_BUILD_TYPE=Debug?

KentuckyFriedData, 2023-07-02:
Running directly on ARM, here is with the Debug build type.
[lldb.txt](https://github.com/SatDump/SatDump/files/11929584/lldb.txt)

KentuckyFriedData, 2023-07-06:
Was the same issue as https://github.com/SatDump/SatDump/issues/302 and i can confirm https://github.com/SatDump/SatDump/commit/089258df38773d7a3dd29d12cceb7ba5ae8a191f fixed it

closed 2023-07-06 by KentuckyFriedData