SatDump/SatDump #298: Add geotiff generation for the main instruments.

lpmrfentazis, 2023-07-03:
**What would you want to be added to satdump?**
It would be great if geotiff and/or, if possible, other formats (L1B, geojson, hdf5, kml, kmz) were also created in combination with png images. This allows you to get additional data for analysis in specialized software (for example, ArcGIS, QGIS, google earth, etc.).

**Describe the feature you'd like to have**
Add the creation of the above formats either always or optionally. (It is important that it can be enabled using the CLI)

**Additional context**
I am primarily interested in AVHRR data (their analogues on FY and METEOR), MODS and VIIRS.

**Off topic**
I'm also wondering when to wait for the next release for windows, I would like to see the feature requested [here](https://github.com/SatDump/SatDump/issues/227). 

It may also be worth creating some kind of community chat to solve not so significant issues in order to create an issue. Maybe telegram or some other platform will do?

Aang23, 2023-07-07:
This could be done relatively easily for many instruments - but perhaps as an option.

Libraries such as HDF5, etc required to write those files are extremely large, and would be a huge dependency many users do not need either. So rather an optional module.

There actually is a chat room - see the Matrix link in the README.

You can get newer Windows builds from the nightlys.

lpmrfentazis, 2023-07-07:
> Это можно было бы сделать относительно легко для многих инструментов - но, возможно, как вариант.
> 
> Библиотеки, такие как HDF5 и т.д., Необходимые для записи этих файлов, Чрезвычайно велики и были бы огромной зависимостью, которая многим пользователям тоже не нужна. Так что скорее необязательный модуль.
> 
> На самом деле есть комната чата - смотрите ссылку на матрицу в README.
> 
> Вы можете получать более новые сборки Windows из nightlys.

 would like to see it even as an option =) is it possible to expect this functionality in the foreseeable future? Maybe I can help in some way?

RAD750, 2023-07-12:
I'd suggest adding GeoTIFF along with other image formats as well, it could enable easy import into GRASS GIS or similar for further processing.

ivica3730k, 2023-10-15:
Hello, just a quick question @Aang23 , any thoughts of adding this in?

would be cool to be able to store geotiffs in something called stac-catalog :) 

https://github.com/stac-utils/stac-fastapi-pgstac

ivica3730k, 2023-10-15:
> Hello, just a quick question @Aang23 , any progress on adding cogs/geotiffs in?
> 
> would be cool to be able to store geotiffs in something called stac-catalog :)
> 
> https://github.com/stac-utils/stac-fastapi-pgstac

https://github.com/developmentseed/rio-stac

RAD750, 2023-10-16:
Work is in progress right now

ivica3730k, 2023-10-16:
> Work is in progress right now

thank you very much!

Aang23, 2024-03-01:
This is now done!

closed 2024-03-01 by Aang23