SatDump/SatDump #301: Crash while generating NOAA 18 APT composite

moh53n, 2023-07-06:
**Description of the issue**
When I try to process a NOAA 18 APT capture, SatDump will throw this error after successfully generating the avhrr_apt_B_Equalized.png:
`[sol2] An error occurred and panic has been invoked: stack index 1, expected string, received no value: bad get from protected_function_result (is not an error) `
`(E) Error generating composite! lua: error: stack index 1, expected string, received no value: bad get from protected_function_result (is not an error)`
**Hardware (SDR/PC/OS)**
Linux moh53n 6.4.1-arch1-1 #1 SMP PREEMPT_DYNAMIC Sat, 01 Jul 2023 16:17:21 +0000 x86_64 GNU/Linux
**Version (Eg, 1.0.0, CI Build #171)**
Built locally on 25199ec with "-DCMAKE_BUILD_TYPE=Debug"
**Logs after the crash (satdump.logs)**
[log.txt](https://github.com/SatDump/SatDump/files/11963810/log.txt)
**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
[product.cbor.zip](https://github.com/SatDump/SatDump/files/11963830/product.cbor.zip)

moh53n, 2023-07-06:
Now I tried to process a daylight capture from NOAA 19, And instead of an error, it simply hung after avhrr_apt_B_Equalized. On ARM64 however, it will end up in SEGFAULT.

thecivvie, 2023-07-30:
I have the same issue with NOAA 15 as well as 18 and 19

Aang23, 2023-08-02:
Can you send over your file and full logs?

thecivvie, 2023-08-03:
Will upload the logs when the next pass goes through. I use a Pi and cli only

thecivvie, 2023-08-03:
Just emailed them to you

JVital2013, 2023-10-30:
@thecivvie is this still happening for you with the latest nightly? We recently made some changes to the APT decoder.

If it is still happening, can you try uninstalling satdump, uninstall all LuaJIT packages from apt, then rebuild/re-install SatDump? I'm not sure it will fix it, but it is worth trying.

JVital2013, 2023-11-17:
We haven't heard from you so I'm going to close this issue. If you'd like continued assistance, feel free to reach back out!

closed 2023-11-17 by JVital2013