SatDump/SatDump #303: UI Scaling issues on MacOS hidpi display

ericye16, 2023-07-07:
**Description of the issue**

On my M1 macbook built-in display which has a retina display with 2x scaling, many UI elements are cut off on the right and bottom of the window and there are no scrollbars, so I can't see half the FFT or any of the waterfall or many other things. I also can't change any settings in the settings tab because all the buttons to change things are too far on the right and cut off. This change: https://github.com/SatDump/SatDump/commit/4d18d3c34f813120ddf604da99f0cfdb8e6cf7f9 fixed it for me, but I'm not able to test it on other platforms to check that I didn't break anything.

**Hardware (SDR/PC/OS)**

**Version (Eg, 1.0.0, CI Build #171)**

**Logs after the cras (satdump.logs)**

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

Aang23, 2023-07-07:
Can you open a PR with it? I'll test it on a few platforms.

ericye16, 2023-07-08:
Please see https://github.com/SatDump/SatDump/pull/304, thanks!

ericye16, 2023-07-10:
Fixed by https://github.com/SatDump/SatDump/pull/304

closed 2023-07-10 by ericye16