SatDump/SatDump #305: Build freezes during startup without any errors

eggix, 2023-07-08:
**Description of the issue**
SatDump freezing during startup. Log stops after/during loading of pipelines. The GUI window is created but is completely transparent with no visiable content. I get an Error (E) if I do not supply any arumgents to `satdump-ui`. Is this normal? With arguments this Error is gone but the program still freezes.

**Hardware (SDR/PC/OS)**
Build and run on Arch Linux (upto-date).

<details>
<summary>Version of dependencies</summary>

```
Name                     : libvolk
Version                  : 2:3.0.0-2

Name                     : glfw-x11
Version                  : 3.3.8-1

Name                     : glew
Version                  : 2.2.0-6

Name                     : fftw
Version                  : 3.3.10-4

Name                     : rtl-sdr
Version                  : 1:0.8.0-4

Name                     : libusb
Version                  : 1.0.26-2

Name                     : libiio
Version                  : 0.24-3

Name                     : libad9361
Version                  : 0.2-8

Name                     : ocl-icd
Version                  : 2.3.2-1

Name                     : zstd
Version                  : 1.5.5-1

Name                     : nng
Version                  : 1.5.2-1

Name                     : libpng
Version                  : 1.6.40-2

Name                     : luajit
Version                  : 2.1.0.beta3.r481.g51fb2f2c-1
```
</details>

**Version**
`1.1.0 / nighly` https://github.com/SatDump/SatDump/commit/57cd337db32d30f0c0ad6aee7ec6312f1a3260dd

**Logs after the crash**
<details>
<summary>satdump.logs</summary>

```
[09:23:15 - 08/07/2023] (E) Usage : ./satdump-ui [downlink] [input_level] [input_file] [output_file_or_directory] [additional options as required]
[09:23:15 - 08/07/2023] (E) Extra options (examples. Any parameter used in modules can be used here) :
[09:23:15 - 08/07/2023] (E)  --samplerate [baseband_samplerate] --baseband_format [f32/i16/i8/w8] --dc_block --iq_swap
[09:23:15 - 08/07/2023] (D) Starting with OpenGL 4.6 (Core Profile) Mesa 23.1.3
[09:23:15 - 08/07/2023] (I)    _____       __  ____                      
[09:23:15 - 08/07/2023] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ 
[09:23:15 - 08/07/2023] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[09:23:15 - 08/07/2023] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[09:23:15 - 08/07/2023] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ 
[09:23:15 - 08/07/2023] (I)                                     /_/      
[09:23:15 - 08/07/2023] (I) Starting SatDump v1.1.0
[... 510 lines cut ...]
Found AVX2 features, enabling AVX2 plugin
Building with Airspy support
AirspyHF Library could not be found! Not building.
Building with HackRF support
LimeSDR Library could not be found! Not building.
SDRlay Library could not be found! Not building.
Building with PlutoSDR support
Building with BladeRF support
-- Performing Test BLADERF_HAVE_WIDEBAND
-- Performing Test BLADERF_HAVE_WIDEBAND - Failed
Building with USRP support
Building with RTL-SDR support
Building with MiriSDR support
Building with SpyServer support
Building with RTL-TCP support
Building with SDR++ Server support
Building with RtAudio sink
AAudio Library could not be found! Not building.
-- Configuring done (14.4s)
-- Generating done (0.6s)
-- Build files have been written to: ...
```
</details>

Is everything properly configured?

Aang23, 2023-07-13:
What are your HW specifications?

eggix, 2023-07-17:
Thinkpad T450s:

- Intel Core i5-5300U
- 8GB DDR3
- Intel HD Graphics 5500
- 1920x1080 Screen variant

RTL-SDR Blog V.3
freezing occurs with and without the stick plugged in.

Aang23, 2023-08-02:
Does this still occur? Can you run it in GDB to see where it hangs?

JVital2013, 2023-10-30:
@eggix is this still happening to you? If so, we'll keep this issue open as long as we need to help you get to the bottom of this.

If not, or we don't hear from you, we'll close this issue in a week or two since we cannot reproduce the problem without more information.

JVital2013, 2023-11-17:
We haven't heard from you so I'm going to close this issue. If you'd like continued assistance, feel free to reach back out!

closed 2023-11-17 by JVital2013

Kappa971, 2024-02-15:
Hi. I don't know if the problem I'm having is the same one described here. I'm trying Satdump v1.1.3 on an old laptop running Debian 12 but it crashes on startup. It crashes here:
```
[11:36:24 - 15/02/2024] (T) Loading plugin /usr/lib/satdump/plugins/libdvb_support.so...
Istruzione non consentita
```
This is the full log: [satdump-ui-log.txt](https://github.com/SatDump/SatDump/files/14297156/satdump-ui-log.txt)

Hardware:
Intel Pentium SU4100 64-Bit (Instrunctions: ht lm nx pae sse sse2 sse3 ssse3)
Intel GMA 4500 MHD (OpenGL 2.1, no OpenCL)
4gb RAM DDR2

Gqrx works fine (uses 70% CPU so it's ok) but Satdump doesn't.

JVital2013, 2024-02-15:
@Kappa971 this sounds like a different issue - besides, our OpenGL system has been largely reworked since this issue. In your case, uninstall SatDump and clear your build folder, then re-run cmake like this:

`cmake -DCMAKE_BUILD_TYPE=Release -DPLUGIN_DVB=OFF -DBUILD_OPENCL=OFF -DCMAKE_INSTALL_PREFIX=/usr ..`

Kappa971, 2024-02-15:
> @Kappa971 this sounds like a different issue - besides, our OpenGL system has been largely reworked since this issue. In your case, uninstall SatDump and clear your build folder, then re-run cmake [...]

Thanks for the reply, I'll try it.

JVital2013, 2024-02-15:
@Kappa971 oops sorry, I'm reviewing the CMake file again and I don't think that will work correctly. If it doesn't, just delete `/usr/lib/satdump/plugins/libdvb_support.so`

Kappa971, 2024-02-15:
> Glad to hear it launches for you! Thanks for the logs; until we get this issue fixed, you'll need to delete that .so file every update.
> 
> There is something wrong here - follow open issue #355 to track this problem, since this issue is unrelated to your problem, and #355 is the same as yours. I should be able to reproduce this issue, and will tag you/others with it once we think we have it fixed.

Okay, thanks for your help.

P.S.
Sorry, I didn't notice that open issue.

Kappa971, 2024-02-15:
I think I discovered another problem in Debian 12 Xfce (I don't know if the same thing happens in other Linux desktop environments, on Windows it works): The `Open` buttons do not open the window to choose the folder, for example in `Viewer` --> `Load Dataset` the Open button doesn't work, none of the `Open` buttons work. 
Should I open a new issue for this?

JVital2013, 2024-02-15:
For different issues, I'd recommend opening a new issue - especially since we're probably sending emails to this issue's creator every time we send a message here.

The issue is probably a missing package - try `sudo apt install zenity`, then try again