SatDump/SatDump #308: two `satdump live` processes on a single host

pluggerlockett, 2023-07-12:
I'm not sure if this already exists, but I was wondering/planning on doing some testing to see if I can get `satdump live` to run two pipelines separately on a single host. I've got two stations in the same physical location, one pointing to Elektro-L3 and using an AirSpy Mini and the other pointing to GK-2A and using a RTL-SDR v3. My hope is that I can `satdump live` for each simultaneously.  Is this supported?

Aang23, 2023-07-12:
There is no reason you couldn't just start 2 instances of `satdump live`. I have done that several times with no issues. You will just need to specify the SDR in some cases, etc.

Aang23, 2023-07-13:
I'll close as this isn't an issue (as of now?).

closed 2023-07-13 by Aang23

pluggerlockett, 2023-07-14:
No worries. Gonna start with a x86 thin client, and then progress to a x64 architectured host if that doesn't work. I'll keep you in the loop!