SatDump/SatDump #315: add Frequency to Spectrum/Waterfall window

db2os, 2023-07-20:
Add frequencies to the X-Axis, draw a small red line indicating the frequency we are currently centered. With a larger text windows, showing the frequency

Something like this:
[image]

db2os, 2023-07-22:
Another suggestion for the Spectrum; not only should it include frequency notation, but it should also carry the date & time somewhere.

db2os, 2023-07-28:
Already in use!  - Thanks :-)
[image]

Aang23, 2023-07-28:
Still somewhat basic, could maybe use some improvements later... But it's there!

closed 2023-07-28 by Aang23