SatDump/SatDump #323: New build from July crrash when started and close all windows

IZ5RZR, 2023-07-26:
**Description of the issue**
In recent Builds downloaded atdump once unpacked and configured the first time it works, all subsequent as soon as the start_ui window opens after a while it crashes and closes. previous June releases work fine
**Hardware (SDR/PC/OS)**
Windows 11 64 bit Intel Core I5 16 Gb ram
**Version (Eg, 1.0.0, CI Build #171)**
a435168
**Logs after the cras (satdump.logs)**
[19:11:37 - 26/07/2023] (I) Loading plugins from ./plugins
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\aim_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin aim_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\airspyhf_sdr_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin airspyhf_sdr_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\airspy_sdr_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin airspy_sdr_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\analog_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin analog_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\bladerf_sdr_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin bladerf_sdr_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\bluewalker3_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin bluewalker3_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\cubesat_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin cubesat_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\dmsp_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin dmsp_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\dvb_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin dvb_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\elektro_arktika_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin elektro_arktika_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\eos_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin eos_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\fengyun2_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin fengyun2_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\fengyun3_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin fengyun3_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\fengyun4_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin fenyun4_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\gcom_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin gcom_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\geonetcast_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin geonetcast_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\gk2a_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin gk2a_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\goes_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin goes_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\hackrf_sdr_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin hackrf_sdr_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\himawari_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin himawari_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\inmarsat_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin inmarsat_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\jason3_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin jason3_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\jpss_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin jpss_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\landsat_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin landsat_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\limesdr_sdr_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin limesdr_sdr_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\mats_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin mats_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\meteor_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin meteor_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\mirisdr_sdr_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin mirisdr_sdr_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\noaa_metop_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin noaa_metop_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\oceansat_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin oceansat_support loaded!
[19:11:37 - 26/07/2023] (T) Loading plugin ./plugins\orbcomm_support.dll...
[19:11:37 - 26/07/2023] (T) Plugin orbcomm_support loaded!
[... 367 lines cut ...]
[19:11:37 - 26/07/2023] (D)  - timed_dump
[19:11:37 - 26/07/2023] (D)  - grace_fo_link
[19:11:37 - 26/07/2023] (D)  - cfosat1_dump
[19:11:37 - 26/07/2023] (D)  - eumetcast_africa
[19:11:37 - 26/07/2023] (D)  - smos_tlm
[19:11:37 - 26/07/2023] (D)  - neossat_dump
[19:11:37 - 26/07/2023] (D)  - formosat5_s_dump
[19:11:37 - 26/07/2023] (D)  - ops_sat_s_link
[19:11:37 - 26/07/2023] (D)  - swot_tlm
[19:11:37 - 26/07/2023] (D)  - tiahne1_tlm
[19:11:37 - 26/07/2023] (D)  - paz_s_tlm
[19:11:37 - 26/07/2023] (D)  - gcom_s_band
[19:11:37 - 26/07/2023] (D)  - cosmos_2518_sband
[19:11:37 - 26/07/2023] (I) Found OpenCL Devices (1) :
[19:11:37 - 26/07/2023] (D)  - gfx803
[19:11:37 - 26/07/2023] (I) Loading TLEs from ./satdump_tles.txt
[19:11:37 - 26/07/2023] (T) Add satellite METOP-A
[19:11:37 - 26/07/2023] (T) Add satellite METOP-B
[19:11:37 - 26/07/2023] (T) Add satellite METOP-C
[19:11:37 - 26/07/2023] (T) Add satellite NOAA 15
[19:11:37 - 26/07/2023] (T) Add satellite NOAA 18
[19:11:37 - 26/07/2023] (T) Add satellite NOAA 19
[19:11:37 - 26/07/2023] (T) Add satellite FENGYUN 3A

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

Aang23, 2023-07-27:
Does that happen only after configuration? Or just after starting it, closing, and attempting to restart?

IZ5RZR, 2023-07-27:
> Does that happen only after configuration? Or just after starting it, closing, and attempting to restart?

After configuration

IZ5RZR, 2023-07-27:
> > Does that happen only after configuration? Or just after starting it, closing, and attempting to restart?
> 
> After configuration

New today version still crash after start UI, My sdr is an SDRPlay and work good on 1 month older version

IZ5RZR, 2023-07-30:
> Can you send me your config file? settings.json

OK, HERE IS FILE
{
    "user": {
        "viewer_state": {
            "panel_ratio": 0.23000000417232513,
            "borders_color": [
                0.0,
                1.0,
                0.0
            ]
        },
        "recorder_state": {
            "fft_avg": 0.05400000140070915,
            "fft_max": -53.39099884033203,
            "fft_min": -94.63500213623047,
            "fft_rate": 120,
            "fft_size": 8192,
            "panel_ratio": 0.39024388790130615,
            "show_waterfall": true,
            "waterfall_palette": "Classic",
            "waterfall_rate": 60,
            "waterfall_ratio": 0.39470013976097107
        },
        "recorder_sdr_settings": {
            "RSP2 180804B911": {
                "agc_mode": 0,
                "am_notch": false,
                "antenna_input": 2,
                "bias": false,
                "dab_notch": false,
                "fm_notch": false,
                "if_gain": 34,
                "lna_gain": 6,
                "samplerate": 3000000,
                "frequency": 1700000000.0
            }
        }
    }
}

IZ5RZR, 2023-07-30:
In this PC won't load tracking interface, I Think that is problem why satdump crash. In other I7 pc with windows 11 it works and not crash . How I can resolve?

Aang23, 2023-07-31:
Can you try from a clean one, and report every button clicked / setting changed? That way I will try to reproduce on Windows. Doesn't occur if I don't change anything. So perhaps it happens after using an SDR, etc.

IZ5RZR, 2023-08-01:
HI. So I did some tests with windows 11, first of all I tested putting the unzipped folder in C: and from properties I told it to run in compatibility mode with windows 8, but nothing, it always crashes after the second boot.
Now checking the gui scroll on startup I saw that it doesn't load some things like:
[14:41:21 - 01/08/2023] (E) Error loading ./plugins\portaudio_audio_sink.dll! Error : ".\plugins\portaudio_audio_sink.dll": Cannot find the specified module.
[14:41:22 - 01/08/2023] (W) Module gcom_amsr2 is not loaded. Pipeline Skipping!
[14:41:22 - 01/08/2023] (W) Module 8psk_demod is not loaded. Pipeline Skipping!
and as you can see at the end of the scroll it does not load the satellites, it stops at fengyun or suomi and closes the gui, the loads of many satellites are missing. I think here is the problem
[14:43:33 - 01/08/2023] (T) Add satellite METOP-A
[14:43:33 - 01/08/2023] (T) Add satellite METOP-B
[14:43:33 - 01/08/2023] (T) Add satellite METOP-C
[14:43:33 - 01/08/2023] (T) Add satellite DMSP 5D-3 F16 (USA 172)
[14:43:33 - 01/08/2023] (T) Add satellite DMSP 5D-3 F17 (USA 191)
[14:43:33 - 01/08/2023] (T) Add satellite DMSP 5D-3 F18 (USA 210)
[14:43:33 - 01/08/2023] (T) Add satellite NOAA 15
[14:43:33 - 01/08/2023] (T) Add satellite NOAA 18
[14:43:33 - 01/08/2023] (T) Add satellite NOAA 19
[14:43:33 - 01/08/2023] (T) Add satellite FENGYUN 3A
[14:43:33 - 01/08/2023] (T) Add satellite FENGYUN 3B
[14:43:33 - 01/08/2023] (T) Add satellite FENGYUN 3C
[14:43:33 - 01/08/2023] (T) Add satellite FENGYUN 3D
[14:41:21 - 01/08/2023] (E) Usage : C:\Users\Utente-XB\Desktop\SatDump-Windows_x64_MSVC (4)\satdump-ui.exe [downlink] [input_level] [input_file] [output_file_or_directory] [additional options as required]
[14:41:21 - 01/08/2023] (E) Extra options (examples. Any parameter used in modules can be used here) :
[14:41:21 - 01/08/2023] (E)  --samplerate [baseband_samplerate] --baseband_format [f32/i16/i8/w8] --dc_block --iq_swap
[14:41:21 - 01/08/2023] (D) Starting with OpenGL 3.2.14761 Core Profile Forward-Compatible Context 21.30.23.04 30.0.13023.4001
[14:41:21 - 01/08/2023] (I)    _____       __  ____
[14:41:21 - 01/08/2023] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____
[14:41:21 - 01/08/2023] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[14:41:21 - 01/08/2023] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[14:41:21 - 01/08/2023] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/
[14:41:21 - 01/08/2023] (I)                                     /_/
[14:41:21 - 01/08/2023] (I) Starting SatDump v1.1.0-8c15183
[14:41:21 - 01/08/2023] (I)
[14:41:21 - 01/08/2023] (I) Loading config satdump_cfg.json
[14:41:21 - 01/08/2023] (W) No user configuration found! Keeping defaults.
[14:41:21 - 01/08/2023] (I) Loading plugins from ./plugins
[14:41:21 - 01/08/2023] (T) Loading plugin ./plugins\aaronia_sdr_support.dll...
[14:41:21 - 01/08/2023] (E) Error loading ./plugins\aaronia_sdr_support.dll! Error : ".\plugins\aaronia_sdr_support.dll": Impossibile trovare il modulo specificato.
[14:41:21 - 01/08/2023] (T) Loading plugin ./plugins\aim_support.dll...
[14:41:21 - 01/08/2023] (T) Plugin aim_support loaded!
[14:41:21 - 01/08/2023] (T) Loading plugin ./plugins\airspyhf_sdr_support.dll...
[14:41:21 - 01/08/2023] (T) Plugin airspyhf_sdr_support loaded!
[14:41:21 - 01/08/2023] (T) Loading plugin ./plugins\airspy_sdr_support.dll...
[14:41:21 - 01/08/2023] (T) Plugin airspy_sdr_support loaded!
[14:41:21 - 01/08/2023] (T) Loading plugin ./plugins\analog_support.dll...
[14:41:21 - 01/08/2023] (T) Plugin analog_support loaded!
[14:41:21 - 01/08/2023] (T) Loading plugin ./plugins\bladerf_sdr_support.dll...
[14:41:21 - 01/08/2023] (T) Plugin bladerf_sdr_support loaded!
[14:41:21 - 01/08/2023] (T) Loading plugin ./plugins\bluewalker3_support.dll...
[14:41:21 - 01/08/2023] (T) Plugin bluewalker3_support loaded!
[14:41:21 - 01/08/2023] (T) Loading plugin ./plugins\cubesat_support.dll...
[14:41:21 - 01/08/2023] (T) Plugin cubesat_support loaded!
[14:41:21 - 01/08/2023] (T) Loading plugin ./plugins\dmsp_support.dll...
[14:41:21 - 01/08/2023] (T) Plugin dmsp_support loaded!
[14:41:21 - 01/08/2023] (T) Loading plugin ./plugins\dscovr_support.dll...
[14:41:21 - 01/08/2023] (T) Plugin dscovr_support loaded!
[14:41:21 - 01/08/2023] (T) Loading plugin ./plugins\dvb_support.dll...
[14:41:21 - 01/08/2023] (T) Plugin dvb_support loaded!
[14:41:21 - 01/08/2023] (T) Loading plugin ./plugins\elektro_arktika_support.dll...
[14:41:21 - 01/08/2023] (T) Plugin elektro_arktika_support loaded!
[14:41:21 - 01/08/2023] (T) Loading plugin ./plugins\eos_support.dll...
[14:41:21 - 01/08/2023] (T) Plugin eos_support loaded!
[14:41:21 - 01/08/2023] (T) Loading plugin ./plugins\fengyun2_support.dll...
[14:41:21 - 01/08/2023] (T) Plugin fengyun2_support loaded!
[14:41:21 - 01/08/2023] (T) Loading plugin ./plugins\fengyun3_support.dll...
[14:41:21 - 01/08/2023] (T) Plugin fengyun3_support loaded!
[14:41:21 - 01/08/2023] (T) Loading plugin ./plugins\fengyun4_support.dll...
[14:41:21 - 01/08/2023] (T) Plugin fenyun4_support loaded!
[14:41:21 - 01/08/2023] (T) Loading plugin ./plugins\gcom_support.dll...
[14:41:21 - 01/08/2023] (T) Plugin gcom_support loaded!
[14:41:21 - 01/08/2023] (T) Loading plugin ./plugins\geonetcast_support.dll...
[14:41:21 - 01/08/2023] (T) Plugin geonetcast_support loaded!
[... 512 lines cut ...]
[14:43:14 - 01/08/2023] (I) http://www.celestrak.org/NORAD/elements/gp.php?CATNR=43215
[... cut ...]
[14:43:33 - 01/08/2023] (T) Add satellite DMSP 5D-3 F16 (USA 172)
[14:43:33 - 01/08/2023] (T) Add satellite DMSP 5D-3 F17 (USA 191)
[14:43:33 - 01/08/2023] (T) Add satellite DMSP 5D-3 F18 (USA 210)
[14:43:33 - 01/08/2023] (T) Add satellite NOAA 15
[14:43:33 - 01/08/2023] (T) Add satellite NOAA 18
[14:43:33 - 01/08/2023] (T) Add satellite NOAA 19
[14:43:33 - 01/08/2023] (T) Add satellite FENGYUN 3A
[14:43:33 - 01/08/2023] (T) Add satellite FENGYUN 3B
[14:43:33 - 01/08/2023] (T) Add satellite FENGYUN 3C
[14:43:33 - 01/08/2023] (T) Add satellite FENGYUN 3D

IZ5RZR, 2023-08-01:
Confirm other test win another core I7 and windows 11, GUI close down after add satellite
[14:43:33 - 01/08/2023] (T) Add satellite FENGYUN 3D

IZ5RZR, 2023-08-01:
HERE IS THE PROBLEM, NOT ALL TLE ARE DOWNLOADING
METOP-A                 
1 29499U 06044A   23212.77596149  .00002968  00000+0  38995-3 0  9996
2 29499  98.3980 274.4651 0194203 162.2251 313.0156 14.68734210873574
METOP-B                 
1 38771U 12049A   23213.42662941  .00000163  00000+0  94437-4 0  9997
2 38771  98.6759 272.0389 0001023 100.9036 319.4594 14.21495947563979
METOP-C                 
1 43689U 18087A   23213.45165807  .00000157  00000+0  91746-4 0  9994
2 43689  98.6804 272.2232 0001625 305.5734  54.5292 14.21513365245557
DMSP 5D-3 F16 (USA 172) 
1 28054U 03048A   23213.09746872  .00000120  00000+0  87163-4 0  9999
2 28054  99.0133 211.6950 0006308 235.6535 212.5927 14.13828629 20909
DMSP 5D-3 F17 (USA 191) 
1 29522U 06050A   23213.08691408  .00000159  00000+0  10568-3 0  9995
2 29522  98.7533 225.1931 0010124 168.4952 191.6454 14.14269133863802
DMSP 5D-3 F18 (USA 210) 
1 35951U 09057A   23213.07225165  .00000355  00000+0  21035-3 0  9999
2 35951  98.7701 188.5673 0011329   7.4051 352.7287 14.13812916710983
<HEAD><TITLE>Connection timed out</TITLE></HEAD>
<BODY BGCOLOR="white" FGCOLOR="black">
<FONT FACE="Helvetica,Arial"><B>
 Connection timed out</B></FONT>

</BODY>
 NOAA 18                 
1 28654U 05018A   23213.38622599  .00000218  00000+0  14059-3 0  9996
2 28654  98.9082 287.5554 0013290 250.1504 109.8236 14.12979093937958
NOAA 19                 
1 33591U 09005A   23213.08236294  .00000245  00000+0  15637-3 0  9995
2 33591  99.0969 259.7201 0014866 115.4222 244.8489 14.12795977746281
FENGYUN 3A              
1 32958U 08026A   23212.84913408  .00000165  00000+0  99387-4 0  9991
2 32958  98.5573 165.1042 0007555 282.7789  77.2543 14.18895941786391
FENGYUN 3B              
1 37214U 10059A   23213.12673796  .00000119  00000+0  86065-4 0  9993
2 37214  99.0831 231.2543 0021146 279.9653 160.2148 14.14360767658868
FENGYUN 3C              
1 39260U 13052A   23213.11126411  .00000134  00000+0  88557-4 0  9993
2 39260  98.4282 224.8412 0011314 167.3104 192.8358 14.16059263509883
FENGYUN 3D              
1 43010U 17072A   23213.11548309  .00000129  00000+0  83161-4 0  9991
2 43010  98.8635 161.7597 0001880 136.9259 223.2064 14.19187451295878
FENGYUN 3E              
1 49008U 21062A   23213.11780982  .00000035  00000+0  37347-4 0  9994
2 49008  98.7229 213.8760 0000833 187.2514 172.8650 14.19689917107438
FENGYUN 3G              
1 56232U 23055A   23212.99041479  .00033241  00000+0  50990-3 0  9994
2 56232  49.9990   6.4992 0008819 328.6340  31.4101 15.53904233 16611
METEOR-M 1              
1 35865U 09049A   23212.93183200  .00000177  00000+0  98639-4 0  9998
2 35865  98.5831 176.5412 0003398 106.1985 253.9568 14.22523325719893
METEOR-M 2              
1 40069U 14037A   23213.13039801  .00000105  00000+0  67799-4 0  9997
2 40069  98.4243 217.0184 0005121 199.9137 160.1842 14.20798846470037
METEOR-M2 2             
1 44387U 19038A   23213.07699444  .00000055  00000+0  43397-4 0  9990
2 44387  98.7786 176.8973 0001839 130.1520 229.9820 14.23795343211699
<HEAD><TITLE>Connection timed out</TITLE></HEAD>
<BODY BGCOLOR="white" FGCOLOR="black">
<FONT FACE="Helvetica,Arial"><B>
 Connection timed out</B></FONT>

</BODY>
 TERRA                   
1 25994U 99068A   23213.10553189  .00000604  00000+0  13754-3 0  9996
2 25994  98.0922 280.3772 0000570 318.1259  99.9924 14.59252655256336
AQUA                    
1 27424U 02022A   23213.09996698  .00001158  00000+0  26101-3 0  9999
2 27424  98.2884 157.3661 0002020  65.7332  62.0858 14.58204705129905
[... 212 lines cut ...]
2 43477  88.9905 343.1959 0011958 218.8595 141.1795 15.26436841288693
CRYOSAT 2               
1 36508U 10013A   23212.70675769  .00000361  00000+0  88820-4 0  9993
2 36508  92.0204  23.1270 0001926 330.5186  29.5913 14.51907893705558
PRISMA                  
1 44072U 19015A   23213.12688756  .00001254  00000+0  15835-3 0  9993
2 44072  97.8704 286.9629 0001399  98.2069 261.9304 14.83644865236209
LANDSAT 8               
1 39084U 13008A   23213.10672652  .00000517  00000+0  12491-3 0  9998
2 39084  98.2243 282.4867 0001423  92.0344 268.1017 14.57104823556650
LANDSAT 9               
1 49260U 21088A   23213.14096347  .00000546  00000+0  13131-3 0  9991
2 49260  98.2278 282.4900 0001260 107.8853 252.2483 14.57097872 97939
YUNHAI 1                
1 41857U 16068A   23213.07850166  .00000445  00000+0  17843-3 0  9998
2 41857  98.5901 216.7062 0006289 221.4139 138.6569 14.33028228351291
YUNHAI 1-02             
1 44547U 19063A   23212.78142503  .00000372  00000+0  15155-3 0  9994
2 44547  98.4683 234.5787 0002688  73.6872 286.4608 14.32982619201290
<HEAD><TITLE>Connection timed out</TITLE></HEAD>
<BODY BGCOLOR="white" FGCOLOR="black">
<FONT FACE="Helvetica,Arial"><B>
 Connection timed out</B></FONT>

</BODY>

Aang23, 2023-08-01:
Can you please send it on paste.altillimity.com or the file itself instead?

Aang23, 2023-08-01:
With some luck https://github.com/SatDump/SatDump/actions/runs/5730238333/job/15528612976 will have it fixed....

Aang23, 2023-08-03:
Can you show the logs of the download error? I doubt it's on my end. But could be the network or such on yours!

I'll close this for now.

closed 2023-08-03 by Aang23