SatDump/SatDump #325: Installing on MacOS

SquashyDisco, 2023-07-26:
Hello,

I'm struggling to compile SatDump on Mac OS 13 Ventura running on the M1 Chipset. I've tried several attempts, but I keep coming to the same issue:

When I reach the _make -j4'nproc'_ stage of compiling, I keep having these errors:

[ 0%] Linking CXX shared library ../libsatdump_core.dylib
ld: warning: directory not found for option '-L/usr/local/opt/libomp/lib'
ld: library not found for -lvolk
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libsatdump_core.dylib] Error 1
make[1]: *** [src-core/CMakeFiles/satdump_core.dir/all] Error 2
make: *** [all] Error 2

My coding experience is fairly limited and I can't see a way forward. I've tried to install Libomp through Ports and Homebrew but I'm not getting anywhere with it. Volk is another issue that I can't find a solution. 

I've managed to compile it on a Raspberry Pi, but I'd really want it on my Macbook Air.

Are you able to help?

Aang23, 2023-08-01:
How did you install volk?

JVital2013, 2023-09-04:
@SquashyDisco while I can re-iterate @Aang23's question, you can also try the new nightly builds for macOS. They're not optimized for M1/M2 macs yet, and compiling it yourself will result in faster processing - but if building does not work for you, the pre-compiled releases may be worth a shot.

jpjonte, 2023-09-17:
On Apple Silicon macs brew changed the prefix from `/usr/local` to `/opt/homebrew`. If you replace all occurrences of `/usr/local` to `/opt/homebrew` in `CMakeLists.txt` it should compile.

JVital2013, 2023-10-30:
This issue has been idle for a while now and cannot be reproduced, so I'm going to close it. If you would still like help, feel free to re-open it. Thanks!

closed 2023-10-30 by JVital2013