SatDump/SatDump #330: abort() in Tracker widget

gvanem, 2023-08-01:
Playing with the *Tracker widget*, selecting `Connect` (to a joke localhost address)
and pressing `Disconnect`, it throws inside `net::TCPClient::recvs()`.

I assume that is since the socket has been closed.

Call-stack:
```
  ucrtbase!abort+0x4e
  ucrtbase!terminate+0x1f
  VCRUNTIME140_1!FindHandler<__FrameHandler4>(struct EHExceptionRecord * pExcept = 0x000000e7`463cf7e0, 
  unsigned int64 * pRN = 0x000000e7`463ce890, struct _CONTEXT * pContext = 0x000000e7`463cf010, struct _xDISPATCHER_CONTEXT 
  * pDC = 0x000000e7`463cee50, struct FH4::FuncInfo4 * pFuncInfo = 0x000000e7`463ce860, unsigned char recursive = 0x00 '', 
  int CatchDepth = 0n0, unsigned int64 * pMarkerRN = 0x00000000`00000000)+0x45b
  VCRUNTIME140_1!__InternalCxxFrameHandler<__FrameHandler4>(struct EHExceptionRecord * pExcept = 0x000000e7`463cf7e0, 
  unsigned int64 * pRN = 0x000000e7`463ce890, struct _CONTEXT * pContext = 0x000000e7`463cf010, struct _xDISPATCHER_CONTEXT 
  * pDC = 0x000000e7`463cee50, struct FH4::FuncInfo4 * pFuncInfo = 0x000000e7`463ce860, int CatchDepth = 0n0, 
  *   unsigned int64 * pMarkerRN = 0x00000000`00000000, unsigned char recursive = 0x00 '')+0x267
  VCRUNTIME140_1!__InternalCxxFrameHandlerWrapper<__FrameHandler4>(struct EHExceptionRecord * pExcept = 
  0x00007ff6`817e0000, unsigned int64 * pRN = 0x000000e7`463cf7e0, struct _CONTEXT * pContext = 0x000000e7`463cf010, 
  struct _xDISPATCHER_CONTEXT * pDC = 0x000000e7`463cee50, struct FH4::FuncInfo4 * pFuncInfo = 0x000000e7`463ce860, 
  int CatchDepth = 0n0, unsigned int64 * pMarkerRN = 0x00000000`00000000, unsigned char recursive = 0x00 '')+0x3e
  VCRUNTIME140_1!__CxxFrameHandler4(struct EHExceptionRecord * pExcept = 0x000000e7`463cf7e0, unsigned int64 RN =
  <Value unavailable error>, struct _CONTEXT * pContext = 0x000000e7`463cf010, struct _xDISPATCHER_CONTEXT * 
  pDC = 0x000000e7`463cee50)+0xa9
  ntdll!RtlpExecuteHandlerForException+0xf
  ntdll!RtlDispatchException+0x244
  ntdll!RtlRaiseException+0x1d7
  KERNELBASE!RaiseException+0x69
  VCRUNTIME140!_CxxThrowException(void * pExceptionObject = 0x000000e7`463cf938, struct _s__ThrowInfo * pThrowInfo = 
  <Value unavailable error>)+0x90
  satdump_ui!net::TCPClient::recvs(void)+0x1f1
  satdump_ui!RotctlHandler::command(class std::basic_string<char,std::char_traits<char>,std::allocator<char> > 
  * cmd = 0x000000e7`463cf9c0 "p ", int * ret_sz = 0x000000e7`463cf9b0)+0x293
  satdump_ui!RotctlHandler::get_pos(float * az = 0x0000025b`2d2908b4, float * el = 0x0000025b`2d2908b8)+0x84
  satdump_ui!satdump::TrackingWidget::rotatorth_run(void)+0xc3
...
```

This is how I patched and fixed it:
```diff
--- a/src-interface/recorder/tracking/rotcl_handler.h 2023-08-01 12:04:21
+++ b/src-interface/recorder/tracking/rotcl_handler.h 2023-08-01 13:37:49
@@ -24,7 +24,16 @@
         std::string result;
         result.resize(1000);

+        try
+        {
         *ret_sz = client->recvs((uint8_t *)result.data(), result.size());
+        }
+        catch (std::exception &e)
+        {
+            logger->error(e.what());
+            disconnect();
+            return "";
+        }

         if (*ret_sz < 0)
             return "";
```

So now it just prints:
```
[11:39:41 - 01/08/2023] (T) Update pass trajectory...
[11:39:55 - 01/08/2023] (E) Error receiving from TCP socket!
[11:39:55 - 01/08/2023] (E) Error getting rotator position!
```

instead of `abort()`-ing.

BTW, there are these important warnings on wrong use of `sscanf()`:
```c
tracking_widget_backend.cpp(304): warning C4473: 'sscanf' : not enough arguments passed for format string
tracking_widget_backend.cpp(304): note: placeholders and their parameters expect 6 variadic arguments, but 5 were provided
tracking_widget_backend.cpp(304): note: the missing variadic argument 6 is required by format string '%lf'
tracking_widget_backend.cpp(306): warning C4473: 'sscanf' : not enough arguments passed for format string
tracking_widget_backend.cpp(306): note: placeholders and their parameters expect 6 variadic arguments, but 5 were provided
tracking_widget_backend.cpp(306): note: the missing variadic argument 6 is required by format string '%lf'
```

I'm on Win-10 using RTLSDR and/or file as input.
MSVC compiled from today's `git master`.

closed 2023-08-09 by ZbychuButItWasTaken

Aang23, 2023-08-09:
Note : I will need to do more testing for this!

gvanem, 2023-09-13:
> Note : I will need to do more testing for this!

The `sscanf()` code is still wrong for MSVC.

commit 7e360413 (2023-08-09): fix #330