SatDump/SatDump #335: Hi, is it possible to have a debian package for raspberry? Thanks

pade1036, 2023-08-09:
**What would you want to be added to satdump?**
A clear and concise description of what the idea/problem is.

**Describe the feature you'd like to have**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Here you can for provide more information about what you'd like to see as the end result. Feel free to also post images if you have ideas regarding the UI

Aang23, 2023-09-03:
It is, but at the moment Github does not offer ARM workers, making it complicated to do cleanly. It will happen when possible, or I put more work into making cross-compilation function (it works fine, but dpkg-sheplib doesn't want to work...).

JVital2013, 2024-02-04:
Hi @pade1036, we now have a package available for arm64, which will work on all Raspberry Pi's, model 3 and newer - assuming you have the modern 64-bit OS installed. Will this meet your needs?

alexmyczko, 2024-02-12:
Yes, https://buildd.debian.org/status/package.php?p=satdump

JVital2013, 2024-02-12:
Good point! With the addition of Debian builds, several more platforms are covered as well, so this issue appears addressed.

closed 2024-02-12 by JVital2013