SatDump/SatDump #355: satdump-ui crashes when "Loading plugin /usr/lib/satdump//plugins/libdvb_support.so..."

gvimlag, 2023-08-19:
**Description of the issue**
  "satdump-ui" crashes when "Loading plugin /usr/lib/satdump//plugins/libdvb_support.so..."
**Hardware (SDR/PC/OS)**
  Hardware: x86 64-bit
  OS:       Debian 11 (bullseye) 64-bit, KDE
  Dongle:   RTL-SDR
**Version (Eg, 1.0.0, CI Build #171)**
  SatDump:  version 1.0.3
**Logs after the cras (satdump.logs)**
TERMINAL CRASH TEXT:
215:~/SDR/SatDump$ satdump-ui 
[06:58:03 - 19/08/2023] (E) Usage : satdump-ui [downlink] [input_level] [input_file] [output_file_or_directory] [additional options as required]
[06:58:03 - 19/08/2023] (E) Extra options (examples. Any parameter used in modules can be used here) :
[06:58:03 - 19/08/2023] (E)  --samplerate [baseband_samplerate] --baseband_format [f32/i16/i8/w8] --dc_block --iq_swap
[06:58:04 - 19/08/2023] (D) Starting with OpenGL 3.3 (Core Profile) Mesa 20.3.5
[06:58:04 - 19/08/2023] (I)    _____       __  ____                      
[06:58:04 - 19/08/2023] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ 
[06:58:04 - 19/08/2023] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[06:58:04 - 19/08/2023] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[06:58:04 - 19/08/2023] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ 
[06:58:04 - 19/08/2023] (I)                                     /_/      
[06:58:04 - 19/08/2023] (I) Starting SatDump v1.1.0-3b6399cb
[06:58:04 - 19/08/2023] (I) 
[06:58:04 - 19/08/2023] (I) Loading config /usr/share/satdump//satdump_cfg.json
[06:58:04 - 19/08/2023] (W) No user configuration found! Keeping defaults.
[06:58:04 - 19/08/2023] (I) Loading plugins from /usr/lib/satdump//plugins
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/liblandsat_support.so...
[06:58:04 - 19/08/2023] (T) Plugin landsat_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libspacex_support.so...
[06:58:04 - 19/08/2023] (T) Plugin spacex_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libothers_support.so...
[06:58:04 - 19/08/2023] (T) Plugin others_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libsimd_sse41.so...
[06:58:04 - 19/08/2023] (E) CPU Does not support SSE4_1. Extension plugin NOT loading!
[06:58:04 - 19/08/2023] (T) Plugin simd_sse41 loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libstereo_support.so...
[06:58:04 - 19/08/2023] (T) Plugin stereo_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libhackrf_sdr_support.so...
[06:58:04 - 19/08/2023] (T) Plugin hackrf_sdr_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libairspyhf_sdr_support.so...
[06:58:04 - 19/08/2023] (T) Plugin airspyhf_sdr_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libjason3_support.so...
[06:58:04 - 19/08/2023] (T) Plugin jason3_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libmeteor_support.so...
[06:58:04 - 19/08/2023] (T) Plugin meteor_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libgoes_support.so...
[06:58:04 - 19/08/2023] (T) Plugin goes_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/librtltcp_support.so...
[06:58:04 - 19/08/2023] (T) Plugin rtltcp_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libumka_support.so...
[06:58:04 - 19/08/2023] (T) Plugin umka_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libfengyun2_support.so...
[06:58:04 - 19/08/2023] (T) Plugin fengyun2_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libbluewalker3_support.so...
[06:58:04 - 19/08/2023] (T) Plugin bluewalker3_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libnoaa_metop_support.so...
[06:58:04 - 19/08/2023] (T) Plugin noaa_metop_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libfengyun3_support.so...
[06:58:04 - 19/08/2023] (T) Plugin fengyun3_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libsimd_avx2.so...
[06:58:04 - 19/08/2023] (E) CPU Does not support AVX2. Extension plugin NOT loading!
[06:58:04 - 19/08/2023] (T) Plugin simd_avx2 loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libjpss_support.so...
[06:58:04 - 19/08/2023] (T) Plugin jpss_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libproba_support.so...
[06:58:04 - 19/08/2023] (T) Plugin proba_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libgk2a_support.so...
[06:58:04 - 19/08/2023] (T) Plugin gk2a_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libspyserver_support.so...
[06:58:04 - 19/08/2023] (T) Plugin spyserver_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libtubsat_support.so...
[06:58:04 - 19/08/2023] (T) Plugin tubsat_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libelektro_arktika_support.so...
[06:58:04 - 19/08/2023] (T) Plugin elektro_arktika_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libfengyun4_support.so...
[06:58:04 - 19/08/2023] (T) Plugin fenyun4_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libgcom_support.so...
[06:58:04 - 19/08/2023] (T) Plugin gcom_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libmirisdr_sdr_support.so...
[06:58:04 - 19/08/2023] (T) Plugin mirisdr_sdr_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libcubesat_support.so...
[06:58:04 - 19/08/2023] (T) Plugin cubesat_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libanalog_support.so...
[06:58:04 - 19/08/2023] (T) Plugin analog_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libsdrpp_server_support.so...
[06:58:04 - 19/08/2023] (T) Plugin spyserver_support loaded!
[06:58:04 - 19/08/2023] (T) Loading plugin /usr/lib/satdump//plugins/libdvb_support.so...
Illegal instruction
**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

Aang23, 2023-08-19:
Use a newer version where this has been fixed.

gvimlag, 2023-08-19:
This version is the release version 1.0.3.
Where do I get a newer version?

Aang23, 2023-08-19:
Ah, I see what's going on.

Shouldn't happen building from source. But I will need to look and fix. That's one issue with providing builds, you just be lacking AVX2.

gvimlag, 2023-08-19:
Thank you.
Please notify me when the fix is in-place.
Again, thank you.

Aang23, 2023-09-14:
What's your exact CPU?

gvimlag, 2023-09-14:
"What's your exact CPU?"
AMD Phenom II X6 1045T

gvimlag, 2023-09-26:
I tested the nightly build (see: github.com/SatDump/SatDump/issues/422) on an "AMD Phenom II X6 1045T", but "SatDump-ui" still crashes on startup due to the X86 CPU lacking AVX2.

I discovered that there are "cmake" options which disable AVX2 on macOS (shown in the macOS build instructions).
So I tried applying them to the Debian 11 X86 build:
`   cmake -DENABLE_AVX=false -DENABLE_AVX2=false ...`
but they also failed.

I understand that you are busy.  I was just wondering when you might address this AVX2 issue?

Thank you.

mrswooz, 2023-10-19:
If you are not interested in DVB support the crash can be fixed by commenting out the following lines in **satdump/plugins/main.cpp** and then rebuild using **sudo make install** from the **satdump/build** folder

#include "core/plugin.h"
#include "logger.h"
#include "core/module.h"

#include "dvbs/module_dvbs_demod.h"
#include "dvbs2/module_dvbs2_demod.h"
#include "dvbs2/module_s2_ts2tcp.h"

class DVBSupport : public satdump::Plugin
{
public:
    std::string getID()
    {
        return "dvb_support";
    }

    void init()
    {
        **//satdump::eventBus->register_handler<RegisterModulesEvent>(registerPluginsHandler);**
    }

    static void registerPluginsHandler(const RegisterModulesEvent &evt)
    {
        //REGISTER_MODULE_EXTERNAL(evt.modules_registry, dvb::DVBSDemodModule);
        //REGISTER_MODULE_EXTERNAL(evt.modules_registry, dvb::DVBS2DemodModule);
        //REGISTER_MODULE_EXTERNAL(evt.modules_registry, dvbs2::S2TStoTCPModule);
    }
};

PLUGIN_LOADER(DVBSupport)

mrswooz, 2023-10-19:
Sorry editor playing up. code should be:
#include "core/plugin.h"
#include "logger.h"
#include "core/module.h"

#include "dvbs/module_dvbs_demod.h"
#include "dvbs2/module_dvbs2_demod.h"
#include "dvbs2/module_s2_ts2tcp.h"

class DVBSupport : public satdump::Plugin
{
public:
    std::string getID()
    {
        return "dvb_support";
    }

    void init()
    {
        //satdump::eventBus->register_handler<RegisterModulesEvent>(registerPluginsHandler);
    }

    static void registerPluginsHandler(const RegisterModulesEvent &evt)
    {
        //REGISTER_MODULE_EXTERNAL(evt.modules_registry, dvb::DVBSDemodModule);
        //REGISTER_MODULE_EXTERNAL(evt.modules_registry, dvb::DVBS2DemodModule);
        //REGISTER_MODULE_EXTERNAL(evt.modules_registry, dvbs2::S2TStoTCPModule);
    }
};

PLUGIN_LOADER(DVBSupport)

gvimlag, 2023-10-19:
Hello swoozeman,
Thank you for the suggestion.
Unfortunately, "satdump-ui" still crashes.

Below is what I did, and the results.

SPECIFICS
As previously instructed, I download from the nightly build: "curl -L https://github.com/SatDump/SatDump/archive/refs/heads/master.tar.gz -o SatDump-master.tar.gz"
I uncompress this download into "/opt/SatDump" which creates "/opt/SatDump/SatDump-master/", then build/compile it.

The file you indicated ("MY_PATH/satdump/plugins/main.cpp") does not exist.
But "MY_PATH/SatDump-master/plugins/dvb_support/main.cpp" does exist and its contents matched what you showed.
So in my existing build, I commented out the contents of "MY_PATH/SatDump-master/plugins/dvb_support/main.cpp" as you instructed.

Here are the results of each step.

--------------------------------------------------------------------------------

COMMENT OUT THE FILE CONTENTS:
$ sudo vi /opt/SatDump/SatDump-master/plugins/dvb_support/main.cpp
```
/*
#include "core/plugin.h"
#include "logger.h"
#include "core/module.h"

#include "dvbs/module_dvbs_demod.h"
#include "dvbs2/module_dvbs2_demod.h"
#include "dvbs2/module_s2_ts2tcp.h"

class DVBSupport : public satdump::Plugin
{
public:
    std::string getID()
    {
        return "dvb_support";
    }

    void init()
    {
        satdump::eventBus->register_handler<RegisterModulesEvent>(registerPluginsHandler);
    }

    static void registerPluginsHandler(const RegisterModulesEvent &evt)
    {
        REGISTER_MODULE_EXTERNAL(evt.modules_registry, dvb::DVBSDemodModule);
        REGISTER_MODULE_EXTERNAL(evt.modules_registry, dvb::DVBS2DemodModule);
        REGISTER_MODULE_EXTERNAL(evt.modules_registry, dvbs2::S2TStoTCPModule);
    }
};

PLUGIN_LOADER(DVBSupport)
*/
```

--------------------------------------------------------------------------------

REBUILD:
$ cd /opt/SatDump/SatDump-master/build
$ sudo make install
```
[ 38%] Built target satdump_core
[ 38%] Built target satdump
[ 40%] Built target satdump_interface
[ 41%] Built target satdump-ui
[ 42%] Built target spacex_support
[ 43%] Built target fengyun2_support
[ 45%] Built target fengyun3_support
[ 46%] Built target fengyun4_support
[ 47%] Built target proba_support
[... 903 lines cut ...]
[23:00:22 - 19/10/2023] (T) Loading plugin /usr/lib/satdump//plugins/libelektro_arktika_support.so...
[23:00:22 - 19/10/2023] (T) Plugin elektro_arktika_support loaded!
[23:00:22 - 19/10/2023] (T) Loading plugin /usr/lib/satdump//plugins/libfengyun4_support.so...
[23:00:22 - 19/10/2023] (T) Plugin fenyun4_support loaded!
[23:00:22 - 19/10/2023] (T) Loading plugin /usr/lib/satdump//plugins/libgcom_support.so...
[23:00:22 - 19/10/2023] (T) Plugin gcom_support loaded!
[23:00:22 - 19/10/2023] (T) Loading plugin /usr/lib/satdump//plugins/libmirisdr_sdr_support.so...
[23:00:22 - 19/10/2023] (T) Plugin mirisdr_sdr_support loaded!
[23:00:22 - 19/10/2023] (T) Loading plugin /usr/lib/satdump//plugins/libcubesat_support.so...
[23:00:22 - 19/10/2023] (T) Plugin cubesat_support loaded!
[23:00:22 - 19/10/2023] (T) Loading plugin /usr/lib/satdump//plugins/libanalog_support.so...
[23:00:22 - 19/10/2023] (T) Plugin analog_support loaded!
[23:00:22 - 19/10/2023] (T) Loading plugin /usr/lib/satdump//plugins/libsdrpp_server_support.so...
[23:00:22 - 19/10/2023] (T) Plugin spyserver_support loaded!
[23:00:22 - 19/10/2023] (T) Loading plugin /usr/lib/satdump//plugins/liblimesdr_sdr_support.so...
[23:00:22 - 19/10/2023] (T) Plugin limesdr_sdr_support loaded!
[23:00:22 - 19/10/2023] (T) Loading plugin /usr/lib/satdump//plugins/libdvb_support.so...
[23:00:22 - 19/10/2023] (W) Possible error loading symbols from plugin!
Segmentation fault

```

So "satdump-up" still crashes.
If I'm doing something wrong, please let me know.
Thank you for the help.

swoozeman, 2023-10-20:
Hello gvimlab
It will crash with the whole of main.cpp commented out. Just comment the lines I suggested and try again.

gvimlag, 2023-10-20:
Hello swoozeman,
I misunderstood your original post.
Thank you for pointing out my mistake.

Unfortunately, when I comment out just the section you indicated (not the whole file),
the rebuild fails.

-----------------------------------------

COMMENT FILE CONTENTS:
$ sudo vi /opt/SatDump/SatDump-master/plugins/dvb_support/main.cpp
```
#include "core/plugin.h"
#include "logger.h"
#include "core/module.h"

#include "dvbs/module_dvbs_demod.h"
#include "dvbs2/module_dvbs2_demod.h"
#include "dvbs2/module_s2_ts2tcp.h"

class DVBSupport : public satdump::Plugin
{
public:
    std::string getID()
    {
        return "dvb_support";
    }

//  void init()
//  {
//      satdump::eventBus->register_handler<RegisterModulesEvent>(registerPluginsHandler);
//  }
//
//  static void registerPluginsHandler(const RegisterModulesEvent &evt)
//  {
//      REGISTER_MODULE_EXTERNAL(evt.modules_registry, dvb::DVBSDemodModule);
//      REGISTER_MODULE_EXTERNAL(evt.modules_registry, dvb::DVBS2DemodModule);
//      REGISTER_MODULE_EXTERNAL(evt.modules_registry, dvbs2::S2TStoTCPModule);
//  }

};

PLUGIN_LOADER(DVBSupport)
```

-----------------------------------------

REBUILD:
$ cd /opt/SatDump/SatDump-master/build
$ sudo make install
```
[ 38%] Built target satdump_core
[ 38%] Built target satdump
[ 40%] Built target satdump_interface
[ 41%] Built target satdump-ui
[ 42%] Built target spacex_support
[ 43%] Built target fengyun2_support
[ 45%] Built target fengyun3_support
[ 46%] Built target fengyun4_support
[ 47%] Built target proba_support
[ 48%] Built target eos_support
[ 50%] Built target meteor_support
[ 50%] Built target jpss_support
[ 53%] Built target goes_support
[ 55%] Built target elektro_arktika_support
[ 56%] Built target gk2a_support
[ 59%] Built target noaa_metop_support
[ 60%] Built target dmsp_support
[ 61%] Built target jason3_support
[ 61%] Built target aim_support
[ 61%] Built target mats_support
[ 61%] Built target oceansat_support
[ 62%] Built target landsat_support
[ 63%] Built target himawari_support
[ 63%] Built target geonetcast_support
[ 63%] Built target bluewalker3_support
[ 64%] Built target analog_support
[ 65%] Built target orbcomm_support
[ 65%] Built target gcom_support
[ 65%] Built target tubsat_support
[ 66%] Built target umka_support
[ 66%] Built target dscovr_support
[ 67%] Built target stereo_support
[ 93%] Built target inmarsat_support
[ 94%] Built target cubesat_support
[ 95%] Built target others_support
Scanning dependencies of target dvb_support
[ 96%] Building CXX object plugins/dvb_support/CMakeFiles/dvb_support.dir/main.cpp.o
In file included from /opt/SatDump/SatDump-master/plugins/dvb_support/main.cpp:1:
/opt/SatDump/SatDump-master/plugins/dvb_support/main.cpp: In function ‘satdump::Plugin* loader()’:
/opt/SatDump/SatDump-master/plugins/../src-core/core/plugin.h:13:55: error: invalid new-expression of abstract class type ‘DVBSupport’
   13 |             return (satdump::Plugin *)new constructor(); \
      |                                                       ^
/opt/SatDump/SatDump-master/plugins/dvb_support/main.cpp:31:1: note: in expansion of macro ‘PLUGIN_LOADER’
   31 | PLUGIN_LOADER(DVBSupport)
      | ^~~~~~~~~~~~~
/opt/SatDump/SatDump-master/plugins/dvb_support/main.cpp:9:7: note:   because the following virtual functions are pure within ‘DVBSupport’:
    9 | class DVBSupport : public satdump::Plugin
      |       ^~~~~~~~~~
In file included from /opt/SatDump/SatDump-master/plugins/dvb_support/main.cpp:1:
/opt/SatDump/SatDump-master/plugins/../src-core/core/plugin.h:28:22: note:     ‘virtual void satdump::Plugin::init()’
   28 |         virtual void init() = 0;
      |                      ^~~~
make[2]: *** [plugins/dvb_support/CMakeFiles/dvb_support.dir/build.make:303: plugins/dvb_support/CMakeFiles/dvb_support.dir/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2262: plugins/dvb_support/CMakeFiles/dvb_support.dir/all] Error 2
make: *** [Makefile:171: all] Error 2
```

Sorry.  I'm not a c++ programmer.  (I'm a hardware guy, not a software guy.)
What am I doing wrong?

Thank you for the help.

swoozeman, 2023-10-20:
You have commented out more lines than I show. You should only have four lines commented out. Do it EXACTLY as shown below:

void init()
{
    //satdump::eventBus->register_handler<RegisterModulesEvent>(registerPluginsHandler);
}

static void registerPluginsHandler(const RegisterModulesEvent &evt)
{
    //REGISTER_MODULE_EXTERNAL(evt.modules_registry, dvb::DVBSDemodModule);
    //REGISTER_MODULE_EXTERNAL(evt.modules_registry, dvb::DVBS2DemodModule);
    //REGISTER_MODULE_EXTERNAL(evt.modules_registry, dvbs2::S2TStoTCPModule);
}

gvimlag, 2023-10-20:
Hello swoozeman,
Following your instructions, the build succeeded and "satdump-ui" executed without crashing!
Thank you for your help and understanding.
I really appreciate it!

JVital2013, 2024-09-08:
It's a long time coming, but this should be fixed as of the latest commit. It will automatically DVB entirely on machines without SSE 4.1 in the stock builds, rather than crashing.

You can manually build a copy of SatDump with working (but very slow) DVB demodulation for non-SSE 4.1 processors by using the following cmake command:

```
cmake -DHAVE_SSE41=OFF -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr ..
```

closed 2024-09-08 by JVital2013

creas002, 2024-09-08:
Thank you Jamie!

commit d83ddaac (2024-09-08): Fix #355