SatDump/SatDump #356: satdump build fails for Raspberry Pi 3B

gvimlag, 2023-08-19:
**Description of the issue**
satdump build FAILS for Raspberry Pi 3B (1GB RAM).
  [NOTE: satdump build SUCCEEDS for a Raspberry Pi 4B (8GB RAM)]

**Hardware (SDR/PC/OS)**
Hardware: Raspberry Pi 3B (1GB RAM)
OS:       Raspberry Pi OS (64-bit) (Debian Bullseye)
Dongle:   RTL-SDR
**Version (Eg, 1.0.0, CI Build #171)**
SatDump:  version 1.0.3
**Logs after the cras (satdump.logs)**
...
[  9%] Building CXX object src-core/CMakeFiles/satdump_core.dir/common/geodetic/euler_coordinates.cpp.o
[  9%] Building CXX object src-core/CMakeFiles/satdump_core.dir/common/geodetic/euler_raytrace.cpp.o
[  9%] Building CXX object src-core/CMakeFiles/satdump_core.dir/common/geodetic/geodetic_coordinates.cpp.o
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [src-core/CMakeFiles/satdump_core.dir/build.make:1135: src-core/CMakeFiles/satdump_core.dir/common/dsp_source_sink/dsp_sample_sink.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:1230: src-core/CMakeFiles/satdump_core.dir/all] Error 2
make: *** [Makefile:171: all] Error 2
...
**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

JVital2013, 2023-08-21:
It looks like the build is failing because it's running out of memory.

When running `make`, try running just `make` and not `make -j4`. This will make it build slower, but it will use less memory

Aang23, 2023-08-22:
100% a RAM issue. @JVital2013's solution should work just fine.

closed 2023-08-22 by Aang23

gvimlag, 2023-08-28:
Increasing the SWAP from 0.1GB to 2GB solved the problem.