SatDump/SatDump #357: Building on macOS without AVX2

Crosswalkersam, 2023-08-20:
When building SatDump on a macOS machine that does not feature AVX2, it is still build with AVX2.
Therefore, resulting builds cannot be run and always terminate on `illegal instruction: 4`.
Tried building on macOS 12.0.1 Monterey, Intel Xeon E5-1680v2 with the latest master.

Crash report:
```
Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 4 Illegal instruction: 4
Terminating Process:   exc handler [9459]

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libfftw3f.3.dylib             	       0x1034b90a8 fftwf_mkplanner + 184
1   libfftw3f.3.dylib             	       0x103484727 fftwf_the_planner + 23
2   libfftw3f.3.dylib             	       0x10347e97d fftwf_mkapiplan + 45
3   libfftw3f.3.dylib             	       0x103484318 fftwf_plan_many_dft + 280
4   libfftw3f.3.dylib             	       0x103483154 fftwf_plan_dft + 52
5   libfftw3f.3.dylib             	       0x103482e25 fftwf_plan_dft_1d + 37
6   libsatdump_core.dylib         	       0x102be7801 dsp::FFTPanBlock::set_fft_settings(int, unsigned long long, int) + 561
7   satdump-ui                    	       0x1022d7ee7 satdump::RecorderApplication::RecorderApplication() + 4423
8   satdump-ui                    	       0x1022c97a1 satdump::initMainUI() + 673
9   satdump-ui                    	       0x1022c28f5 main + 3573
10  dyld                          	       0x1107d34fe start + 462
```

JVital2013, 2023-08-24:
@crosswalkersam @aang23 I just tested this on a 2012 iMac that does not have AVX2 and it compiled/ran fine.

**However,** when running the cmake step of compiling libfftw3, you need to specify `-DENABLE_AVX2=false`. Otherwise, the library compiles with AVX2.

Uninstall satdump and libfftw3, then rebuild libfftw3 using this cmake command:

```sh
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTS=false -DENABLE_FLOAT=true -DENABLE_THREADS=true -DENABLE_SSE=true -DENABLE_SSE2=true -DENABLE_AVX=true -DENABLE_AVX2=false ..
```

Afterwards, SatDump should build/run fine

Crosswalkersam, 2023-08-24:
Just verified, that solves it. Thanks a lot!

closed 2023-08-24 by Crosswalkersam