SatDump/SatDump #371: error: ‘GLFW_SCALE_TO_MONITOR’ was not declared in this scope

K4KDR, 2023-08-31:
Google searches did not provide me with any tips on what I might be missing, so I'll have to ask for help.

After going back through all the dependencies to be sure I had the latest of everything, when attempting to install the latest

(https://github.com/SatDump/SatDump/commit/fdbbac5a219967f28b4ed7e710e4c5b966b1906f as of the time of this post)

on Ubuntu 18.04, I get the following at the end of the 'make' command:
```
[ 99%] Linking CXX shared library ../../libportaudio_audio_sink.so
[ 99%] Built target portaudio_audio_sink
[ 99%] Building CXX object src-ui/CMakeFiles/satdump-ui.dir/imgui/imgui_image.cpp.o
[ 99%] Building CXX object src-ui/CMakeFiles/satdump-ui.dir/imgui/imgui_impl_glfw.cpp.o
[ 99%] Building CXX object src-ui/CMakeFiles/satdump-ui.dir/imgui/imgui_impl_opengl3.cpp.o
[ 99%] Building CXX object src-ui/CMakeFiles/satdump-ui.dir/loading_screen.cpp.o
[ 99%] Linking CXX shared library ../../libudp_source_support.so
[ 99%] Built target udp_source_support
[ 99%] Building CXX object src-ui/CMakeFiles/satdump-ui.dir/main.cpp.o
In file included from /home/k4kdr/satdump/src-ui/loading_screen.cpp:4:
/home/k4kdr/satdump/src-ui/loading_screen.h: In constructor ‘satdump::LoadingScreenSink::LoadingScreenSink(GLFWwindow*, float, GLFWimage*)’:
/home/k4kdr/satdump/src-ui/loading_screen.h:18:21: warning: ‘satdump::LoadingScreenSink::window’ will be initialized after [-Wreorder]
         GLFWwindow* window;
                     ^~~~~~
/home/k4kdr/satdump/src-ui/loading_screen.h:17:15: warning:   ‘float satdump::LoadingScreenSink::scale’ [-Wreorder]
         float scale;
               ^~~~~
/home/k4kdr/satdump/src-ui/loading_screen.cpp:9:5: warning:   when initialized here [-Wreorder]
     LoadingScreenSink::LoadingScreenSink(GLFWwindow* window, float scale, GLFWimage* img) : window{window},
     ^~~~~~~~~~~~~~~~~
/home/k4kdr/satdump/src-ui/main.cpp: In function ‘void window_content_scale_callback(GLFWwindow*, float, float)’:
/home/k4kdr/satdump/src-ui/main.cpp:25:48: warning: unused parameter ‘window’ [-Wunused-parameter]
 void window_content_scale_callback(GLFWwindow* window, float xscale, float yscale)
                                    ~~~~~~~~~~~~^~~~~~
/home/k4kdr/satdump/src-ui/main.cpp:25:76: warning: unused parameter ‘yscale’ [-Wunused-parameter]
 void window_content_scale_callback(GLFWwindow* window, float xscale, float yscale)
                                                                      ~~~~~~^~~~~~
/home/k4kdr/satdump/src-ui/main.cpp: In function ‘int main(int, char**)’:
/home/k4kdr/satdump/src-ui/main.cpp:80:20: error: ‘GLFW_SCALE_TO_MONITOR’ was not declared in this scope
     glfwWindowHint(GLFW_SCALE_TO_MONITOR, GLFW_TRUE);
                    ^~~~~~~~~~~~~~~~~~~~~
/home/k4kdr/satdump/src-ui/main.cpp:80:20: note: suggested alternative: ‘GLFW_KEY_LEFT_CONTROL’
     glfwWindowHint(GLFW_SCALE_TO_MONITOR, GLFW_TRUE);
                    ^~~~~~~~~~~~~~~~~~~~~
                    GLFW_KEY_LEFT_CONTROL
/home/k4kdr/satdump/src-ui/main.cpp:223:5: error: ‘glfwSetWindowContentScaleCallback’ was not declared in this scope
     glfwSetWindowContentScaleCallback(window, window_content_scale_callback);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/k4kdr/satdump/src-ui/main.cpp:223:5: note: suggested alternative: ‘glfwSetWindowFocusCallback’
     glfwSetWindowContentScaleCallback(window, window_content_scale_callback);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     glfwSetWindowFocusCallback
/home/k4kdr/satdump/src-ui/main.cpp:224:5: error: ‘glfwGetWindowContentScale’ was not declared in this scope
     glfwGetWindowContentScale(window, &display_scale, nullptr);
     ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/k4kdr/satdump/src-ui/main.cpp:224:5: note: suggested alternative: ‘glfwGetWindowMonitor’
     glfwGetWindowContentScale(window, &display_scale, nullptr);
     ^~~~~~~~~~~~~~~~~~~~~~~~~
     glfwGetWindowMonitor
src-ui/CMakeFiles/satdump-ui.dir/build.make:131: recipe for target 'src-ui/CMakeFiles/satdump-ui.dir/main.cpp.o' failed
make[2]: *** [src-ui/CMakeFiles/satdump-ui.dir/main.cpp.o] Error 1
CMakeFiles/Makefile2:1046: recipe for target 'src-ui/CMakeFiles/satdump-ui.dir/all' failed
make[1]: *** [src-ui/CMakeFiles/satdump-ui.dir/all] Error 2
Makefile:155: recipe for target 'all' failed
make: *** [all] Error 2
```

... and since the error mentioned '**GLFW**', I double-checked my install of that dependency:

```
k4kdr@3010i5:~/satdump/build$ apt-cache policy libglfw3-dev
libglfw3-dev:
  Installed: 3.2.1-1
  Candidate: 3.2.1-1
```

... any instruction on how I can correct this would be MUCH appreciated!

-Scott,  K4KDR

JVital2013, 2023-08-31:
Hmm, it looks like the new version requires glfw 3.3 or higher to compile. Version 3.2 is (apparently) missing some of the display scaling APIs. 

I'll see if I can patch it later to compile on 3.2

closed 2023-09-01 by Aang23

K4KDR, 2023-09-01:
Thanks a million! 

-Scott

PR #373 (2023-09-01): Fix GLFW 3.2 Support
Disables DPI awareness on older versions of GLFW since it's not supported. Display scaling in user settings should still work. Fixes #371