SatDump/SatDump #376: UI scaling weird on Macbook pro 14" 2021

Smile249, 2023-09-02:
**Description of the issue**

The UI doesn't seem to be scaled properly unless I use a scaled resolution such as 2048x1330 instead of 1512x982 (default res).

**Hardware (SDR/PC/OS)**

Macbook pro 14" 2021 (edition with notch), M1 Pro, MacOS Ventura 13.4

**Version (Eg, 1.0.0, CI Build #171)**

SatDump v1.1.0-37c1b6e
https://github.com/SatDump/SatDump/commit/57cd337db32d30f0c0ad6aee7ec6312f1a3260dd

**Logs after the cras (satdump.logs)**

no crash; visual bug

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

*Insanely annoying on live processing tab as the UI takes up the entirety of the FFT.*

Default resolution:
[image]

Scaled resolution:
[image]

JVital2013, 2023-09-02:
There must be a bug on some retina displays. I'll try to reproduce and will get back to you!

In the meantime, you can change the "UI scale" within SatDump settings to 0.5 to work around the issue without changing your system preferences.

JVital2013, 2023-09-04:
@Smile249 I have reproduced the bug on a test machine and I have it fixed on my end. Could you try the build here and let me know if it works for you: https://github.com/JVital2013/SatDump/suites/15785586318/artifacts/900809641

Don't forget to set UI Scaling back to 1.0 when testing. Thanks!

Smile249, 2023-09-04:
Everything looks good! Thanks a lot for fixing this issue!

PR #384 (2023-09-04): macOS: Improve Retina Support
This PR improves display scaling for retina displays on macOS. It also improves font rendering on the same screens to make text look crisper.

`style::macos_framebuffer_scale()` is a bit of a workaround: we should be able to use `ImGui::GetIO().DisplayFramebufferScale` to get the same value, but it doesn't work consistently for me. Maybe I'm doing something wrong, but the native API seems to work more reliably.

Fixes #376
closed 2023-09-04 by Smile249