SatDump/SatDump #377: error when installing on linux debian

JimmyeJones, 2023-09-02:
**Consolidate compiler generated dependencies of target satdump_core
[ 97%] Built target satdump_core
Consolidate compiler generated dependencies of target satdump
Consolidate compiler generated dependencies of target fengyun2_support
Consolidate compiler generated dependencies of target spacex_support
Consolidate compiler generated dependencies of target satdump_interface
[ 97%] Built target satdump
[ 97%] Built target spacex_support
[ 97%] Built target satdump_interface
Consolidate compiler generated dependencies of target fengyun4_support
Consolidate compiler generated dependencies of target proba_support
[ 97%] Built target fengyun2_support
Consolidate compiler generated dependencies of target fengyun3_support
Consolidate compiler generated dependencies of target eos_support
[ 97%] Built target fengyun4_support
[ 97%] Built target proba_support
Consolidate compiler generated dependencies of target meteor_support
Consolidate compiler generated dependencies of target jpss_support
[ 97%] Built target fengyun3_support
[ 97%] Built target jpss_support
[ 97%] Built target meteor_support
[ 97%] Built target eos_support
Consolidate compiler generated dependencies of target goes_support
Consolidate compiler generated dependencies of target gk2a_support
Consolidate compiler generated dependencies of target elektro_arktika_support
Consolidate compiler generated dependencies of target noaa_metop_support
[ 97%] Built target gk2a_support
Consolidate compiler generated dependencies of target dmsp_support
[ 97%] Built target elektro_arktika_support
[ 97%] Built target dmsp_support
[ 97%] Built target goes_support
Consolidate compiler generated dependencies of target jason3_support
[ 97%] Built target noaa_metop_support
Consolidate compiler generated dependencies of target aim_support
Consolidate compiler generated dependencies of target oceansat_support
Consolidate compiler generated dependencies of target mats_support
[ 97%] Built target jason3_support
[ 97%] Built target aim_support
Consolidate compiler generated dependencies of target himawari_support
Consolidate compiler generated dependencies of target bluewalker3_support
[ 97%] Built target oceansat_support
[ 97%] Built target himawari_support
[ 97%] Built target bluewalker3_support
Consolidate compiler generated dependencies of target analog_support
Consolidate compiler generated dependencies of target orbcomm_support
[ 97%] Built target mats_support
Consolidate compiler generated dependencies of target others_support
[ 97%] Built target orbcomm_support
[ 97%] Built target analog_support
Consolidate compiler generated dependencies of target airspy_sdr_support
Consolidate compiler generated dependencies of target hackrf_sdr_support
Consolidate compiler generated dependencies of target airspyhf_sdr_support
[ 97%] Built target airspy_sdr_support
[ 97%] Built target others_support
Consolidate compiler generated dependencies of target rtlsdr_sdr_support
[ 97%] Built target hackrf_sdr_support
Consolidate compiler generated dependencies of target mirisdr_sdr_support
[ 97%] Built target airspyhf_sdr_support
[ 97%] Building CXX object plugins/sdr_sources/rtlsdr_sdr_support/CMakeFiles/rtlsdr_sdr_support.dir/rtlsdr_sdr.cpp.o
Consolidate compiler generated dependencies of target spyserver_support
[ 97%] Built target mirisdr_sdr_support
Consolidate compiler generated dependencies of target rtltcp_support
Consolidate compiler generated dependencies of target sdrpp_server_support
[ 97%] Built target spyserver_support
Consolidate compiler generated dependencies of target satdump-ui
[ 97%] Built target rtltcp_support
[ 97%] Built target sdrpp_server_support
[ 97%] Built target satdump-ui
/home/orangepi/SatDump/SatDump-1.0.3/plugins/sdr_sources/rtlsdr_sdr_support/rtlsdr_sdr.cpp: In member function ‘void RtlSdrSource::set_bias()’:
/home/orangepi/SatDump/SatDump-1.0.3/plugins/sdr_sources/rtlsdr_sdr_support/rtlsdr_sdr.cpp:84:5: error: ‘rtlsdr_set_bias_tee’ was not declared in this scope; did you mean ‘rtlsdr_set_xtal_freq’?
   84 |     rtlsdr_set_bias_tee(rtlsdr_dev_obj, bias_enabled);
      |     ^~~~~~~~~~~~~~~~~~~
      |     rtlsdr_set_xtal_freq
make[2]: *** [plugins/sdr_sources/rtlsdr_sdr_support/CMakeFiles/rtlsdr_sdr_support.dir/build.make:90: plugins/sdr_sources/rtlsdr_sdr_support/CMakeFiles/rtlsdr_sdr_support.dir/rtlsdr_sdr.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1579: plugins/sdr_sources/rtlsdr_sdr_support/CMakeFiles/rtlsdr_sdr_support.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
**

**I am running debian on orange pi 3 LTS**

**Version 1.0.3**

JVital2013, 2023-09-02:
What version of Debian are you using? Have you installed librtlsdr-dev from apt?

SatDump wants a version of librtlsdr that supports Bias-T, and it looks like the version you installed is missing the necessary functions.

JimmyeJones, 2023-09-03:
Thanks for the reply!
I am using Debian 22.04 - this is the only version that I know of that works on the orange pi
I have installed librtlsdr-dev (version 0.6.0-4)

JimmyeJones, 2023-09-03:
I installed librtlsdr-dev from github, and I am no longer getting the error, however I am getting a new error;

[ 95%] Linking CXX shared library ../libdvb_support.so
[ 95%] Built target dvb_support
[ 95%] Building CXX object plugins/sdr_sources/usrp_sdr_support/CMakeFiles/usrp_sdr_support.dir/main.cpp.o
In file included from /usr/include/uhd/error.h:70,
                 from /usr/include/uhd.h:12,
                 from /home/orangepi/SatDump-master/plugins/sdr_sources/usrp_sdr_support/usrp_sdr.h:4,
                 from /home/orangepi/SatDump-master/plugins/sdr_sources/usrp_sdr_support/main.cpp:3:
/usr/include/uhd/config.hpp:10:10: fatal error: boost/config.hpp: No such file or directory
   10 | #include <boost/config.hpp>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [plugins/sdr_sources/usrp_sdr_support/CMakeFiles/usrp_sdr_support.dir/build.make:76: plugins/sdr_sources/usrp_sdr_support/CMakeFiles/usrp_sdr_support.dir/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2206: plugins/sdr_sources/usrp_sdr_support/CMakeFiles/usrp_sdr_support.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 95%] Building CXX object plugins/inmarsat_support/CMakeFiles/inmarsat_support.dir/stdc/decode_utils.cpp.o
[ 95%] Linking CXX shared library ../../libhackrf_sdr_support.so
[ 96%] Building CXX object plugins/inmarsat_support/CMakeFiles/inmarsat_support.dir/stdc/egc_parser.cpp.o
[ 96%] Built target hackrf_sdr_support
[ 96%] Building CXX object plugins/inmarsat_support/CMakeFiles/inmarsat_support.dir/stdc/module_stdc_decoder.cpp.o
[ 96%] Building CXX object plugins/inmarsat_support/CMakeFiles/inmarsat_support.dir/stdc/module_stdc_parser.cpp.o
[ 96%] Building CXX object plugins/sdr_sources/bladerf_sdr_support/CMakeFiles/bladerf_sdr_support.dir/bladerf_sdr_source.cpp.o
[ 96%] Building CXX object plugins/inmarsat_support/CMakeFiles/inmarsat_support.dir/stdc/msg_parser.cpp.o
[ 96%] Building CXX object plugins/inmarsat_support/CMakeFiles/inmarsat_support.dir/stdc/pkt_parser.cpp.o
[ 96%] Building CXX object plugins/sdr_sources/bladerf_sdr_support/CMakeFiles/bladerf_sdr_support.dir/main.cpp.o
[ 96%] Building CXX object plugins/inmarsat_support/CMakeFiles/inmarsat_support.dir/stdc/pkt_structs.cpp.o
[ 96%] Linking CXX shared library ../../libbladerf_sdr_support.so
[ 96%] Built target bladerf_sdr_support
[ 96%] Linking CXX shared library ../libinmarsat_support.so
[ 96%] Built target inmarsat_support
make: *** [Makefile:156: all] Error 2

JVital2013, 2023-09-03:
It looks like you're missing libboost-dev. Try installing that, then compile again.

JimmyeJones, 2023-09-03:
It installed successfully, however it is now giving an error when I try to run the UI.
This is what happens;

[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030060
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 765461532.69072896
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030063
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 69056512.-939512400
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030062
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL -402643216.764675102
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030074
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 765461533.-1071656960
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030066
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 69072896.19766273
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030032
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL -939512400.-66838080
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030033
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 764675102.570163500
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030060
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL -1071656960.2917380
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030055
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 19766273.-1677450576
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030062
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL -66838080.619184429
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030074
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 570163500.2547716
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x0003006C
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 2917380.-2080103032
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030060
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL -1677450576.630718508
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030074
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 619184429.19766276
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030066
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 2547716.-603707996
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030075
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL -2080103032.632291367
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030075
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 630718508.2922500
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030066
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 19766276.-1811667524
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030073
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL -603707996.696516653
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x0003004A
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 632291367.-2013265916
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030054
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 2922500.784598058
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030075
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL -1811667524.69901312
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030032
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 696516653.-1342165316
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030031
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL -2013265916.783549482
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030076
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 784598058.69911552
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x0003006F
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 69901312.-671076692
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x0003006A
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL -1342165316.768345130
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030072
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 783549482.69971968
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030076
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 69911552.-2147472208
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030066
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL -671076692.652739628
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030060
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 768345130.70094848
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030071
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 69971968.16788916
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030075
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL -2147472208.0
[... cut ...]
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030060
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 0.0
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030053
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 230784.230912
[20:49:53 - 03/09/2023] (E) Glfw Error 65539: Invalid client API 0x00030066
[20:49:53 - 03/09/2023] (C) Could not init GLFW Window with OpenGL 0.0
Segmentation fault (core dumped)

*this is only part of the message*

Aang23, 2023-09-03:
Which GPU are you using? Does OpenGL work on your system?

JVital2013, 2023-09-04:
From what I'm reading, The Mali T720 supports OpenGL ES, which is slightly different from "standard" OpenGL. Try invoking OpenGL from the command line like this (same as Raspberry Pi):

```bash
export MESA_GL_VERSION_OVERRIDE=4.5
satdump-ui
```

If that still does not work, try running `apt install mesa-utils`, then run `glxgears` and make sure it renders properly. If it does not, something is wrong with your OpenGL install. Running `glxinfo` should give you some more information.

JimmyeJones, 2023-09-04:
It works! Thanks so much for all of your help :)

closed 2023-09-04 by JimmyeJones