SatDump/SatDump #385: Not properly decoding WAV on nightly Mac build

RCCRAFT1, 2023-09-04:
**Description of the issue**
The software crashes when I try to process a WAV recording file. It also makes me choose a baseband format when I never recorded it in baseband.
The software also only allows me to decode in baseband, otherwise it will crash. (even though as I said before, I never recorded it in baseband)

**Hardware (SDR/PC/OS)**
MacBook Air M2 running MacOS Sonoma 14.0

**Version (Eg, 1.0.0, CI Build #171)**
Latest nightly build Mac version (No specific build number)

**Logs after the cras (satdump.logs)**
This file isn't on Mac. I have the Apple report logs

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
I cannot gather any screenshots (since it crashes as soon as I hit start), but if somebody would like my Apple report logs I will give them to you

JVital2013, 2023-09-04:
Thanks for trying the new Mac nightly builds!

What type of wav file are you decoding - is it a NOAA APT auto recording in a wav file? How was it recorded?

Can you please send:

- That Apple crash report
- A screenshot of SatDump from right before you click start (to make sure I try to exactly reproduce the scenario)
- the wav that causes the crash, if possible. 

Thanks!

Aang23, 2023-09-04:
It would be nice to have the satdump.logs as well. I guess it's dumped somewhere "contained" on MacOS?

JVital2013, 2023-09-04:
@Aang23 If the app is in a writable location, satdump.logs is within the app bundle itself in Contents/Resources/satdump.logs. If it's not in a writable location, like the disk image or /Applications folder, no log is saved. 

If you'd like, I can push a PR that saves it in the normal macOS ~/Library/Logs folder. For simplicity sake, I'd probably make it always save the log there on macOS, no matter how it's invoked.

Would that be helpful?

Aang23, 2023-09-04:
I'd say that's a good idea. The log file should be easy to find to debug this sort of stuff.

JVital2013, 2023-09-04:
Sounds good - I'll push it later today!

Until then, it would be great to see what other info you have @RCCRAFT1 for your specific issue.

RCCRAFT1, 2023-09-04:
> Thanks for trying the new Mac nightly builds!
>
> What type of wav file are you decoding - is it a NOAA APT auto recording in a wav file? How was it recorded?
>
> Can you please send:
>
> * That Apple crash [...]

Here are the logs Apple produced:
[logs.txt.txt](https://github.com/SatDump/SatDump/files/12517011/logs.txt.txt)

I am unable to send the file because it is 12GB. The error doesn't seem to be produced on my Windows machine, but I cannot send the WAV to my PC because of the size of the file. 

Heres a screenshot of what I put before I click start
[image]

JVital2013, 2023-09-04:
I see the problem - you're asking it to decode a **baseband** .wav, but the input level you selected is for an **audio** wav. You want SatDump configured like this (assuming that .wav came from something like SDR++, and it was set for Int16 on record):

[image]

Some other things to note:

- Also keep an eye on the sample rate. The samplerate setting should match what's in the filename (assuming the filename is correct)
- I'd recommend getting the latest SatDump build, published a few minutes ago. It should fix that blurry text, and makes logs available in the Console app.

RCCRAFT1, 2023-09-04:
> I see the problem - you're asking it to decode a **baseband** .wav, but the input level you selected is for an **audio** wav. You want SatDump configured like this (assuming that .wav came from [...]

I am currently recording directly in satdump (I have tried to record using other software before, and they either don't work, are a pain in the ass to get a recording out of, or in the case of SDR++ it just doesn't install)

If I try to process it as a baseband file, it doesn't generate a full image like I'm assuming it should.

JVital2013, 2023-09-04:
If you're not getting an image on decode, make sure you match the baseband format and sample rate to what you used when you recorded it. It might not be s16 - I think the default is f32, but we can't tell what you recorded it at. 

Also, was the satellite centered in the baseband when you recorded it? If not, you'll need to set a frequency offset.

RCCRAFT1, 2023-09-04:
> If you're not getting an image on decode, make sure you match the baseband format and sample rate to what you used when you recorded it. It might not be s16 - I think the default is f32, but we [...]

I set the recording format to "wav16" and it recorded it as a .wav file. The signal was also perfectly centered on the waterfall and it did not have any problems recording it.

JVital2013, 2023-09-04:
I'd encourage you to check the samplerate again. In the screenshot you sent, you had one too few zeros specified in the UI. 

If that's not it, are you getting any image at all? Getting the logs after an attempted decode would be useful as well, now that logging on macOS is fixed

RCCRAFT1, 2023-09-05:
> I'd encourage you to check the samplerate again. In the screenshot you sent, you had one too few zeros specified in the UI.
> 
> If that's not it, are you getting any image at all? Getting the logs after an attempted decode would be useful as well, now that logging on macOS is fixed

I recorded the data before the newest nightly update, so I have no idea where the logs would be. 

I do get an image, but it doesn't generate completely because of the formats not matching up.

What settings should I use for when I'm attempting to decode an actual .wav file?

JVital2013, 2023-09-05:
> I recorded the data before the newest nightly update, so I have no idea where the logs would be.

That's OK - the logs for your failed decode attempts are mostly what we're looking for since I (currently) don't have any reason to believe the baseband recorded incorrectly. They are at ~/Library/Logs/satdump.log (or in the Console program under "Log Reports") once you update and try to decode your recording again.

> What settings should I use for when I'm attempting to decode an actual .wav file?

I guess that depends on what you mean by an "actual wav"

- If you mean an audio file, then it should be set up like you had in the first screenshot. SatDump will auto detect the samplerate, and it'll be something much lower since audio files have _very_ low sample rates compared to basebands.
- If you mean a wav16 like SatDump records, it's like my screenshot. In that situation, it looks like you need to specify the samplerate yourself (a bug we should look at...)

I hope this helps!

JVital2013, 2023-09-05:
The log is likely missing since the older version you recorded it on didn't save it there, and now you don't have the file to decode again - no worries!

You can also try decoding it live to see how that works for you. Good luck on the next pass!

RCCRAFT1, 2023-09-07:
> The log is likely missing since the older version you recorded it on didn't save it there, and now you don't have the file to decode again - no worries!
> 
> You can also try decoding it live to see how that works for you. Good luck on the next pass!

Hello again,

I was able to capture a file from NOAA 19 and process it though NOAA-APT on my Windows machine when I recorded it in plain WAV in SDR++. Satdump still crashes with the same error when I try to process this file. Is there any way I can fix this?

JVital2013, 2023-09-08:
Can you share the plain wav, and the log file located at /Users/\<username\>/Library/Logs/satdump.log so we can troubleshoot?

RCCRAFT1, 2023-09-11:
> Can you share the plain wav, and the log file located at /Users/<username>/Library/Logs/satdump.log so we can troubleshoot?

The software no longer crashes, but it still does not decode WAV properly and I get no image, but I am able to process them through NOAA-APT on my windows machine and I still have a good image

Aang23, 2023-09-14:
Can you show what you get @RCCRAFT1 ? And if you can send the file.

JVital2013, 2023-10-30:
@RCCRAFT1 any updates to this issue? Others can decode apt wavs on macOS. If you're still having this issue, let us know and we'll keep this issue open as long as we need to get to the bottom of this.

If not, or we don't hear back from you in a week or two, we will close this issue since it cannot be reproduced. Thanks!

JVital2013, 2023-11-17:
We haven't heard from you so I'm going to close this issue. If you'd like continued assistance, feel free to reach back out!

closed 2023-11-17 by JVital2013