SatDump/SatDump #406: Better algorithm for geometric image correction

LEDFlighter, 2023-09-16:
**What would you want to be added to satdump?**
I would like to have implemented a better algorithm for the geometric correction, so that the edges of the corrected images look less pixelated and more sharper and clearer instead.

**Describe the feature you'd like to have**
Something like this (https://github.com/Xerbo/meteor_corrector) would be nice to have, I heard that the same algorithm is most likely also implemented in Lean-HRPT (https://github.com/Xerbo/LeanHRPT-Decode).

**Example of the current one**
[image]

**Example of the better one**
[image]

In the "meteor_corrector.py"-Script I described earlier, there are also arguments for swath width, altitude, resolution in km/px and the option to flip the image. That would be also nice to have as different satellites have different parameters for altitude, swath width etc. We already have the "flip"-option implemented into SatDump, so that isn't needed.

Aang23, 2023-09-17:
This is more about adding interpolation than an actual algorithm change. Nothing too complicated, will probably do this quickly.

All parameters like swath, etc are also available and configured in "resources", as this is not something the user is expected to change once configured right for a specific satellite/instrument (but you can if you wish).

Aang23, 2023-09-20:
Added.

closed 2023-09-20 by Aang23