SatDump/SatDump #408: new satdump pipeline

ramers87, 2023-09-18:
Hello 
how i can add new satdump pipeline satellite ?
Thanks

Marhmarchi, 2023-09-18:
You must create a json file in the pipelines folder, following the same structure as the others.

ramers87, 2023-09-18:
Hi,
Thank you so much for your help i solved it.

Marhmarchi, 2023-09-18:
You're welcome 🤩

ramers87, 2023-09-18:
Just one more thing how i can make desktop icon for satdump to lunch it up

Marhmarchi, 2023-09-18:
Depends on your OS (Mac, Linux, Windows..), but when you build the package it creates the shortcut. If you didn't install it system-wide, then look for instructions for your OS, then use the satdump-ui as the executable on the shortcut.

ramers87, 2023-09-18:
Yes i use ./satdump-ui,but i need icon desktop,i use Debian Linux

Aang23, 2023-09-18:
If you just do `sudo make install` you should get the Desktop icon by default.

Also @ramers87 feel free to join the Matrix group mentioned in the README if you need more help!

ramers87, 2023-09-20:
Hi,
How to add a new pipeline satellite? ,only way is modify json file?

ramers87, 2023-09-20:
I didn't know the staeps  to join MATRIX groupe?

Crosswalkersam, 2023-09-20:
Yes, the only way to get SatDump to support a satelite is to modify or create a new .json file in the pipeline folder.

JVital2013, 2023-09-26:
@ramers87 To join Matrix, go to https://matrix.to/#/#satdump:altillimity.com.

Is there anything specific we can help you with?

JVital2013, 2023-10-30:
Since we haven't heard back from you, I'm going to close this issue. If you still need help, feel free to re-open it. Thanks!

closed 2023-10-30 by JVital2013

ramers87, 2023-10-30:
Hi
Thank you so much it was helpful.
Regards

commit 5660036e (2023-09-20): Interpolate in earth curvature correction. Fixes #408