SatDump/SatDump #413: Satdump crash with Windows 11 systems how to fix!

IZ5RZR, 2023-09-20:
In this video I show you what the problem is with the new satdump builds and how to solve it. the crash is caused by errors in downloading the 2line elements from celestrack, which has changed the download locations, therefore when the program tries to download the tles of some staellites it does not find them and reports error codes in the Satdump.tles text, therefore the the next time the program is started it fails to process the satdump.tles file and crashes.
The only way to solve it is to edit the Satdump.tles file and remove all the error codes, there are several, save and then it works!
I have make a video tutorial [https://youtu.be/U4HCo-WmYt4?si=hqDotg8_AnubrpS2](url)

IZ5RZR Max

Aang23, 2023-09-20:
Which build are you using exactly?

IZ5RZR, 2023-09-20:
HI

   all last builds from 28 jun 2023
   Da "Aang23" ***@***.***
   A "SatDump/SatDump" ***@***.***
   Cc "Max" ***@***.***,"Author" ***@***.***
   Data Wed, 20 Sep 2023 02:45:23 -0700
   Oggetto Re: [SatDump/SatDump] Satdump crash with Windows 11 systems how to fix! (Issue #413)

   Which build are you using exactly?

IZ5RZR, 2023-09-20:
> 28 jun is old. Can you try newer?

All recently buils are affected

IZ5RZR, 2023-09-20:
> Ok, can you send the bugger satdump_tles.txt?

Have you seen video?
here is satdump.tles , look all <HEAD><TITLE>Connection timed out</TITLE></HEAD>
<BODY BGCOLOR="white" FGCOLOR="black">
<FONT FACE="Helvetica,Arial"><B>
-----------------------------------------------------------------
METOP-A                 
1 29499U 06044A   23263.43761744  .00002357  00000+0  31089-3 0  9999
2 29499  98.3907 326.5172 0192964   3.4631   6.8172 14.68981089881011
METOP-B                 
1 38771U 12049A   23263.39222104  .00000240  00000+0  12941-3 0  9996
2 38771  98.6691 321.1331 0001366  98.9864 271.9342 14.21516599571072
METOP-C                 
1 43689U 18087A   23263.42708251  .00000231  00000+0  12550-3 0  9990
2 43689  98.6738 321.3529 0003400 134.6858 225.4597 14.21519325252657
DMSP 5D-3 F16 (USA 172) 
1 28054U 03048A   23263.34797512  .00000280  00000+0  17184-3 0  9991
2 28054  99.0179 262.3704 0008391  96.9624  15.1941 14.13849910 28008
DMSP 5D-3 F17 (USA 191) 
1 29522U 06050A   23263.31728388  .00000288  00000+0  17361-3 0  9995
2 29522  98.7544 274.4244 0010629  37.5973 322.5942 14.14292059870906
DMSP 5D-3 F18 (USA 210) 
1 35951U 09057A   23263.38920944  .00000399  00000+0  23366-3 0  9992
2 35951  98.7781 237.9586 0010274 212.9423 147.1110 14.13852682718092
NOAA 15                 
1 25338U 98030A   23263.34799656  .00000289  00000+0  13818-3 0  9998
2 25338  98.6005 290.7554 0011009  41.4520 318.7494 14.26391782318821
NOAA 18                 
1 28654U 05018A   23263.37894546  .00000327  00000+0  19909-3 0  9993
2 28654  98.9029 337.2825 0015298 110.0555 250.2265 14.13009150945011
NOAA 19                 
1 33591U 09005A   23263.36460833  .00000279  00000+0  17469-3 0  9991
2 33591  99.0905 310.7732 0013356 339.1005  20.9619 14.12823802753381
FENGYUN 3A              
1 32958U 08026A   23263.40962569  .00000259  00000+0  14427-3 0  9996
2 32958  98.5640 213.9408 0009469 129.0572 231.1448 14.18918012793560
FENGYUN 3B              
1 37214U 10059A   23263.36294661  .00000154  00000+0  10422-3 0  9996
2 37214  99.0822 282.3343 0022958 135.8180 351.6642 14.14376749665961
FENGYUN 3C              
1 39260U 13052A   23263.34916345  .00000190  00000+0  11660-3 0  9998
2 39260  98.4295 272.4041 0011732  34.0601 326.1325 14.16077245516996
FENGYUN 3D              
1 43010U 17072A   23263.38336897  .00000210  00000+0  12166-3 0  9999
2 43010  98.8696 212.0575 0002392  53.0078 307.1316 14.19205407303007
FENGYUN 3E              
1 49008U 21062A   23263.36821941  .00000045  00000+0  42166-4 0  9993
2 49008  98.7270 263.4042 0002702  90.7998 269.3487 14.19697300114566
FENGYUN 3F              
1 57490U 23111A   23263.40024024  .00000152  00000+0  92732-4 0  9999
2 57490  98.8060 329.5744 0001765 118.3135 241.8219 14.19706324  6856
FENGYUN 3G              
1 56232U 23055A   23263.37264179  .00042918  00000+0  68353-3 0  9999
2 56232  49.9971 106.8077 0007809  18.4953 341.6297 15.52752829 24444
METEOR-M 1              
1 35865U 09049A   23263.36340876  .00000239  00000+0  12616-3 0  9994
2 35865  98.5911 225.6952 0002119 359.7541   0.3635 14.22546184727066
METEOR-M 2              
1 40069U 14037A   23263.41217421  .00000185  00000+0  10491-3 0  9993
2 40069  98.4277 264.9784 0006915  69.9292 290.2630 14.20813117477172
METEOR-M2 2             
1 44387U 19038A   23263.39332889  .00000099  00000+0  62746-4 0  9994
2 44387  98.7866 227.1498 0002116  52.6209 307.5161 14.23806221218855
**<HEAD><TITLE>Connection timed out</TITLE></HEAD>
<BODY BGCOLOR="white" FGCOLOR="black">
<FONT FACE="Helvetica,Arial"><B>**
 Connection timed out</B></FONT>

</BODY>
[... 218 lines cut ...]
</BODY>
 GRACE-FO 2              
1 43477U 18047B   23262.74813337  .00004979  00000+0  18734-3 0  9990
2 43477  88.9901 336.3321 0015494  43.0659 317.1799 15.26819655296421
CRYOSAT 2               
1 36508U 10013A   23262.53582343  .00000456  00000+0  11247-3 0  9997
2 36508  92.0283  35.1649 0005778 112.8905 247.2912 14.51911460712783
PRISMA                  
1 44072U 19015A   23263.43852262  .00001691  00000+0  21127-3 0  9993
2 44072  97.8707 336.5405 0000989  82.5882 277.5445 14.83650365243668
LANDSAT 8               
1 39084U 13008A   23263.37297932  .00000746  00000+0  17557-3 0  9998
2 39084  98.2182 332.1000 0001194  95.5721 264.5613 14.57113250563975
LANDSAT 9               
1 49260U 21088A   23263.20130934  .00000773  00000+0  18171-3 0  9995
2 49260  98.2217 331.9212 0001385  92.8226 267.3131 14.57116895105229
YUNHAI 1                
1 41857U 16068A   23263.41954987  .00000513  00000+0  20330-3 0  9998
2 41857  98.5931 266.6485 0008467  81.5660 278.6483 14.33076135358502
YUNHAI 1-02             
1 44547U 19063A   23263.33373996  .00000405  00000+0  16374-3 0  9994
2 44547  98.4666 284.0068 0000262 241.4320 118.6838 14.33023061208535
YUNHAI 1-03             
1 53877U 22115A   23263.36944600  .00000486  00000+0  19359-3 0  9996
2 53877  98.5116 260.4015 0001081  67.9359 292.1940 14.33005077 52182

Aang23, 2023-09-20:
Can you send the file as a file and not copy-paste it over?

IZ5RZR, 2023-09-23:
> Can you send the file as a file and not copy-paste it over?
Ok here is file. just now updated from satdump, You will see many downloadings erroe that go into crash satdump
[satdump_tles.txt](https://github.com/SatDump/SatDump/files/12706873/satdump_tles.txt)

JVital2013, 2023-09-26:
Hello @IZ5RZR!

Thanks for the TLE file. We just made two changes that should fix your problem:

- SatDump now only queries a few URLs that get multiple TLEs at once, to prevent rate-limiting
- SatDump now verifies downloaded TLEs. Instead of crashing, it will gracefully skip any TLEs it failed to download.

Please try the latest nightly at https://github.com/SatDump/SatDump/releases/tag/nightly and let us know if it works for you!

JVital2013, 2023-09-30:
Hello again @IZ5RZR,

I see your posts on social media that this issue is now fixed for you. I'm going to go ahead and close this, but feel free to reach out if it acts up again!

closed 2023-09-30 by JVital2013