SatDump/SatDump #422: Please update the "Source code" for the "Nightly Builds"

gvimlag, 2023-09-24:
The "Source code" for the "**_Nightly Builds_**" has not been updated since **_June 28_** (nearly 3 months old).
Please update the "Source code" (both "nightly.zip" and "nightly.tar.gz").
Thank you.

JVital2013, 2023-09-25:
I bumped the nightly tag to the latest commit, which updated the source code archive. However, it will go stale again with time as this type of GitHub nightly tends to do.

The compiled builds on the nightly are always up to date, and the latest source code can always be downloaded from:

- https://github.com/SatDump/SatDump/archive/refs/heads/master.zip,
- https://github.com/SatDump/SatDump/archive/refs/heads/master.tar.gz
- A good ol' `git clone https://github.com/SatDump/SatDump`

closed 2023-09-25 by JVital2013

gvimlag, 2023-09-25:
I've been downloading "nightly builds" from "https://github.com/SatDump/SatDump/archive/refs/tags/nightly.tar.gz".
I'll switch to downloading "nightly builds" from "https://github.com/SatDump/SatDump/archive/refs/heads/master.tar.gz".
Thank you for the information and for updating the current "nightly builds".

JVital2013, 2023-09-25:
I added a note on the nightly release as well to help clear up any confusion. Thanks for reaching out!