SatDump/SatDump #423: The DC-Blocking issue with Mirics-based sticks

sergsero, 2023-09-26:
I am very glad that the program supports working with Mirics-based sticks, which I use, as well as RSP1A.

However, I have the DC-Blocking issue with MiriSDR. I always have a DC Spike in the center no matter where I go frequency. Unfortunately I didn't find a possibility to get rid of it. 

[image]

At the same time I don't have any issue with the DC spike on the RSP1A.

[image]

I know that nearly every practical SDR with a quadrature mixer has this issue. Of course for signals that don't require the total bandwidth of a device can tune above or below the signal frequency so it doesn't overlap with the DC offset and than use "Frequency shift".  But for some reason it doesn't work for me either.

Would it be possible to do something about it for Mirics-based sticks?

JVital2013, 2023-09-26:
Hi @sergsero,

For your primary question about blocking the DC spike, I don't think the underlying MiriSDR library has built-in support for DC blocking. I'll let @Aang23 speak to that if something can be done manually.

As for the "frequency shift" not working for you: the Frequency shift setting does not change what is shown on the FFT/waterfall, but it **should** affect what goes into the demodulator for processing. I use this with my MiriSDR.

For example, if you:

- Have a signal at 1693 MHz
- Tune your SDR to 1694 MHz
- Set "Frequency Shift" to 1 MHz

This should allow the signal to decode, even though the signal still shows as offset to the left.

sergsero, 2023-09-27:
Hello Jamie,

Thanks for your reply, I will do that.
Indeed, it was not clear to me that when the Frequency shift setting, there are no changes on the spectrum scope (FFT)/waterfall.

Maybe if there was a vertical red line in the center of the FFT window that shows which frequency the receiver is currently tuned to, it would be more visual. And its position will depending on the specified frequency shift, if there is one :).

I will be very grateful if you can do something with 'DC spike' using software processing.

By the way, the 'DC blocking' control is located in the 'Processing' section now, although most likely, is it more related to the Device or not?

I would say again, thank you so much for the hard work of your team!

Best regards,
sergsero

JVital2013, 2023-09-27:
**About the DC block button** - honestly, that button somehow slipped my mind in my first response. Sorry about that! It should do exactly what you want - it removes the DC spike from the center, so there should not be a need to change the MiriSDR implementation if everything's working right. 

Just like the frequency shift button though, the change is not reflected in the FFT/waterfall of the recorder UI. However, if you do offline processing, these setting _are_ reflected in the "show fft" floating FFT/waterfall due to how offline processing works.

The reason these functions are under processing instead of with SDR settings, is because they don't control how the SDR operates. Instead, they are "post-processing" operations done on the signal after being received, but before being processed. If you record a baseband recording while one or both of these options are set, the baseband will also not reflect the settings.

**About the visual indicator for frequency shift:** there are changes planned to implement multi-VFO in the UI. It would make sense to add a visual indicator along with that to show a given demodulator's center frequency, so stay tuned!