SatDump/SatDump #432: The frozen images from EWS-G1 in the latest commits

sergsero, 2023-09-30:
For some time now I have a problem with getting pictures from EWS-G1 (GOES-13).
As know, the GVAR transmission session from this satellite is from the 15th minute to the ~40th minute of the hour and from the 45th minute to the ~10th minute of the next hour.

In earlier versions, if I was start receiving at an arbitrary moment during the transmission session, then part of the hemisphere image was successfully created, starting from this moment until the end of the session.

Further, all consecutive reception sessions resulted in obtaining full-fledged images. Both in the GUI and in the output directory.

However, in the latest versions of the program, I get, both in the GUI window and in the output directory, about the same frozen double image on the equatorial zone of the transmitted snapshot. The line of percent counter is also wrong.

[image]

[image]

[image]
[image]
[image]

The same erroneous behavior occurs in the receive mode from the command line (./satdump live ...).

Is the reason in my setup? However, it has not changed since the time when the reception was successful (lshw -short: Linux 6.2.0-33-generic #33~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Sep  7 10:33:52 UTC 2 x86_64 GNU/Linux, 11th Gen Intel(R) Core(TM) i7-11700K @ 3.60GHz, RocketLake-S GT1 [UHD Graphics 750]) 

I also noticed that after all functions stopping in the GUI version (the stop on all buttons), the processor utilization is preserved, as during the time when there was reception and processing (~500% by 'top').

[image]

sergsero, 2023-09-30:
Folders are formed in the output directory, the creation time of which and the names do not coincide with the standard periods of data transmission from the satellite.

The files contain the same fragment (maybe with different duration) of the equatorial part of the image.

The file size, for example, G13_FC_20230930T102547Z.png, is correct: 12156x18956. But the image occupies only part of the field, the rest of the picture is a black field.

[image]

JVital2013, 2023-10-01:
On September 27th, the scan mode was changed for EWS-G1 - https://cimss.ssec.wisc.edu/satellite-blog/archives/54685. This likely broke something in SatDump. Probably best to see if @Aang23 of @ZbychuButItWasTaken has any insights on why your issue is happening now.

Sorry about that!

ZbychuButItWasTaken, 2023-10-01:
Yes, the satellite has swiched modes and we do not yet have support for the new settings.

Aang23, 2023-10-01:
Actually, support is in. But it was made for the US configuration, not GOES-13, hence the partially working result.

ZbychuButItWasTaken, 2023-10-01:
Yes, that's what I meant

sergsero, 2023-10-02:
Thank you for clarifying the situation. 
I hope that changes will be made that will allow to receive images in the new scan mode for EWS-G1.

Considering that on the resource by the link above, it is reported about GOES-15, which has become EWS-G2 and is moving to some point over the Indian Ocean. Now it's already somewhere around 96°E.

I made the GVAR record. Maybe it will help somehow. The file contains a record today (20231002) with a duration of 1 hour and 25 minutes.
https://www.dropbox.com/scl/fi/w40jv0rlz3bvf9m5ol2wg/goes-13-gvar.tgz?rlkey=fo6rb181i84hj22pvaodqjtb3&dl=0

ZbychuButItWasTaken, 2023-10-02:
Oh! That's great, a recording is always useful. Thanks! We'll see what can be done, but it won't be the highest priority for now, with all the stuff currently in development..