SatDump/SatDump #433: Error when installing with cmake

JimmyeJones, 2023-09-30:
When I run this command; "cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr .." I get this error; 

-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Compiling with OpenMP support!
-- Found ZSTD: /usr/lib/aarch64-linux-gnu/libzstd.so  
-- Commit has no tag (undefined). Assume dev build
CMake Error at CMakeLists.txt:125 (add_compile_definitions):
  Unknown CMake command "add_compile_definitions".

-- Configuring incomplete, errors occurred!
See also "/home/orangepi/satdump/build/CMakeFiles/CMakeOutput.log".

**Hardware - Orange Pi 3 LTS 
OS - NAME="Ubuntu"
        VERSION="18.04.6 LTS (Bionic Beaver)"

SatDump Version - 1.1.0

Any help is greatly apreciated:)
**

JVital2013, 2023-10-01:
SatDump requires CMake 3.12 or higher - you must be on an old version. The following instructions worked on Ubuntu 18.04 for me: https://askubuntu.com/questions/355565/how-do-i-install-the-latest-version-of-cmake-from-the-command-line

I think your gcc version is OK, but you may need to update it as well. Here are instructions for that if you run into issues: https://devicetests.com/install-gcc-9-ubuntu-18-04-lts

Please keep in mind that Ubuntu 18.04 is no longer supposed by Canonical, unless you pay for the extended support. Either way, upgrading is recommended as you will likely keep running into this type of issue with other software.

Let us know how it works for you!

Aang23, 2023-10-01:
Indeed, you will need to update your CMake. It is very complicated to keep support for older distributions without running through far too many hoops.

It is still possible to build SatDump even on Ubuntu 16.04, but you will need to manually update quite a bit of stuff.

I will not be modifying the CMake code to be compatible with older versions. Users still running on such distributions are expected to go through what @JVital2013 documented, or similar.

closed 2023-10-01 by Aang23

JimmyeJones, 2023-10-01:
**Thanks for the reply!**
**I updated Cmake and ggc, but I am now getting multiple failed tests.**

-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Compiling with OpenMP support!
-- Found ZSTD: /usr/lib/aarch64-linux-gnu/libzstd.so  
-- Commit has no tag (undefined). Assume dev build
-- Looking for CL_VERSION_3_0
-- Looking for CL_VERSION_3_0 - not found
-- Looking for CL_VERSION_2_2
-- Looking for CL_VERSION_2_2 - found
-- Found OpenCL: /usr/lib/aarch64-linux-gnu/libOpenCL.so (found version "2.2") 
OpenCL Found! SatDump will support accelerated GPU computing.
Building with ZIQ Support
-- Performing Test STD_HAS_FILESYSTEM
-- Performing Test STD_HAS_FILESYSTEM - Failed
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Checking for module 'volk'
--   Found volk, version 1.3
-- Performing Test VOLK_HAS_ALLOC
-- Performing Test VOLK_HAS_ALLOC - Failed
-- VOLK does not have volk_alloc.hh, including volk_includes
-- Found ZLIB: /usr/lib/aarch64-linux-gnu/libz.so (found version "1.2.11") 
-- Found PNG: /usr/lib/aarch64-linux-gnu/libpng.so (found version "1.6.34") 
-- Checking for module 'fftw3f'
--   Found fftw3f, version 3.3.7
-- Found LuaJit: /usr/lib/aarch64-linux-gnu/libluajit-5.1.so  
-- Building with LuaJIT!
-- Performing Test VOLK_HAS_volk_32fc_x2_add_32fc
-- Performing Test VOLK_HAS_volk_32fc_x2_add_32fc - Failed
Building the GUI
-- Found OpenGL: /usr/lib/aarch64-linux-gnu/libOpenGL.so   
-- Found GLEW: /usr/include (found version "2.0.0") 
Could not find HDF5 libraries. Geonetcast HDF5 processing disabled!
-- Try SSE41 C flag = [ ]
-- Performing Test HAVE_SSE41
-- Performing Test HAVE_SSE41 - Failed
-- Try SSE41 C flag = [-march=corei7]
-- Performing Test HAVE_SSE41
-- Performing Test HAVE_SSE41 - Failed
-- Try SSE41 C flag = [-msse4]
-- Performing Test HAVE_SSE41
-- Performing Test HAVE_SSE41 - Failed
-- Try SSE41 C flag = [-msse4.1]
-- Performing Test HAVE_SSE41
-- Performing Test HAVE_SSE41 - Failed
-- Could NOT find SSE41 (missing: SSE41_C_FLAGS) 
SSE Features not found! This is only an error on x86 CPUs that should support SSE4.1
-- Could NOT find SSE41 (missing: SSE41_C_FLAGS) 
SSE41 Features not found! This is only an error on x86 CPUs that should support SSE41
-- Try NEON C flag = [ ]
-- Performing Test HAVE_NEON
-- Performing Test HAVE_NEON - Success
-- Found NEON:    
Found NEON features, enabling NEON plugin
-- Performing Test C_HAS_AVX_1
-- Performing Test C_HAS_AVX_1 - Failed
-- Performing Test C_HAS_AVX_2
-- Performing Test C_HAS_AVX_2 - Failed
-- Performing Test C_HAS_AVX_3
-- Performing Test C_HAS_AVX_3 - Failed
-- Performing Test C_HAS_AVX2_1
-- Performing Test C_HAS_AVX2_1 - Failed
-- Performing Test C_HAS_AVX2_2
-- Performing Test C_HAS_AVX2_2 - Failed
-- Performing Test C_HAS_AVX2_3
-- Performing Test C_HAS_AVX2_3 - Failed
-- Performing Test C_HAS_AVX512_1
-- Performing Test C_HAS_AVX512_1 - Failed
-- Performing Test C_HAS_AVX512_2
-- Performing Test C_HAS_AVX512_2 - Failed
-- Performing Test C_HAS_AVX512_3
-- Performing Test C_HAS_AVX512_3 - Failed
-- Performing Test CXX_HAS_AVX_1
-- Performing Test CXX_HAS_AVX_1 - Failed
-- Performing Test CXX_HAS_AVX_2
-- Performing Test CXX_HAS_AVX_2 - Failed
-- Performing Test CXX_HAS_AVX_3
-- Performing Test CXX_HAS_AVX_3 - Failed
-- Performing Test CXX_HAS_AVX2_1
-- Performing Test CXX_HAS_AVX2_1 - Failed
-- Performing Test CXX_HAS_AVX2_2
-- Performing Test CXX_HAS_AVX2_2 - Failed
-- Performing Test CXX_HAS_AVX2_3
-- Performing Test CXX_HAS_AVX2_3 - Failed
-- Performing Test CXX_HAS_AVX512_1
-- Performing Test CXX_HAS_AVX512_1 - Failed
-- Performing Test CXX_HAS_AVX512_2
-- Performing Test CXX_HAS_AVX512_2 - Failed
-- Performing Test CXX_HAS_AVX512_3
-- Performing Test CXX_HAS_AVX512_3 - Failed
AVX2 Features not found! This is only an error on x86 CPUs that should support AVX2
Building with Airspy support
Building with AirspyHF support
Building with HackRF support
Building with LimeSDR support
SDRlay Library could not be found! Not building.
IIO or AD9361 Library could not be found! Not building.
Building with BladeRF support
-- Performing Test BLADERF_HAVE_WIDEBAND
-- Performing Test BLADERF_HAVE_WIDEBAND - Failed
UHD Library could not be found! Not building.
Building with RTL-SDR support
Building with MiriSDR support
Building with SpyServer support
Building with RTL-TCP support
Building with SDR++ Server support
Building with UDP Source support
Building with Remote Source support
RtAudio Library could not be found! Not building.
AAudio Library could not be found! Not building.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/orangepi/satdump/build

JVital2013, 2023-10-01:
A few things:

- CMake is still reporting that your gcc/g++ versions are 7.5.0. Make sure old versions are removed, or that calling "gcc/g++" calls the new versions
- All the "failures" you have listed look normal - it won't compile with UHD, SDRPlay, IIO/AD9361, etc since the libraries are not installed. This is not a problem if you don't use these SDRs. If you need them, install their development libraries. The CPU feature checks are also OK since this is for x86-based computers, but you're on an orange pi so these features do not exist.

I hope this helps!

JimmyeJones, 2023-10-01:
**Thanks for the help :) I ran the next command, and got this error;**

[ 15%] Building CXX object src-core/CMakeFiles/satdump_core.dir/common/widgets/notated_num.cpp.o
[ 15%] Building CXX object src-core/CMakeFiles/satdump_core.dir/common/widgets/snr_plot.cpp.o
[ 15%] Building CXX object src-core/CMakeFiles/satdump_core.dir/common/widgets/switch.cpp.o
[ 15%] Building CXX object src-core/CMakeFiles/satdump_core.dir/common/widgets/value_plot.cpp.o
[ 16%] Building CXX object src-core/CMakeFiles/satdump_core.dir/common/widgets/waterfall_plot.cpp.o
[ 16%] Building CXX object src-core/CMakeFiles/satdump_core.dir/common/ziq.cpp.o
[ 16%] Building CXX object src-core/CMakeFiles/satdump_core.dir/common/ziq2.cpp.o
[ 16%] Building CXX object src-core/CMakeFiles/satdump_core.dir/core/config.cpp.o
[ 16%] Building CXX object src-core/CMakeFiles/satdump_core.dir/core/live_pipeline.cpp.o
/home/orangepi/satdump/src-core/common/ziq.cpp: In constructor ‘ziq::ziq_writer::ziq_writer(ziq::ziq_cfg, std::ofstream&)’:
/home/orangepi/satdump/src-core/common/ziq.cpp:27:46: error: ‘ZSTD_c_compressionLevel’ was not declared in this scope
   27 |             ZSTD_CCtx_setParameter(zstd_ctx, ZSTD_c_compressionLevel, zst_level);
      |                                              ^~~~~~~~~~~~~~~~~~~~~~~
/home/orangepi/satdump/src-core/common/ziq.cpp:27:13: error: ‘ZSTD_CCtx_setParameter’ was not declared in this scope
   27 |             ZSTD_CCtx_setParameter(zstd_ctx, ZSTD_c_compressionLevel, zst_level);
      |             ^~~~~~~~~~~~~~~~~~~~~~
/home/orangepi/satdump/src-core/common/ziq.cpp:28:46: error: ‘ZSTD_c_checksumFlag’ was not declared in this scope
   28 |             ZSTD_CCtx_setParameter(zstd_ctx, ZSTD_c_checksumFlag, 1);
      |                                              ^~~~~~~~~~~~~~~~~~~
/home/orangepi/satdump/src-core/common/ziq.cpp:29:46: error: ‘ZSTD_c_nbWorkers’ was not declared in this scope
   29 |             ZSTD_CCtx_setParameter(zstd_ctx, ZSTD_c_nbWorkers, zst_workers);
      |                                              ^~~~~~~~~~~~~~~~
/home/orangepi/satdump/src-core/common/ziq.cpp: In member function ‘int ziq::ziq_writer::compress_and_write(uint8_t*, size_t)’:
/home/orangepi/satdump/src-core/common/ziq.cpp:61:71: error: ‘ZSTD_e_continue’ was not declared in this scope
   61 |             ZSTD_compressStream2(zstd_ctx, &zstd_output, &zstd_input, ZSTD_e_continue);
      |                                                                       ^~~~~~~~~~~~~~~
/home/orangepi/satdump/src-core/common/ziq.cpp:61:13: error: ‘ZSTD_compressStream2’ was not declared in this scope; did you mean ‘ZSTD_compressStream’?
   61 |             ZSTD_compressStream2(zstd_ctx, &zstd_output, &zstd_input, ZSTD_e_continue);
      |             ^~~~~~~~~~~~~~~~~~~~
      |             ZSTD_compressStream
/home/orangepi/satdump/src-core/common/ziq.cpp: In member function ‘bool ziq::ziq_reader::seekg(size_t)’:
/home/orangepi/satdump/src-core/common/ziq.cpp:180:49: error: ‘ZSTD_reset_session_only’ was not declared in this scope
  180 |                 err = ZSTD_DCtx_reset(zstd_ctx, ZSTD_reset_session_only);
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~
/home/orangepi/satdump/src-core/common/ziq.cpp:180:23: error: ‘ZSTD_DCtx_reset’ was not declared in this scope; did you mean ‘ZSTD_DCtx_s’?
  180 |                 err = ZSTD_DCtx_reset(zstd_ctx, ZSTD_reset_session_only);
      |                       ^~~~~~~~~~~~~~~
      |                       ZSTD_DCtx_s
/home/orangepi/satdump/src-core/common/ziq.cpp: In member function ‘int ziq::ziq_reader::decompress_at_least(int)’:
/home/orangepi/satdump/src-core/common/ziq.cpp:221:47: error: ‘ZSTD_reset_session_only’ was not declared in this scope
  221 |                     ZSTD_DCtx_reset(zstd_ctx, ZSTD_reset_session_only);
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~
/home/orangepi/satdump/src-core/common/ziq.cpp:221:21: error: ‘ZSTD_DCtx_reset’ was not declared in this scope; did you mean ‘ZSTD_DCtx_s’?
  221 |                     ZSTD_DCtx_reset(zstd_ctx, ZSTD_reset_session_only);
      |                     ^~~~~~~~~~~~~~~
      |                     ZSTD_DCtx_s
src-core/CMakeFiles/satdump_core.dir/build.make:2329: recipe for target 'src-core/CMakeFiles/satdump_core.dir/common/ziq.cpp.o' failed
make[2]: *** [src-core/CMakeFiles/satdump_core.dir/common/ziq.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:1131: recipe for target 'src-core/CMakeFiles/satdump_core.dir/all' failed
make[1]: *** [src-core/CMakeFiles/satdump_core.dir/all] Error 2
Makefile:155: recipe for target 'all' failed
make: *** [all] Error 2

Aang23, 2023-10-01:
You should update "libzstd" :
https://github.com/facebook/zstd

JimmyeJones, 2023-10-01:
I updated libzstd, however I am still getting the same error

JVital2013, 2023-10-03:
The current version of libzstd is 1.5.5.

I'm guessing you tried updating it for APT - the version of zstd in Ubuntu 18.04's apt repos is too old, so you'll need to remove it from apt and compile it yourself.

Alternatively, you can just uninstall libzstd. SatDump will compile without it - you just won't be able to record/play ZIQ files. If you don't know what this is, you're probably OK without it 😄