SatDump/SatDump #436: The new Indian satellite INSAT-3DS

sergsero, 2023-10-06:
In the near future an ISRO GSLV-II rocket will launch the new Indian satellite INSAT-3DS . The launch date is currently targeted for November 01, 2023 (UTC).

There is no file for earlier Indian INSAT-3DR among the pipelines. 

Is this due to some circumstances of decoding signals from these satellites, besides the fact that the downlink transmission is in the extended C-band, near 4.78 GHz?

Aang23, 2023-10-06:
The reason is just that nodody has ever provided data I could work off to implement a decoder. If you are able to receive it, I don't see why it would not be possible though.

Raw data and HRIT should be on C-Band.

sergsero, 2023-10-07:
I tried to receive signals from INSAT-3DS at frequencies of 4798 MHz (SOUNDER) and 4781 MHz (IMAGER).

My LNB has LO=5950 MHz, so the reception frequencies were 1152.465 MHz and 1169.465 MHz.

I tried the HRIT processing options from different satellites  for "IMAGER" and some SNR values were displayed for "GK-2A HRIT".
For "SOUNDER" some values were obtained in the mode from "GOES-N Sounder SD".

[image]

[image]

[image]

I recorded the signals in s8 format. I don't know if these records will be useful for development.

SOUNDER https://drive.google.com/file/d/1A0y2enabf22Sp11pIpt0cJT38xNkFX-V/view?usp=sharing
IMAGER https://drive.google.com/file/d/1A-5IvtMwsm9I_UstJBTVVdYGSAQEBGsk/view?usp=sharing
IMAGER https://drive.google.com/file/d/1A1FGtM0wI39WZAbmrwXs9wMSjucTQv2j/view?usp=sharing
IMAGER https://drive.google.com/file/d/1A1XU19XjT5eD1C9OEwcSli_46d_BKdaN/view?usp=sharing

Unfortunately, the SNR value of 5-6 dB for "HRIT" seems to be the maximum that can be obtained on the my setup for this satellite.
I hope this will prove useful.

sergsero, 2023-10-08:
Today I checked the signal receiving from Insat-3ds again after accurate guidance of antenna on the target satellite and it was slightly better by 1-2 dB, but still does not reach SNR 7 dB on the GK-2A HRIT scale.

sergsero, 2023-10-17:
Launch Overview: Liftoff Time - November 01, 2023 +07 (00:00 UTC).

Mission Overview:
INSAT-3DS is an Indian weather satellite built by the Indian Space Research Organization and operated by ISRO as part of the Indian National Satellite System. It provides meteorological services to India using a 6-channel imager and a 19-channel sounder, as well as search and rescue information and message relay for terrestrial data collection platforms. The satellite will be a follow-up to INSAT-3DR.

sergsero, 2023-11-12:
There are two different modes in the transmission. Probably one of its contains data.

[image]

[image]

sergsero, 2023-12-17:
I checked if there is still signal reception from Insat-3dr and made two .s8 recordings of 6 MSpS samplerate. I still hope that they will prove useful and that someday the possibility of decoding the will be implemented.

https://drive.google.com/file/d/1oNGbgWKL_W6zB6ZN3GYbgYTBzqnCqxt6/view?usp=drive_link
https://drive.google.com/file/d/1oSab218smbWtqWv3vALb8FlUfTOOa1_L/view?usp=drive_link

[image]

Aybarnur, 2023-12-18:
I also recorded the baseband from insat3dr via the same C-band antenna LNB LO=5.95 GHz (5950-4780.5=1169.5):
2023-12-18_08-46-20_6000000SPS_1169500000Hz.s8 (~6.55GB).
https://drive.google.com/file/d/1gIcR4isM5dz5efmReq69hhhkVPJh2sxE/view?usp=drive_link

To avoid DC-spike, the receiver's tune was 50 kHz lower in frequency from the signal carrier.
[image]

For these gain's values, the constellation does not yet become square.
[image]

Aybarnur, 2023-12-21:
INSAT-3DS - 4th orbital launch attempt
Launch Time: Wed Jan 10, 2024 18:30 GMT+7

sergsero, 2024-02-17:
[image]

Aybarnur, 2024-02-18:
Insat-3DR satellite Imager:
insat3dr_20240218_043551_10MSPS.s16.tgz  (8479MB)
https://drive.google.com/file/d/1SOd0SFbI2kHCn1afftxhR2GRpgXUUDOT/view?usp=drive_link

insat3dr_20240218_044126_10MSPS.s16.tgz (1680MB)
https://drive.google.com/file/d/1aS59t60r2fc2DwqeUJMBSxyRzBOE6GvM/view?usp=drive_link

Aybarnur, 2024-02-20:
12 MSPS .soft
1 hour and 20 minutes. The file size is 34 GB:
https://drive.google.com/file/d/1_7SHpzMDoAFytrDFS4h1FG5P1rHb5l-g/view?usp=drive_link 

1 hour and 50 minutes. The file size is 46 GB:
https://drive.google.com/file/d/1wLAYyKCyMui8O5k_FPr11k8COWD7cfwq/view?usp=drive_link

Aang23, 2024-02-20:
@sergsero it is likely 3DS will use some slightly newer protocols if it follows the general trend of Indian missions.

Bosslike65, 2024-11-02:
> I tried to receive signals from INSAT-3DS at frequencies of 4798 MHz (SOUNDER) and 4781 MHz (IMAGER).
> 
> My LNB has LO=5950 MHz, so the reception frequencies were 1152.465 MHz and 1169.465 MHz.
> 
> I tried the HRIT processing options from different satellites for "IMAGER" and some SNR values were displayed for "GK-2A HRIT". For "SOUNDER" some values were obtained in the mode from "GOES-N Sounder SD".
> 
> ![2023-10-07_15-46-31_Insat-3dr_4798MHz_sounder_1MSpS_s8](https://private-user-images.githubusercontent.com/4005442/273386924-3f815c17-e9e7-4a6c-a875-4a8fa88120fe.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA1NzQ2NTIsIm5iZiI6MTczMDU3NDM1MiwicGF0aCI6Ii80MDA1NDQyLzI3MzM4NjkyNC0zZjgxNWMxNy1lOWU3LTRhNmMtYTg3NS00YThmYTg4MTIwZmUuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEwMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMDJUMTkwNTUyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MmFjYjc1ZmNjNDM0YTkxMTllYWE3YmZlYjc5MGNhODlmYzA5OGU1M2MxNmRiYjg3MTkzZDUzMGVlYzIwZjdhNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.WmvB200mEgsBuq_TtXnyBwisk25zRyeKmamA1a-G8e8)
> 
> ![2023-10-07_16-10-59_Insat-3dr_4781MHz_imager_10MSpS_s8](https://private-user-images.githubusercontent.com/4005442/273386932-a3921efc-689e-48d6-8b09-ec44e3a323fa.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA1NzQ2NTIsIm5iZiI6MTczMDU3NDM1MiwicGF0aCI6Ii80MDA1NDQyLzI3MzM4NjkzMi1hMzkyMWVmYy02ODllLTQ4ZDYtOGIwOS1lYzQ0ZTNhMzIzZmEuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEwMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMDJUMTkwNTUyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTZjNjJhMzY1ODUxZmRiYTg3YTIyMGExMWI3OWY2ODU2YTg5MDJiMzc3ODlmOThjYjQ1YzEyOGI0NzZkMTllMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.rt0EcHe2VwyzBsk9tBcqYCyN91a97HH4TP-1yyyK4B4)
> 
> ![2023-10-07_16-15-03_Insat-3dr_4781MHz_imager_10MSpS_s8](https://private-user-images.githubusercontent.com/4005442/273386941-6e212a29-1b5a-40b5-868b-eda371dbdb8f.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA1NzQ2NTIsIm5iZiI6MTczMDU3NDM1MiwicGF0aCI6Ii80MDA1NDQyLzI3MzM4Njk0MS02ZTIxMmEyOS0xYjVhLTQwYjUtODY4Yi1lZGEzNzFkYmRiOGYuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEwMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMDJUMTkwNTUyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YWZlMjVkMGJkMTdkOThlMDYyNTRjM2RkZWNhMDkxZDRkYWUyZmNlNTIzMjk4MDAzOTA1OWZiZDA3OGY0N2JmNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.vflK_AowcW7bKUq8SS9NXe7rJtQOtlhDN5tnIcFFQqY)
> 
> I recorded the signals in s8 format. I don't know if these records will be useful for development.
> 
> SOUNDER https://drive.google.com/file/d/1A0y2enabf22Sp11pIpt0cJT38xNkFX-V/view?usp=sharing IMAGER https://drive.google.com/file/d/1A-5IvtMwsm9I_UstJBTVVdYGSAQEBGsk/view?usp=sharing IMAGER https://drive.google.com/file/d/1A1FGtM0wI39WZAbmrwXs9wMSjucTQv2j/view?usp=sharing IMAGER https://drive.google.com/file/d/1A1XU19XjT5eD1C9OEwcSli_46d_BKdaN/view?usp=sharing
> 
> Unfortunately, the SNR value of 5-6 dB for "HRIT" seems to be the maximum that can be obtained on the my setup for this satellite. I hope this will prove useful.

Hey, if you see this, could you maybe upload the files once again? or record new ones? Let me know if you could do that, thanks.

sergsero, 2024-11-03:
Hi, only two files remain from the recording made in April 2024:

https://drive.google.com/file/d/11MKYqE_4y1SRpUWjXYoz-OqnaaWI2WWa/view?usp=sharing 
https://drive.google.com/file/d/16ce4GhC4qH3ffcPucY5nUzJ9PN-hvzZZ/view?usp=sharing 

As Alan informed us, this satellite employs a distinct data format that cannot be decoded, which is why the receiving C-band antenna has been removed.
However, I suspect that the issue lies in the poor reception of data due to the limited size of the 2-meter antenna.

juliasatt, 2024-11-03:
> Hi, only two files remain from the recording made in April 2024:
>
> https://drive.google.com/file/d/11MKYqE_4y1SRpUWjXYoz-OqnaaWI2WWa/view?usp=sharing [...]

Yea, I'm finishing up processing the bigger recording but the problem seems to be the SNR (at least for those 2 recordings).

I'm not sure about the data format tho, if Alan said it's not possible then it's probably something cursed, but even then I'd like to try out the other recordings that had more SNR, so hopefully @Aybarnur might still have it.

From one document that explains the format briefly it seems like it should be simple, but again, with Alan already looking into it my hopes aren't that high.

sergsero, 2024-11-04:
I think the main issue was a low-quality recording with a low signal-to-noise ratio. This prevented Alan from decoding it.
If your recordings have a better ratio, I hope for success in the C-band.

Bosslike65, 2024-11-25:
> I think the main issue was a low-quality recording with a low signal-to-noise ratio. This prevented Alan from decoding it. If your recordings have a better ratio, I hope for success in the C-band.

From one of the ealier recordings, it looks like the signal was strong enough, but yeah... Those two were too weak... If you stumble across the ealier recordings, let us know... Saddly, @Aybarnur hasnt replied yet.