SatDump/SatDump #439: SatDump 1.1.0 crashes when loading

afernan10, 2023-10-08:
**Description of the issue**
SatDump 1.1.0 crashes when loading TLEs on Windows 10
Same problem with official release and previous releases

**Hardware (SDR/PC/OS)**
RSP1A/Lenovo Legion Core i7 Geforce RTX/Windows 10

**Version (Eg, 1.0.0, CI Build 171)**
1.1.0 

**Logs after the crash (satdump.log)**
file satdump.log not written 

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
[image]

JVital2013, 2023-10-08:
This is a known bug with the 1.1.0 release. It has been fixed in the nightlies. Please use a nightly build from the releases page to fix your issue.

afernan10, 2023-10-08:
Thanks so much! I tried one of the nightlies  with no success, will try again and report back

afernan10, 2023-10-08:
In fact my report was one the nightlies

JVital2013, 2023-10-08:
In that case, can we have the log file? It should be in %appdata%\satdump\satdump.log.

JVital2013, 2023-10-08:
If there's no log file in appdata, try uninstalling satdump and reinstall it in the default install location.

It looks like you're using the installer, but installing it in a non-standard location. That should work, but if no log file is created and it's crashing, installing it in the default location is worth trying.

afernan10, 2023-10-08:
I am attaching the satdump.log file. THANKS! for your help
[satdump.log](https://github.com/SatDump/SatDump/files/12841794/satdump.log)

afernan10, 2023-10-08:
> If there's no log file in appdata, try uninstalling satdump and reinstall it in the default install location.
>
> It looks like you're using the installer, but installing it in a non-standard location. [...]

will try installing it in the default location

afernan10, 2023-10-08:
> will try installing it in the default location

I uninstalled and then installed the nightly version on the default location as per the installer program C:/program files/satdump. It keeps crashing with exactly the same symptoms

afernan10, 2023-10-08:
I found that the tles file name is satdump_tles.txt.tmp the I renamed it to satdump_tles.txt with no tmp extension. Now the program load the tles but keep crashing after loading. I attach the new log file 
[satdump.log](https://github.com/SatDump/SatDump/files/12841821/satdump.log)

JVital2013, 2023-10-08:
Thanks for the logs! Also thanks for trying a reinstall - it was worth a shot, but I don't think it's causing your problem. It should work fine wheee you installed it.

**Next thing to try:** can you remove %appdata%\satdump\satdump_tles.txt if it's there, then launch SatDump again? I wonder if the file is corrupted in some way. Keep a copy of the file somewhere else - if this fixes it, I would like to get a copy for troubleshooting.

afernan10, 2023-10-09:
> Thanks for the logs! Also thanks for trying a reinstall - it was worth a shot, but I don't think it's causing your problem. It should work fine where you installed it.
>
Thanks so much, Jamie! your help is most appreciated!
 
> **Next thing to try:** can you remove %appdata%\satdump\satdump_tles.txt if it's there, then launch SatDump again? I wonder if the file is corrupted in some way. Keep a copy of the file somewhere else - if this fixes it, I would like to get a copy for troubleshooting.

I moved the file to another location and SatDump crashed after the 0 TLEs loaded message :(
It is really weird.

afernan10, 2023-10-10:
TLEs loading OK now! but Crashing after. I attach the log file. More testing tomorrow
[satdump.log](https://github.com/SatDump/SatDump/files/12862311/satdump.log)

JVital2013, 2023-10-11:
The output doesn't look right when you ran `.\satdump sdr_probe` - the error you listed is normal, but it should list several SDR sources after that. Was the error you put there the only thing it output? If so, I think it's crashing the UI too when probing SDRs.

What version of the SDRPlay API do you have? Also, is your Microsoft Visual C++ Runtime up-to-date? It never hurts to run the installer again from https://aka.ms/vs/17/release/vc_redist.x64.exe to update it if you're not sure.

afernan10, 2023-10-11:
> The output doesn't look right when you ran `.\satdump sdr_probe` - the error you listed is normal, but it should list several SDR sources after that. Was the error you put there the only thing it output? If so, I think it's crashing the UI too when probing SDRs.
yes, that was all. I suspect also the UI crash
> 
> What version of the SDRPlay API do you have? Also, is your Microsoft Visual C++ Runtime up-to-date? It never hurts to run the installer again from https://aka.ms/vs/17/release/vc_redist.x64.exe to update it if you're not sure.

SDRPlay API is ver 2.13 released on 2018/06/20

I have updated also the Microsoft Visual C++ Runtime. After updating, I was able to briefly run SatDump once, then crashed (screen capture attached). It no longer works, keeps crashing again :( 
[image]

JVital2013, 2023-10-11:
If possible, try uninstalling the SDRPlay driver and install version 3.09 from https://www.sdrplay.com/api/ - you can try newer versions as well, but I'm on 3.09 so I know that one works 😄.

afernan10, 2023-10-11:
> If possible, try uninstalling the SDRPlay driver and install version 3.09 from https://www.sdrplay.com/api/ - you can try newer versions as well, but I'm on 3.09 so I know that one works 😄.

Thanks so much Jamie! I updated the SDRPlay driver/API to 3.09, reinstalled SatDump nightly build, but keeps crashing on this machine. Really, I do not want to bother you. Close to giving-up :(

JVital2013, 2023-10-11:
Any ideas @Aang23? I'm a bit baffled. It's especially confusing since it looks like it started to work one time!

dan6114, 2023-10-16:
I'm having the same issues here. I also tried running the portable version with some success but it crashed while I was in it then kept crashing on boot. If I keep unzipping and running it I get the same results, runs for a bit then stops. I'll attach my log file as well. Satdump had been working for me throughout the week but suddenly stopped today. Thanks in advance!

[satdump.log](https://github.com/SatDump/SatDump/files/12911946/satdump.log)

JVital2013, 2023-10-16:
@dan6114 just to confirm: have you tried the latest 1.1.1 nightly build? There was a crash in 1.1.0 that has been fixed, and @afernan10 is experiencing a different crash we haven't figured out yet.

Just want to make sure you're not experiencing the since-fixed bug!

dan6114, 2023-10-16:
Correct I have tried it. I've gone through uninstalling Satdump as well as deleting %appdata%/Satdump and trying fresh installs with both the nightly and stable builds.  It's able to open once after a fresh install briefly then close out and crash on boot. I apparently did this too much last night as now I'm IP blocked from celestrak for updating TLEs too much.

JVital2013, 2023-10-18:
Could either of you try running as administrator to see if that fixes the issue? We'd like to rule out any potential permissions issues

dan6114, 2023-10-18:
Ok so today I deleted all the files again and reinstalled the stable build again and now it's working for me. But I will say when it was crashing I did try running as an administrator and it did nothing. I also got temporarily IP banned from celestrak for updating tle too much while it was crashing a lot. I'll let you know if you it does start happening again.

afernan10, 2023-10-18:
trying to catch-up here, been busy recently, will try newest nigthly build and will report back 

> @dan6114 just to confirm: have you tried the latest 1.1.1 nightly build? There was a crash in 1.1.0 that has been fixed, and @afernan10 is experiencing a different crash we haven't figured out yet.
> 
> Just want to make sure you're not experiencing the since-fixed bug!

trying to catch-up here, been busy recently, will try newest nightly build running as admin and will report back

JVital2013, 2023-10-18:
Sounds good - thanks! If it does crash, we'd like to see the satdump_tles file from right after the crash in case there's useful info in there.

afernan10, 2023-10-18:
[image]
No luck, latest nightly build, downloaded a few minutes ago, crashes at the point shown in the screen capture :(

JVital2013, 2023-10-18:
And the TLEs file?

JVital2013, 2023-10-18:
Also, are you able to load this page: http://celestrak.org/NORAD/elements/active.txt

JVital2013, 2023-10-20:
We pushed some changes to TLE loading. I'm not 100% sure your problem is TLE related - but if it is, it might be worth trying the latest nightly.

JVital2013, 2023-10-30:
There have been a number of changes since we last touched base: if you're still interested, can you try updating and let us know what happens?

If we don't hear back within the next few weeks, we'll close this issue since this issue cannot be preproduced on other systems. As long as you're willing to troubleshoot, though, we'll keep it open. Thanks!

afernan10, 2023-10-30:
Thanks so much, Jamie!

Yes, I will try latest release and will report back!

I assume you are referring to the latest nightly build.

cheers,

a

> There have been a number of changes since we last touched base: if 
> you're still interested, can you try updating and let us know what 
> happens?
> 
> If we don't hear back within the next few weeks, we'll close this issue 
> since this issue cannot be preproduced on other systems. As long as 
> you're willing to troubleshoot, though, we'll keep it open. Thanks!
> 
> [2].

Links:
------
[1] 
https://github.com/SatDump/SatDump/issues/439#issuecomment-1785336313
[2] 
https://github.com/
--=_0a6b4254959ceccf76ca44032ab6c355
Content-Transfer-Encoding: quoted-printable
Content-Type: text/html; charset=UTF-8

<html><head><meta http-equiv=3D"Content-Type" content=3D"text/html; charset=
=3DUTF-8" /></head><body style=3D'font-size: 10pt; font-family: Verdana,Gen=
eva,sans-serif'>
<p>Thanks so much, Jamie!</p>
<p>Yes, I will try latest release and will report back!</p>
<p>I assume you are referring to the latest nightly build.</p>
<p>cheers,</p>
<p>a</p>
<p id=3D"reply-intro">El 2023-10-30 14:27, Jamie Vital escribi&oacute;:</p>
<blockquote type=3D"cite" style=3D"padding: 0 0.4em; border-left: #1010ff 2=
px solid; margin: 0">
<div id=3D"replybody1">
<p><br /></p>
<p dir=3D"auto">There have been a number of changes since we last touched b=
ase: if you're still interested, can you try updating and let us know what =
happens?</p>
<p dir=3D"auto">If we don't hear back within the next few weeks, we'll clos=
e this issue since this issue cannot be preproduced on other systems. As lo=
ng as you're willing to troubleshoot, though, we'll keep it open. Thanks!</=
p>
<p style=3D"font-size: small; -webkit-text-size-adjust: none; color: #666;"=
atDump/SatDump/issues/439#issuecomment-1785336313" target=3D"_blank" rel=3D=
"noopener noreferrer">view it on GitHub</a>, or <a href=3D"https://github.c=
AA5XTPVDWVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMYTOOBVGMZTMMZRGM" target=
=3D"_blank" rel=3D"noopener noreferrer">unsubscribe</a>.<br />You are recei=
ving this because you were mentioned.<img src=3D"https://github.com/notific=
ations/beacon/AHUJZKKFHEGTZYKTQCL2PALYB62NDA5CNFSM6AAAAAA5XTPVDWWGG33NNVSW4=
5C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTKNII7S.gif" width=3D"1" h=
eight=3D"1" /><span style=3D"color: transparent; font-size: 0; display: non=
e; visibility: hidden; overflow: hidden; opacity: 0; width: 0px; height: 0p=
mp/SatDump/issues/439/1785336313</span><span>@</span><span>github</span><sp=
an>.</span><span>com&gt;</span></span></p>
</div>
</blockquote>
</body></html>

--=_0a6b4254959ceccf76ca44032ab6c355--

JVital2013, 2023-10-30:
Yes, latest nightly would be best. Thanks!

afernan10, 2023-10-30:
Just tried 1.1.1 version and did some offline decoding, as I am away from home and have no chance to do a live test till next week. However, everything seems to be fine! :) 
Thanks so much!!!
[image]

JVital2013, 2023-10-30:
That's great! I think I'll close this since it worked for you, but if you have any issues with live, feel free to open an issue about it then. Since it works offline though, I'd say you should be good to go :-)

closed 2023-10-30 by JVital2013