SatDump/SatDump #440: Adjust when to stop decoding NOAA-APT

Akito01, 2023-10-10:
Satdump stops decoding an NOAA-APT wav file too soon when it hits noise.
It would be great if Satdump would check all of the file to see if later on an image appears again.

For example sometimes it hits a bad spots in reception which will cause a few seconds of total noise after which the signal comes back again. Satdump doesn't notice it properly and as such will too quick stop building a picture out of it missing a part.

Maybe during the build process, or in a config file, 2 settings could be set and applied:
1 - The amount of noise or SNR after which it is considered actual noise and stop decoding
2 - An amount of time in seconds where the noise, defined in param 1, should apply before it's considered the end of the decoding progress.

Aang23, 2023-10-10:
What you're saying makes little sense. It's fully analog, and unless you check the "Autocrop Wedges" option, SatDump will process everything, noise or not. Do you have an example on hand?

Akito01, 2023-10-10:
Yes I do, here's 2 links.
This one is decoded using Satdump with the autocrop wedges turned on: https://i.imgur.com/FbHcSpZ.png
Here is the same capture decoded by wxtoimg: https://i.imgur.com/4PFSGoA.png

You can see that in this capture wxtoimg continues decoding after an x-amount of noisier lines where Satdump would stop.
It also looks like wxtoimg despeckles differently.

Akito01, 2023-10-10:
Another example. These were made at a very low elevation <7 .
Satdump: https://i.imgur.com/kgePf35.png
Wxtoimg: https://i.imgur.com/c1TUeZ2.png

JVital2013, 2024-01-15:
Hi @Akito01,

We have made a number of updates to the NOAA APT decoder since this issue was opened. Could you try the latest release (or better - nightly!) build to see if it's still autocropping too aggressively?

Akito01, 2024-01-15:
It seems like it's gotten better, but Wxtoimg still creates 'more image' out of the same WAV-file.

JVital2013, 2024-03-06:
Hello again @Akito01,

I just pushed an update that completely revamps the autocrop logic in SatDump. It's now more willing to save images with a bit of noise like the example you shared, and it can be tuned by editing a config file to allow more or less noise through.

If you're still interested, can you try the latest nightly at https://github.com/SatDump/SatDump/releases and let us know your thoughts?

Akito01, 2024-06-11:
Can you tell me the name of the config file where to change the settings regarding noise?

JVital2013, 2024-11-28:
In `pipelines\NOAA.json`, it is `max_crop_stddev` near the bottom, under `noaa_apt`: https://github.com/SatDump/SatDump/blob/4ee04d09ed0cfd8fcaeed55406b53506e84b0ecd/pipelines/NOAA.json#L261

You can read about it here: https://docs.satdump.org/pipelines.html#noaa.

I'm going to close this since this has worked for folks I talked to, but let us know if there are issues!

closed 2024-11-28 by JVital2013