SatDump/SatDump #443: GUI showing as blank, no apparent crash

TravisKG, 2023-10-13:
**Description of the issue**
SatDump window loads, but GUI is a black screen.

**Hardware (SDR/PC/OS)**
Ubuntu 22.04.3 LTS
Intel Xeon E5-2670 2.60 GHz x2
128GB RAM

Tried with:
RTL SDR V3
Nooelec Smartee XTR

**Version (Eg, 1.0.0, CI Build 171)**
SatDump GUI v1.1.1-08ccf68

**Logs after the crash (satdump.log)**
[satdump.txt](https://github.com/SatDump/SatDump/files/12900573/satdump.txt)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
[image]

JVital2013, 2023-10-14:
Hello! It looks like this might be a graphics driver problem. What graphics card and driver are you using?

TravisKG, 2023-10-14:
It would be whatever's integrated with the motherboard, which is apparently a Matrox Electronics MGA G200EH. 
[image]

JVital2013, 2023-10-15:
Have you confirmed OpenGL is working? SatDump requires working OpenGL. Please run `sudo apt install mesa-utils` and run `glxinfo` and report its output. You can also try running `glxgears` to see if the gears are rendered properly.

You many need to figure out how to get software OpenGL rendering to work on that machine: see below for a brief deep-dive I did.

Matrox Electronics MGA G200 is an extremely old chip from 1998. Various server manufacturers still use them since Xeon processors do not have an iGPU, and they need a way to do video output. Specific manufacturers have kept the driver up-to-date for modern versions of Windows [here](https://internal.support.hpe.com/connect/s/search?language=en_US#q=Matrox%20G200eH&t=DriversandSoftware&sort=relevancy&numberOfResults=25), but I can't tell if there's a modern driver for Ubuntu, let alone one with OpenGL support. There is some info [here](https://askubuntu.com/questions/1231887/20-04-lts-boot-hang-matrox-g200ew-after-successful-install). There is also info at https://forums.linuxmint.com/viewtopic.php?t=387751, which is for a chip in the same family - but seems to report there is no 3D acceleration available.

TravisKG, 2023-10-15:
Thanks for the suggestions.

Here was the output from glxinfo after installing mesa-utils:
[glxinfo.txt](https://github.com/SatDump/SatDump/files/12910342/glxinfo.txt)

glxgears also seems to work:
[image]

RAD750, 2023-10-16:
Try launching satdump as follows:
`MESA_GL_VERSION_OVERRIDE=4.5 satdump-ui`.
Should fix it.

The Matrox MGA G200 has very limited OpenGL support.

TravisKG, 2023-10-16:
That didn't seem to do the trick either unfortunately. I suppose there's just the option of sticking to the CLI version.

JVital2013, 2023-10-17:
Yes, the CLI version is certainly an option. We can provide assistance in constructing the command if you need it!

It does look like something is going on here that should be fixed with the UI version - or, at least, have a workaround found. If you have time, could you please try running satdump with this command?

`LIBGL_ALWAYS_SOFTWARE=1 satdump-ui`

This should force it to use software rendering, at the cost of high CPU usage. Although, it looks like it may already be doing that - so if it doesn't work, try this:

`LIBGL_DEBUG=verbose satdump-ui`

This will output extra OpenGL debugging information to the console. Some warnings in this mode are normal, but it can still give useful information. It won't show in the log file, but above/within the satdump log in the console.

TravisKG, 2023-10-17:
This is what I'm using in the CLI, the program starts but I can't get the Deframer to sync. As a sanity check I tested the GUI version on my laptop to confirm it worked, then tried the CLI on my laptop as well with the same command and no luck getting the Deframer to sync.
> satdump live goes_hrit output_test --source rtlsdr --samplerate 2.4e6 --frequency 1694.1 --bias --gain 35

As for the original problem, I tried starting with LIBGL_ALWAYS_SOFTWARE=1 satdump-ui and ended up with the same black screen.

Here's the what showed up in console when launching with the second suggested command.
[satdump.txt](https://github.com/SatDump/SatDump/files/12955326/satdump.txt)

Not sure if it's worth mentioning but I was having problems with my machine about a week prior to installing SatDump. It was just an HP firmware bug where sometimes it couldn't see storage on reboot, but it happened during an update and I had to repair Grub. Again. Not sure if that might be related but I figure more context doesn't hurt.

JVital2013, 2023-10-18:
Starting with the CLI: it looks like the you're missing the `e6` after the frequency. The command should look like this:

```bash
satdump live goes_hrit output_test --source rtlsdr --samplerate 2.4e6 --frequency 1694.1e6 --bias --gain 35
```

Back to the GUI: all those logs look fine. You're using the software OpenGL renderer anyway per the logs, but this is supported. I'm guessing there's something specific about your chip that's causing trouble. I'll let this for other SatDump devs to take a look at, but it may be difficult to troubleshoot without access to a similar machine.

JVital2013, 2023-11-06:
@TravisKG we just reworked OpenGL loading. Would you have time to pull the latest commit, rebuild, and see if it works for you?

If it doesn't work right away, try `LIBGL_ALWAYS_SOFTWARE=1 satdump-ui` again to see if that makes a difference.

TravisKG, 2023-11-07:
Thanks for your assistance. Unfortunately still having the same trouble with the GUI on the latest build. The CLI version is now working though.

JVital2013, 2023-11-07:
I'm glad to hear the CLI version is working for you, but this GUI glitch is an odd one. I probably should have asked this before, but: are you remotely connecting to this machine, or is the display directly connected?

Now that you've updated, there's one more startup method you can try: `MESA_GL_VERSION_OVERRIDE=2.1 satdump-ui` (note the override version is different from the one we recommended before). This will force SatDump to use a new (old?) built-in OpenGL rendering engine that's more basic.

TravisKG, 2023-11-08:
I have a monitor attached via VGA but occasionally connect remotely using VNC viewer.

Launching with `MESA_GL_VERSION_OVERRIDE=2.1 satdump-ui` it still doesn't work. Though it shows a completely grey screen instead of a black screen.

JVital2013, 2023-11-08:
Interesting - I'm surprised such a basic rendered doesn't work.

If you have some time, there's a piece of software called apitrace (https://github.com/apitrace/apitrace) that might help here. If you compile apitrace, you can then generate a record of what satdump-ui does when you try to launch it. I'd do it like this:

`MESA_GL_VERSION_OVERRIDE=2.1 ./apitrace trace satdump-ui`

Let it run for a few seconds, then ctrl-c it. We would be interested in both the .trace file that is created, and the SatDump log from the console. The console will have some more information in it, like this (this is from my system which works):

```
apitrace: loaded into /usr/bin/satdump-ui
[20:07:09 - 08/11/2023] (E) Usage : satdump-ui [downlink] [input_level] [input_file] [output_file_or_directory] [additional options as required]
[20:07:09 - 08/11/2023] (E) Extra options (examples. Any parameter used in modules can be used here) :
[20:07:09 - 08/11/2023] (E)  --samplerate [baseband_samplerate] --baseband_format [f32/i16/i8/w8] --dc_block --iq_swap
apitrace: redirecting dlopen("libGL.so.1", 0x1) from /lib/x86_64-linux-gnu/libglfw.so.3
apitrace: tracing to /home/jamie/Programs/apitrace/build/satdump-ui.trace
[20:07:09 - 08/11/2023] (E) Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig
[20:07:09 - 08/11/2023] (W) Could not init GLFW Window; falling back to OpenGL 2.1...
[20:07:09 - 08/11/2023] (D) Starting with OpenGL 2.1 Mesa 23.0.4-0ubuntu1~22.04.1
apitrace: warning: glVertexPointer: call will be faked due to pointer to user memory (https://git.io/JOMRv)
apitrace: warning: glTexCoordPointer: call will be faked due to pointer to user memory (https://git.io/JOMRv)
apitrace: warning: glColorPointer: call will be faked due to pointer to user memory (https://git.io/JOMRv)
[20:07:09 - 08/11/2023] (I)    _____       __  ____                      
[20:07:09 - 08/11/2023] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ 
[20:07:09 - 08/11/2023] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[20:07:09 - 08/11/2023] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[20:07:09 - 08/11/2023] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ 
[20:07:09 - 08/11/2023] (I)                                     /_/      
[20:07:09 - 08/11/2023] (I) Starting SatDump v1.1.2-7e7d28cb
...
```

JVital2013, 2023-11-09:
Quick add-on: I got access to a system running a similar chip. It's running Windows, but SatDump works fine on it once Software OpenGL is enabled - it's an extra DLL on the Windows side.

I'm suspicious there's some very specific interaction between your hardware, Mesa on Ubuntu, and OpenGL Immediate Mode. 

[image]

TravisKG, 2023-11-13:
I'll give that a shot when I have a second this week.

TravisKG, 2023-12-06:
Sorry about the long delay. I haven't been able to get apitrace figured out because I'm a bit of a noob with it. I have it compiled and installed but I haven't been able to get it to do a trace.

JVital2013, 2023-12-15:
And in turn, sorry for my delay!

**To record a trace attemping OpenGL 3+, simply run this command in a terminal:**
```bash
apitrace trace satdump-ui
```

**To record a trace attempting OpenGL 2.1, run:**
```bash
MESA_GL_VERSION_OVERRIDE=2.1 apitrace trace satdump-ui
```

In either case, `satdump-ui.trace` will be saved in your current directory, which can then be shared here. Don't let SatDump run too long or it will get quite large!

JVital2013, 2024-01-15:
Hi @TravisKG,

Are you interested in pursuing this issue further? We'd like to help if you need it, but if not, I'm thinking we should close this issue since we have had no other reports of this.

TravisKG, 2024-01-15:
> Hi @TravisKG,
> 
> Are you interested in pursuing this issue further? We'd like to help if you need it, but if not, I'm thinking we should close this issue since we have had no other reports of this.

I have a lot on my plate at the moment and I do want to get around to figuring it out, but for the time being it's probably best to close the topic.

JVital2013, 2024-01-16:
OK, let us know if/when we can help, and we'll happily re-open this.

closed 2024-01-16 by JVital2013