SatDump/SatDump #447: Decoding baseband coming from stdin

luarvique, 2023-10-17:
I am somewhat lost trying to decode baseband data coming from the standard input. Using things like

`... | satdump noaa_apt baseband /dev/stdin /tmp --samplerate 50000 --baseband_format f32
`
does not appear to produce any result. Am I doing something wrong? Is it possible at all in the current SatDump?

luarvique, 2023-10-17:
(to clarify: the input data in this case is a 50kHz IQ stream made of interleaved F32s, it is not demodulated in any way)

JVital2013, 2023-10-18:
SatDump can handle input from stdin - I successfully decoded a METEOR HRPT baseband like this:

```bash
cat 2023-07-07_14-44-25_6000000SPS_1700000000Hz.s8 | satdump meteor_hrpt baseband /dev/stdin test --samplerate 6e6 --baseband_format s8
```

So, in theory, what you're trying to accomplish should work. What exactly are you seeing? My first guess is that your samplerate is very low - are you able to increase it at all?

luarvique, 2023-10-18:
Seeing broken progress messages (since stdin has no length):
```
Oct 17 18:41:50 orange openwebrx[771068]: [22:41:50 - 17/10/2023] (I) Progress 1844674407370955161600.000000%
Oct 17 18:41:50 orange openwebrx[771068]: [22:41:50 - 17/10/2023] (I) Progress nan%
Oct 17 18:42:00 orange openwebrx[771068]: [22:42:00 - 17/10/2023] (I) Progress 1844674407370955161600.000000%
Oct 17 18:42:00 orange openwebrx[771068]: [22:42:00 - 17/10/2023] (I) Progress nan%
```
Not seeing anything else. One important thing is that my baseband data stream is essentially endless (since it is live baseband), so "live" option sounds more fitting, but I do not see any command line options to make satdump consume stdin in the "live" mode.

As far as sample rate is concerned, I can make it higher, but since NOAA satellites use 50kHz bandwidth to transmit, will it really make any difference?

PS: Sorry for somewhat unusual usage mode: I am trying to integrate satdump inside OpenWebRX+, so that a user could click on a satellite transmission in their waterfall and immediately start receiving images.

luarvique, 2023-10-21:
A little update:

So far, we have been able to decode Meteor M2 data to the .cadu file by running SatDump from OpenWebRX. This has been achieved by running SatDump with the following command line:
`
satdump meteor_m2-x_lrpt baseband /dev/stdin /tmp --samplerate 150000 --baseband_format f32
`
NOAA decoding does not appear to work, at least when using the following command line:
`
satdump noaa_apt baseband /dev/stdin /tmp --samplerate 50000 --baseband_format f32 --satellite_number 19 --start_timestamp <current-utc-timestamp>
`
I have tried upping the sample rate to 150000, but it did not help.

In general, I have been wondering if it is going to be possible to run SatDump in the "live" mode while feeding it with samples via standard input? The above scenarios are clearly designed for offline use and that might be the reason why the NOAA pipeline does not work.

JVital2013, 2023-10-30:
**I've done some testing** - first, your samplerate of 50000 for APT should be fine, so my "first guess" sure was wrong :-).

**Here's some boilerplate that should get you going in the right direction.** First, you do want to switch to live mode. Here's a command to do that for APT:

```
satdump live noaa_apt /output/dir --source file --file_path /dev/stdin --samplerate 50e3 --frequency 137.9125e6 --satellite_number 18 --start_timestamp <current-utc-timestamp> --finish_processing
```

**And for METEOR:**
```
satdump live meteor_m2-x_lrpt /output/dir --source file --file_path /dev/stdin --samplerate 150e3 --frequency 137.9e6 --finish_processing
```

**Some notes:**

- Send SIGINT (Ctrl-C) to SatDump when the pass is over, and products will be spit out with the above commands
- The frequency parameter is required, but it does not need to be correct.
- The "finish processing" flag is important as well so products are produced. 

**Note that no wav is produced by default for APT:** if you want that, the NOAA pipeline should be modified. Edit pipelines\NOAA.JSON, and remove the `[2,0]` from within `live_cfg`

I'm thinking, with this info, you should be able to do what you need. Let us know!

luarvique, 2023-10-30:
Thanks a lot for the hints! One thing I want to ask is what input data format do your examples expect? Can I use "--baseband_format f32" to make it accept an IQ stream of f32 sample pairs?

JVital2013, 2023-10-31:
In my test, I fed it f32 and it just worked. It appears to be expecting f32 by default, so I forgot to include that argument😄.

The argument is `--baseband_type` though (different from the offline parameter), and it probably should be specified to be safe. Using the file input live from the cli isn't documented anywhere, but you can see the parameters in source here: 

https://github.com/SatDump/SatDump/blob/c0b784ca4277801512307309a6744e1154e7aba1/src-core/common/dsp_source_sink/file_source.cpp#L36-L39

So, the updated commands are, for APT:
```
satdump live noaa_apt /output/dir --source file --file_path /dev/stdin --samplerate 50e3 --baseband_type f32 --frequency 137.9125e6 --satellite_number 18 --start_timestamp <current-utc-timestamp> --finish_processing
```

And for LRPT:
```
satdump live meteor_m2-x_lrpt /output/dir --source file --file_path /dev/stdin --samplerate 150e3 --baseband_type f32 --frequency 137.9e6 --finish_processing
```

luarvique, 2023-10-31:
Acknowledged. Set things up, waiting for a satellite to pass now...

Aang23, 2023-11-05:
Is everything fine now?

luarvique, 2023-11-06:
Sorry for a bit of the delay, have been busy. So far, there are some good news and some bad news.

The good news is that Meteor decoder works as good as can be expected. Runs in the background, leaves .cadu files in the output folder, which can later be converted to images.

The bad news is that we just cannot receive anything from NOAA satellites. No matter what we do, there are no output files in the output folder. We seem to be doing something wrong.

After removing "[2,0]" from live_cfg, NOAA starts dumping what looks like unprocessed baseband .wav file to the output folder.

luarvique, 2023-11-06:
The SatDump NOAA decoder is called like this:

satdump live noaa_apt /tmp --source file --file-path /dev/stdin --samplerate 50000 --frequency 137000000 --baseband_format f32 --finish_processing --satellite_number 19 --start_timestamp <timestamp>

It is stopped with SIGTERM initially. Then we close pipes, wait 50x100ms for it to stop, and finally issue SIGKILL if it is still alive at that point.

JVital2013, 2023-11-06:
Ah, that's the problem - we weren't catching SIGTERM. I just made a change so it should work for you; please update and try again.

luarvique, 2023-11-07:
Aha! Pulled the latest code from master, rebuilt, reinstalled, waiting for a satellite to arrive (for myself or a friend of mine).

studentkra, 2023-11-07:
I don't think anything has changed. Only a cadu file is created for Meteor and nothing else happens. And what should be? Automatic decoding of images from this file?

JVital2013, 2023-11-07:
Looks like it's working right now from the log, but you're only giving it 5 seconds to terminate? That's not nearly enough time for it to do anything.

_After_ sending SIGTERM/SIGINT is when it actually creates products images - depending on the machine, this may take a few minutes. You can see in your log it starts processing but only gets to 0.1% before you kill it.

studentkra, 2023-11-07:
Thanks! Before you made changes to the SatDump code, we did not even receive messages about the start of decoding. Now we will work on our code. However, the issue with NOAA remains open for now. There may be a problem on our side.

JVital2013, 2023-11-07:
Yeah, nothing will show for APT for the first 20 seconds after stopping - everything is done in memory for the first bit.

Honestly, I'd completely remove the SIGKILL, see how long it takes for things to process, then put SIGKILL back in with a sensible limit.

studentkra, 2023-11-07:
I think I got it. Let's wait for **luarvique** to respond as developer of OpenWebRX+.

luarvique, 2023-11-07:
The SIGKILL thing is automatic and part of the ExecModule that is used for many decoders. The schedule for running SatDump can be adjusted to give it 5-10 additional minutes after a satellite passes. What exactly triggers the start of image processing in the NOAA decoder? Does it start working on an image once the signal disappears?

studentkra, 2023-11-07:
If I understand correctly, nothing in SatDump captures the disappearance of the signal. For NOAA APT, it will just be a huge wav file, and METEOR will add data to a cadu file.

JVital2013, 2023-11-07:
You are correct @studentkra: With NOAA, a wav of the entire session is recorded. The crop wedges option will crop it down to just the actual image. For METEOR, data is added to the CADU when the decoder has a lock on the signal.

Processing of the image is triggered when you send SIGTERM to SatDump.

luarvique, 2023-11-07:
I see the actual problem now. When OWRX sends a SIGTERM, it expects the decoder to terminate immediately. OWRX has no idea that the signal has a special meaning for SatDump, nor would it know when to send it, since it is just a machine, running decoders on schedule.

Can SatDump start processing the data once it does not see modulated baseband for a certain amount of time?

JVital2013, 2023-11-07:
Now that SatDump responds to SIGTERM, one thing you can do is to remove the `--finish-processing` flag from the call. This flag is explicitly asking SatDump to not exit as soon as possible. Removing it will prevent SatDump from producing all products, but SatDump will exit much more quickly. You can then use saved data for processing later.

It will need longer than 5 seconds, especially for APT on some systems. SatDump does a lot of its work in memory, and needs time to clean up/save everything to disk when you ask it to exit. Your problem can be solved by allowing SatDump to run for long enough after sending a SIGTERM.

Another thing you could do is specify the `--timeout` flag to specify how long the decoder should run for. I'm not sure how well this will work in your situation, since you need to know how long the decoder will run for when you start it.

Right now, SatDump does not have functionality to start and stop a decoder based on "seeing" modulated data in a baseband.

luarvique, 2023-11-08:
Understood. Thanks a lot for clearing up the horizon. Trying to figure out how this can be integrated with the regular workflow (e.g. receiver starts decoder app on schedule, lets it run for a given amount of time, then terminates it). A longer schedule slot + --timeout may actually do the job, but it gets hackier and hackier with each step.

I remember seeing a module within OWRX that essentially works in chunks, recording a piece of baseband, then starting an app to process it. Maybe **that one** will do a better job...

Aang23, 2023-11-24:
Another way that may fit the usecase better for OWRX would be to use SatDump's API to implement things, allowing much more integration and access to things the user may like to have shown back in the UI. However that's a bit more involved.

JVital2013, 2024-02-15:
Is this still a project you are pursuing? Do you need anything from us?

luarvique, 2024-02-15:
The project is still going on, with several people using OWRX+ and SatDump to receive satellite images. I am unable to figure out how to integrate this functionality for common use though, since a lot of processing happens offline. Either way, the ball is on my side of the court at the moment, there is nothing you can do there. Lots of thanks for all the help though, resulting in somewhat hacky, but usable solution.

JVital2013, 2024-02-15:
If you need anything from us in the future, don't be afraid to ask!