SatDump/SatDump #457: Automated the "Viewer"

NicolasGagne, 2023-10-29:
Frist I would like to say that I really appreciated the new Scheduling feature Great job. I also appreciated the new NOAA ATP enhancement (Similar to Wxtoimg) Great Job again. 

**Describe the feature you'd like to have**
I would like to suggest to automate the "Viewer" tab.  
It would be helpful that after the satellite pass is done all the request images variant are automatically generated and save. 

**Describe alternatives you've considered**
If that feature already exist I haven't found it. Can you point me in the right direction?
I have also consider using the CLI version to do this task with some kind of script but it seem that the "viewer" tab CLI is undocumented.

**Additional context**
Ultimately some kind of website generation would be nice.

I am happy to offer my help/time but I would need some pointer.

JVital2013, 2023-10-30:
Hello! This feature is already there. To turn it on, make sure "Automatically process products" is turned on in Settings:

[image]

**Not all products are automatically created to keep processing from taking forever.** To customize what gets created, take a look at the viewer->instruments->\[sensor\]->rgb_composites section of satdump_cfg.json. I'd recommend doing a Ctrl+F to search for the sensor you're looking to customize (ex, AVHRR) and edit the products underneath.

If a product does not specify `"autogen": false`, it will automatically be created. So, you can remove this line from any products you want to auto-create.

[image]

**If you want to automatically project a product,** it needs to have a `project` section within its config. Both non-projected and projected images will be saved when a projection config is provided. For an example, the AVHRR 221 projection will automatically be created with the default config, which looks like this:

[image]

Let us know if you have issues with this!

JVital2013, 2023-10-30:
Quick addition: doing viewer tasks/projection from the command line is a planned feature, so stay tuned for that!

NicolasGagne, 2023-10-31:
Thanks, I will give it a try,

NicolasGagne, 2023-11-05:
I am doing something wrong but I can't figure out what.

I edited the satdump_cfg.json file from the main satdump directory and it dosen't seem to have any effect. 
I have also noted some discrepancy between the .json file and the parameter in the setting section of satdump-ui. (exemple: Lat, Long, folders, etc) 

Is there an other satdump_cfg.json file somewhere ? or a way to know where are the parameter from the setting tab come from?
[satdump_cfg.json](https://github.com/SatDump/SatDump/files/13261004/satdump_cfg.json)

JVital2013, 2023-11-06:
Some settings are overridden by settings.json.

It's possible SatDump is loading the wrong config file. SatDump will try to load a satdump_cfg.json in your current working directory (or the folder satdump.exe is in); if it's not there, it will look for it at `/usr/share/satdump/satdump_cfg.json` (on Linux), or `C:\Program Files\Satdump\share\satdump_cfg.json` (Windows installed version).

Assuming you have SatDump installed, it's simplest to make sure there's no satdump_cfg.json in your current working directory, and you're editing the system installed one.

JVital2013, 2024-01-15:
Hi @NicolasGagne,

Do you need any further assistance with this?

NicolasGagne, 2024-01-20:
It all good

closed 2024-01-20 by NicolasGagne