SatDump/SatDump #458: SatDump Crashing

ve3elb, 2023-10-30:
**Description of the issue**

I have tried on many devices with the same results SatDump crashes between 10mins - 30mins. 

Desktop PC Windows11
Laptop PC Windows11
Raspberry Pi 4GB PiOS Bullseye
Raspberry Pi 400 PiOS Bullseye. 

All crashing. 
**Hardware (SDR/PC/OS)**

**Version (Eg, 1.0.0, CI Build 171)**

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

JVital2013, 2023-10-30:
What are you doing when SatDump crashes? Do you have any logs from the crash? Logs are now dated and can be found in ~/.config/satdump (macOS, Linux) or %appdata%\satdump (windows)

ve3elb, 2023-10-30:
I did not have logs enabled. I will enable it and run SatDump again than post the loss here.

JVital2013, 2023-10-30:
OK - once we have the logs we can go from there!

ve3elb, 2023-10-31:
Got the log files but will have to upload the. Tonight as I can't upload them from my phone. 

Running SatDump/GOES-R HRIT decoder SatDump crash's and shutdown within minutes. 

However after testing I found out running SatDump/GOES-R HRIT to goestools with Goesrecv SatDump did not crash or shutdown. within the few hours it and received many images. I left it running for the day to see what will happen. Lastnight images from SatDump and goestools 

[image]

[image]

ve3elb, 2023-10-31:
There is one thing I can’t figure out. Pointed at goes16 Running SatDump or SatDump/goestools I always seem to get an image folder for goes 18 with FD images in it. Why is that. Is there a bug with both software. Have you ever seen that on your system.

JVital2013, 2023-10-31:
GOES-16 relays full disk images from GOES-18 - not a bug, but a feature 😄.

Once we get the log of the standalone SatDump crash we can really get started, but I already have an idea of where to look...

JVital2013, 2023-10-31:
@ve3elb I found and fixed a crash in the GOES-R HRIT decoder. If you can, please update and let us know if it still crashes for you. If it does, send the log over.

Thanks!

ve3elb, 2023-10-31:
Jamie Thank You. I will update from in a bit and keep you updated.

ve3elb, 2023-11-01:
[satdump-20231031T180754.log](https://github.com/SatDump/SatDump/files/13222346/satdump-20231031T180754.log)
[satdump-20231031T181410.log](https://github.com/SatDump/SatDump/files/13222347/satdump-20231031T181410.log)

JVital2013, 2023-11-01:
Thanks so much! I assume you're using the UI, not the CLI?

If you can, try updating again - I just fixed another issue that caused a crash if there were too many CRC failures at a time. I'm seeing a lot of those in the log, and I suspect that's what's causing the crash now.

ve3elb, 2023-11-01:
Thank you I will update and give it another go.

ve3elb, 2023-11-01:
May I ask why does SatDump make a new folder everytime it starts. This screw up VitalityGoes and you have to change the path everytime.  I like Goestools it uses the same path all the time.   Is there something I can do to have SatDump to return to the last used file folder/path.

JVital2013, 2023-11-01:
To turn off new folder generation in the recorder, turn the setting shown below off in Settings and close/re-open SatDump:

[image]

Afterwards, when using the Recorder screen, it will prompt you where to save the files. It won't remember this, but you can point it to the same location each time.

[image]

ve3elb, 2023-11-01:
@JVital2013  SatDump has been running for about 8Hrs with no crash/shutdown. The fix you did worked. 

I left it running through the day and will check it when I get home and report back. 

Thank You.

ve3elb, 2023-11-01:
@JVital2013 Here are two more logs. came home to a frozen satdump on the screen. 

[satdump-20231031T190955.log](https://github.com/SatDump/SatDump/files/13231134/satdump-20231031T190955.log)
[satdump-20231031T205853.log](https://github.com/SatDump/SatDump/files/13231135/satdump-20231031T205853.log)

JVital2013, 2023-11-01:
Thanks again! Sounds like we're chipping away at the problem. 12 hours is certainly an improvement over 30 minutes 😄.

I don't have any solid answers yet, but I do have some thoughts:
- You can try updating again - I again made some changes to the HRIT decoder this morning to correct some minor issues when there are lots of data errors. There's a chance it's related to this lock-up.
- Is there anything you can do to increase the signal strength? Your SNR is averaging around 1, which is very low and is resulting in a large number of packet drops. I find it likely this is contributing to the problem - especially since it looks like your signal all but disappeared for a whole minute before the lock-up. In an ideal world, a program would never crash due to bad data, but it's hard to find all the edge cases!
- I see you're using a RSP1A - after exiting the locked-up SatDump, were you able to use the SDR without restarting your computer? The reason I ask this oddly-specific question is: we've had issues with the SDRPlay API in the past, and I wonder if it's possible the SDRPlay API locked up and caused an issue.

ve3elb, 2023-11-01:
@JVital2013  yes I am using an RSPA1. I am at the max gain setting. Sorry when I seen the lockup I just rebooted the PC. if it happens again I will check and see if the API is active. As I mentioned it's a completely new setup and there's nothing on this pc. I installed the latest API v3 along with SatDump and goestools.

ve3elb, 2023-11-03:
@JVital2013  Heads up SatDump v1.1.2-c4752d2 has been up and running for 24Hrs with no issues decoding GOES-16. Thank you for all you work, patience and help. 

Now only if we can get the satellite tracking and scheduler to start the radio and image decoders for each satellite pass then stop the radio and process the images after the pass that would be great all fully automated. Or maybe am doing something wrong as I can auto track the satellites but have to start/stop the APT decoders manually for each pass.. 

Once again BIG thank you.

[image]

ve3elb, 2023-11-03:
@JVital2013  

I noticed one processing bug SatDump does not produce or prosess these images.

[image]

[image]

[image]

JVital2013, 2023-11-05:
I'm glad to hear SatDump seems to be running stably for you now! Since your original issue seems to be fixed, I'm going to close this issue for now.

**About those colored images that SatDump doesn't save:** those "enhanced" images are composites created by goestools using a gradient, and are not actually present on the GOES-R HRIT downlink. SatDump will have similar support in the future.

**Auto-decoding is a feature already in SatDump:** to use it, go to the Tracking section of the recorder, and click "Schedule and config":

[image]

In here, search for the satellite you want to track, and double-click on it to move it over to the right. Repeat for all satellites you want to auto-decode:

[image]

At first, the information at the bottom will be wrong. Set the correct frequency and decoder, and check "Live":

[image]

Next, click "Configure" and set up the decoder options.

[image]

Then, click "Update Passes" to refresh the list of upcoming passes. 

[image]

Finally, check "Engage autotrack" and it will handle the passes for you. The radio will be left running after the pass ends.

closed 2023-11-05 by JVital2013