SatDump/SatDump #460: Android, UI: crash upon textfield or volume button usage (Oneplus Nord)

Strikeless, 2023-11-02:
**Description of the issue**
SatDump on Android 12 running on my Oneplus Nord always crashes whenever I use the phone's volume buttons, or press the backspace or enter/submit key in a textfield.

**Hardware (SDR/PC/OS)**
Oneplus Nord phone running OxygenOS 12.1 based on Android 12. Stock.
Occurs no matter any connected SDRs or other external devices.

**Version (Eg, 1.0.0, CI Build 171)**
1.1.1

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
I couldn't find where to get SatDump's log file from, but I found this with adb logcat:
```
11-02 22:20:10.273 10149 10149 F DEBUG   : Process name is org.satdump.SatDump, not key_process
11-02 22:20:10.273 10149 10149 F DEBUG   : keyProcess: 0
11-02 22:20:10.273 10149 10149 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
11-02 22:20:10.273 10149 10149 F DEBUG   : Build fingerprint: 'OnePlus/Nord_EEA/Nord:12/RKQ1.211119.001/Q.GDPR.202309140004:user/release-keys'
11-02 22:20:10.273 10149 10149 F DEBUG   : Revision: '0'
11-02 22:20:10.273 10149 10149 F DEBUG   : ABI: 'arm64'
11-02 22:20:10.273 10149 10149 F DEBUG   : Timestamp: 2023-11-02 22:20:09.810797749+0200
11-02 22:20:10.273 10149 10149 F DEBUG   : Process uptime: 0s
11-02 22:20:10.273 10149 10149 F DEBUG   : Cmdline: org.satdump.SatDump
11-02 22:20:10.273 10149 10149 F DEBUG   : pid: 9669, tid: 9720, name: Thread-117  >>> org.satdump.SatDump <<<
11-02 22:20:10.273 10149 10149 F DEBUG   : uid: 10424
11-02 22:20:10.273 10149 10149 F DEBUG   : signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
11-02 22:20:10.273 10149 10149 F DEBUG   : Abort message: '/home/alan/Documents/SatDump_ReWork/src-core/imgui/imgui.cpp:1460: void ImGuiIO::AddKeyAnalogEvent(ImGuiKey, bool, float): assertion "KeyMap[n] == -1 && "Backend needs to either only use io.AddKeyEvent(), either only fill legacy io.KeysDown[] + io.KeyMap[]. Not both!"" failed'
11-02 22:20:10.273 10149 10149 F DEBUG   :     x0  0000000000000000  x1  00000000000025f8  x2  0000000000000006  x3  0000006fe7d09a10
11-02 22:20:10.273 10149 10149 F DEBUG   :     x4  0000000080808080  x5  0000000080808080  x6  0000000080808080  x7  8080808000000000
11-02 22:20:10.273 10149 10149 F DEBUG   :     x8  00000000000000f0  x9  7fbcb83c1ac486d8  x10 0000000000000000  x11 ffffff80fffffbdf
11-02 22:20:10.273 10149 10149 F DEBUG   :     x12 0000000000000001  x13 0000000100000000  x14 0000000000000000  x15 0000000000000068
11-02 22:20:10.273 10149 10149 F DEBUG   :     x16 00000070fffb1bc0  x17 00000070fff8c680  x18 0000006fd9f12000  x19 00000000000025c5
11-02 22:20:10.273 10149 10149 F DEBUG   :     x20 00000000000025f8  x21 00000000ffffffff  x22 0000000000000000  x23 0000000000000000
11-02 22:20:10.273 10149 10149 F DEBUG   :     x24 0000006fe7d09cb0  x25 0000006fe7d09cb0  x26 0000006fe7d09ff8  x27 00000000000fc000
11-02 22:20:10.273 10149 10149 F DEBUG   :     x28 0000006fe7c11000  x29 0000006fe7d09a90
11-02 22:20:10.273 10149 10149 F DEBUG   :     lr  00000070fff3c99c  sp  0000006fe7d099f0  pc  00000070fff3c9c8  pst 0000000000000000
11-02 22:20:10.273 10149 10149 F DEBUG   : backtrace:
11-02 22:20:10.273 10149 10149 F DEBUG   :       #00 pc 00000000000799c8  /apex/com.android.runtime/lib64/bionic/libc.so (abort+168) (BuildId: bbbdeb7c87c74f1491f92c6e605095b0)
11-02 22:20:10.273 10149 10149 F DEBUG   :       #01 pc 0000000000079d94  /apex/com.android.runtime/lib64/bionic/libc.so (__assert2+40) (BuildId: bbbdeb7c87c74f1491f92c6e605095b0)
11-02 22:20:10.273 10149 10149 F DEBUG   :       #02 pc 000000000059c484  /data/app/~~YaSWYKPxqeLTP23WMR8p4g==/org.satdump.SatDump-OvyJt3wFFtfzFIXCF8vFqQ==/lib/arm64/libsatdump_core.so (ImGuiIO::AddKeyAnalogEvent(ImGuiKey, bool, float)+808) (BuildId: 920bf45fc4bf767e582eec17dd90a81f36e7cd95)
11-02 22:20:10.273 10149 10149 F DEBUG   :       #03 pc 0000000000065a3c  /data/app/~~YaSWYKPxqeLTP23WMR8p4g==/org.satdump.SatDump-OvyJt3wFFtfzFIXCF8vFqQ==/lib/arm64/libandroid_imgui.so (ImGui_ImplAndroid_HandleInputEvent(AInputEvent*)+120) (BuildId: b12f8cdb69608102787c53004f6e5fa6853b981f)
11-02 22:20:10.273 10149 10149 F DEBUG   :       #04 pc 0000000000068628  /data/app/~~YaSWYKPxqeLTP23WMR8p4g==/org.satdump.SatDump-OvyJt3wFFtfzFIXCF8vFqQ==/lib/arm64/libandroid_imgui.so (BuildId: b12f8cdb69608102787c53004f6e5fa6853b981f)
11-02 22:20:10.273 10149 10149 F DEBUG   :       #05 pc 000000000005f674  /data/app/~~YaSWYKPxqeLTP23WMR8p4g==/org.satdump.SatDump-OvyJt3wFFtfzFIXCF8vFqQ==/lib/arm64/libandroid_imgui.so (android_main+252) (BuildId: b12f8cdb69608102787c53004f6e5fa6853b981f)
11-02 22:20:10.273 10149 10149 F DEBUG   :       #06 pc 0000000000068464  /data/app/~~YaSWYKPxqeLTP23WMR8p4g==/org.satdump.SatDump-OvyJt3wFFtfzFIXCF8vFqQ==/lib/arm64/libandroid_imgui.so (BuildId: b12f8cdb69608102787c53004f6e5fa6853b981f)
11-02 22:20:10.273 10149 10149 F DEBUG   :       #07 pc 00000000000deb44  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+264) (BuildId: bbbdeb7c87c74f1491f92c6e605095b0)
11-02 22:20:10.273 10149 10149 F DEBUG   :       #08 pc 000000000007b2fc  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+68) (BuildId: bbbdeb7c87c74f1491f92c6e605095b0)
```

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
I am not able to reproduce this bug with SatDump version 1.1.0 on the same phone.
The crash does not seem to get triggered by "typical" keyboard input, but always occurs when I press backspace or enter.
The given snippet from adb logcat seems to have a quite descriptive abort message.

JVital2013, 2023-11-03:
I can reproduce the issue - we'll keep you updated when we have a fix!

JVital2013, 2023-11-03:
This should now be fixed - could you verify it works for you by downloading the latest nightly build from https://github.com/SatDump/SatDump/actions/runs/6746203388?

Thanks for the report!

Strikeless, 2023-11-03:
Great stuff, works as expected. Thank you for the quick fix :)

closed 2023-11-03 by Strikeless