SatDump/SatDump #463: GOES-R HRIT to GOEStools

ve3elb, 2023-11-03:
Hello there seem to be an issue with data transfer.  When Running SatDump/GOES-R HRIT to Goestools with Goestools/Goesproc.
the following full disk image is not received.

[image]

JVital2013, 2023-11-05:
These false color images look slightly different when saved with goestools/goesproc. Assuming you're using Goestools for Windows, it's saved by default in the `goes16/fd/fc/` folder, and it has a black background. Goestools does not support saving this image with a white background.

If the file is not there, and other files are being saved by goestools, then the issue is with goestools and not SatDump. SatDump does not know what kind of data it is sending to goesproc, and therefore cannot filter out specific products.

ve3elb, 2023-11-05:
@JVital2013  Yes I am using goestools for windows. But these color images are not in the folder. The image I shown was from SatDump

JVital2013, 2023-11-06:
Do you have anything in `goes16/fd/fc/` at all? You should also verify a false color handler is in your goesproc config - it should be by default. Just do a Ctrl+F for "fc".

That being said, goestools will completely discard some files if they have black lines across them like many of yours have. Improving your SNR will go a long way in fixing these issues.

Either way, this is not a SatDump issue, but a goestools issue (if not a signal strength issue). If you want to continue this conversation, look for help on one of the various Geostationary Satellite/goestools groups, or open a ticket on the Goestools for Windows repo.

Closing.

JVital2013, 2023-11-06:
So, I believe I spoke too soon. Try updating satdump from the latest nightly and see if it fixes your issue.

I did some digging in the goesrecv source code, and I found some behavioral differences between SatDump and goesrecv. Since goesproc expects bad data to be handled certain way, it can cause problems if SatDump acts differently. I made a change that should provide data to a format goesproc is expecting. See if this helps with using goesproc on Windows.

JVital2013, 2024-02-04:
Closing again as this has been working in my tests - feel free to reach back out if you have continued issues

closed 2024-02-04 by JVital2013