SatDump/SatDump #464: satdum live SDR device name argument ?

f6bvp, 2023-11-05:
Working via Internet ssh to remote site running  SatDump-ui it is unusable for it is to low despite fast links on both sides.

I wanted to try satdump_sdr_server or satdump service

However arguments for  --source RTL2832U or other spelling did not work.

lsusb shows :
Realtek Semiconductor Corp. RTL2838 DVB-T

SaDump-ui device says :

Generic RTL2832U OEM #0

What argument should I put to select this RTL SDR device with live or server application ?

Suggestion : SatDump could display the device names list to use when started.

Regards,

Bernard
f6bvp /ai7bg

f6bvp, 2023-11-06:
Trying SatDump live,  not 'satdump service' as written before.
And suggestion is that SatDump could display the exact SDR device names that should be used on command line not the one in the list of device that do not work.

JVital2013, 2023-11-06:
To get the list of available SDRs in a format that works for the command line, run `satdump sdr_probe`. The "type" is the parameter to use for `--source` in your live command.

[image]

In this case, you will want `rtlsdr`. If you have more than one connected, look at the "ID" from `satdump sdr_probe`, and then add a `--source_id <id>` flag to your satdump live command.

The `satdump sdr_probe` command should be better documented - I added a line about it in the main README for now.

f6bvp, 2023-11-07:
> image

Hi Jamie,

Thanks a lot for your help providing additional documentation and very 
usefull information about devices (sources).

Bernard
f6bvp / ai7bg

f6bvp, 2023-11-07:
The following should probably not here but I don't know where to chat ?
I translated (with Google translate help ...) Satdump basic use in french for our radiotelescope blog :
https://radiotelescopelavillette.wordpress.com/application-satdump/
Actual pictures are not included and there is a link toward SatDump Basic Usage.
Are you interested in linking such translation to https://www.satdump.org/

By the way. https://www.satdump.org/ Menu on the left does not refer to https://www.satdump.org/posts/basic-usage/

It could be nice to let it know on the Menu ?

Best regards,

Bernard

f6bvp, 2023-11-07:
A more interesting proposal. You could include a Translate Widget on Satdump.org Menu ? 
I did on our site and it it is very effective.

Bernard

f6bvp, 2023-11-07:
Correct display after pluging two RTL  :
satdump sdr_probe
[16:55:14 - 07/11/2023] (I) Found devices (sources) :
[16:55:14 - 07/11/2023] (I) - File Source
[16:55:14 - 07/11/2023] (D)     - type : file
[16:55:14 - 07/11/2023] (D)     - id : 0
[16:55:14 - 07/11/2023] (I) - Generic RTL2832U OEM #0
[16:55:14 - 07/11/2023] (D)     - type : rtlsdr
[16:55:14 - 07/11/2023] (D)     - id : 0
[16:55:14 - 07/11/2023] (I) - Generic RTL2832U OEM #1
[16:55:14 - 07/11/2023] (D)     - type : rtlsdr
[16:55:14 - 07/11/2023] (D)     - id : 1
[16:55:14 - 07/11/2023] (I) - RTL-TCP
[16:55:14 - 07/11/2023] (D)     - type : rtltcp
[16:55:14 - 07/11/2023] (D)     - id : 0
[16:55:14 - 07/11/2023] (I) - SDR++ Server
[16:55:14 - 07/11/2023] (D)     - type : sdrpp_server
[16:55:14 - 07/11/2023] (D)     - id : 0
[16:55:14 - 07/11/2023] (I) - SpyServer
[16:55:14 - 07/11/2023] (D)     - type : spyserver
[16:55:14 - 07/11/2023] (D)     - id : 0
[16:55:14 - 07/11/2023] (I) - UDP Source
[16:55:14 - 07/11/2023] (D)     - type : udp_source
[16:55:14 - 07/11/2023] (D)     - id : 0
[16:55:14 - 07/11/2023] (I) Found devices (sinks) :
[16:55:14 - 07/11/2023] (I) Done! Goodbye

and satdump live  ok with

satdump live noaa_apt noaa_apt_live_test --source rtlsdr --source_id 0 --frequency 1544.5e6 --general_gain 29.6 --dc_block --samplerate 2e6

Thanks again !

Bernard

JVital2013, 2023-11-07:
I'm glad to hear you got the CLI working!

I'll leave this open for other devs to look at re: the translation suggestion.

closed 2023-11-08 by JVital2013