SatDump/SatDump #476: Device after restart is gone

georgjorge, 2023-11-08:
**What would you want to be added to satdump?**
A clear and concise description of what the idea/problem is.

**Describe the feature you'd like to have**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Here you can for provide more information about what you'd like to see as the end result. Feel free to also post images if you have ideas regarding the UI

georgjorge, 2023-11-08:
Hello,
when I restart SatDump my device is gone. Is this a regular behavior?

JVital2013, 2023-11-08:
Sorry, we will need more details here. What do you mean by "your device is gone?"

georgjorge, 2023-11-08:
Manually, I choos RTL-2832 as input device. After restarting the program, the input device was cleared.

JVital2013, 2023-11-08:
Ah, I understand now, thanks! Yes, this is currently how the software is designed to work.

georgjorge, 2023-11-08:
Thank you for the information!
George

JVital2013, 2024-01-06:
Hi @georgjorge,

Sorry for letting this sit for so long! We have implemented a change so SatDump now remembers the last used SDR. This change will be a part of 1.1.3

commit 183a21bf (2024-01-06): Remember last used SDR in recorder (fixes #476)
closed 2024-01-06 by JVital2013