SatDump/SatDump #477: error opening dataset

georgjorge, 2023-11-08:
Hi, 
I am still struggling with SatDump. I am not able to decode the received data from noaa 15 with format f32 and also noaa 18 with format .wav.
Always I get an exception-parse-error 101 at line 1 column 1, syntax error while parsing value - invalid literal, last read ´R´. Could anyone point me to set the correct parameter for decoding the file?
Further more I am not able to get an automatic start to receive satellites even though it was scheduled.

George

JVital2013, 2023-11-08:
You cannot decode basebands with the viewer screen - you should use the "offline processing" screen for that. The viewer screen is for opening dataset.json/product.cbor files from already-processed data.

I'd recommend reviewing this web page to get started: https://www.satdump.org/posts/basic-usage/

georgjorge, 2023-11-08:
Thank you for your quick answer, but I still does not get an picture with offline processing. I tried noaa apt, dsb and gac decoder with input level basement. With noaa apt it says error 302. With dsb decoder it reports "none of the channel is present".  The signal is approximately 12 dB over noise. The basic-usage document does not help me. Sorry molesting you.
George

JVital2013, 2023-11-10:
**Let's start with the NOAA 15 f32 recording you have.** I'm assuming you got it's APT signal on 137.62 MHz? Set up SatDump like this. If the signal was not centered in the baseband when you recorded it, you'll need to set a frequency shift.

[image]

**For the NOAA 18 that's a wav:** is this a baseband wav, or an audio wav (where you can hear the ticking)? If it's an audio wav, you need to set satdump up like this:

[image]

JVital2013, 2023-11-25:
Hi @georgjorge, would you like any further assistance with this issue?

Aang23, 2024-01-05:
I'll close this for now. I'll gladly re-open on any further reply.

closed 2024-01-05 by Aang23