SatDump/SatDump #483: Tracking display is broken

db2os, 2023-11-12:
Recently the AZ/EL diagram is showing strange behavior, in particular with objects from Horizons.
Latest release..

[image]

JVital2013, 2023-11-12:
This has been fixed - please update to the latest nightly and try again. If it still happens, let us know.

Thanks for the report!

db2os, 2023-11-14:
@JVital2013 it seems there is a new issue..   at LOS (Elevation < 0) the display is cleared and does not show any updates and next AOS time..
[image]

JVital2013, 2023-11-18:
Update to the latest nightly and try again - I was able to reproduce the issue and fix it on my end, but I'd like to make sure it works for you too.

db2os, 2023-11-19:
Thanks @JVital2013 - it seems to work fine now..

JVital2013, 2023-11-19:
Great, glad to hear it!

closed 2023-11-19 by JVital2013