SatDump/SatDump #484: Automatic tracking and recording (+processing) via CLI

variorex, 2023-11-12:
Hey SatDump Team, first of all, big thanks for your incredible work! The workflow of recording, processing and enhancing data from all these different satellites is a dream-come-true compared to what was necessary only a few years ago. The automated tracking function is definitely the icing on top of the cake, since I really like to automate the workflow.
So currently, I use the tracking option located within the recorder to record and process the NOAAs and Meteor, it works really well. But using that requires the use of the GUI/Desktop version of SatDump.

**What would you want to be added to satdump?**
Would it be possible to add an option to the CLI version of SatDump to track and record, maybe even process the data?
This would allow the usage of SatDump for example on a home server running as a background process, streamlining and automating the whole process even further.
If adding all the processing/decoding functionalities to SatDump is too much to ask for, even the functionality of only automated recording controlled by the tracking via CLI, without the processing/decoding, would be a huge step towards that automated workflow.

**Describe alternatives you've considered**
Keeping the desktop computer running 24/7 with the SatDump GUI version tracking option enabled...

**Additional context**
Just an idea: Since the tracking option would probably add a whole lot of necessary arguments to the command (all these settings done in the "Tracking Configuation"-window in the GUI), maybe just a simple "--engage-autotrack" option would be fine, that only reads the tracking config made with the GUI, and just enables it, having the identical functionality to enabling the checkbox "Engage Autotrack" in the "Tracking Configuration"-Menu in the GUI.

Thanks 😃

Aang23, 2023-11-15:
Yes, this is planned at some point - the idea is to have feature parity (as possible) between the UI and CLI. Automation will of course be covered.

Aang23, 2024-02-01:
This is now mostly done :-)

closed 2024-05-27 by Aang23