SatDump/SatDump #493: NOAA ATP avhrr calibration

Phoenix-64, 2023-11-15:
What would you want to be added to satdump?
Curently the NOAA ATP avhrr pictures are only color value calibrated and that is awesome. Now I wanted to ask whether it be possible to add the HRPT avhrr calibration to the NOAA ATP pipeline. The algorithm is identical. The only adjustment needed is multiplying all input values by 4 because the HRPT images are 10bit and the ATP ones 8bit so the count values are off. 
The PRT counts are in wedges 10, 11, 12, 13 and the back scan value, Cbb is in wedge 15.
Only the space data would need to be additionally extracted from the image, it is in the strip left to the data channel, see the market strip in the image
[image]

I took a look at the code but am just not good enough to figure out the structure used as far as I can see the code from noaa_meteop_supports/instruments/avhrr can be nearly identically copied.
Thank you so much for your help and everything you have done sofar this is truly the only one-stop shop for all things satellite decoding related.

Aang23, 2023-11-15:
Hello!
This is planned - and will probably get looked at quite soon. Really the only thing missing for APT to be "fully supported".

ZbychuButItWasTaken, 2023-11-23:
Hi, 
the feature is currently undergoing testing in the `apt_calib` branch.
You can give it a shot when you have some time! It will be soon merged into master

Phoenix-64, 2023-11-24:
Uhh cool will give it a compile and try it out

closed 2023-11-25 by Aang23