SatDump/SatDump #499: Mute option for decoders with audio output

variorex, 2023-11-16:
**Describe the feature you'd like to have**
Since some decoders like the NOAA APT decoder enable audio output when a signal is being processed live within the "recorder"-tab, an option to mute the satdump audio output would be really useful.
In my usecase with satdump running in the background in tracking mode, I always get a minor heart attack when the APT decoder automatically enables itself and the pretty loud decoder noise comes on😁 

**Describe alternatives you've considered**
Muting the entire PC audio output.

**Additional context**
I think a simple checkbox-based option to mute the audio output or something like a volume slider would be very nice. Also, making the last state of the mute-option/slider position "non-volatile" so that after closing and reopening satdump that state is still enabled, would make the usage even more smoother.

Thanks 😄

klhutchins1, 2023-11-17:
"I always get a minor heart attack when the APT decoder automatically enables itself and the pretty loud decoder noise comes on". haha, that brought me here for this reason as well.

RAD750, 2023-11-17:
An additional workaround for the time being is muting SatDump only from the audio control panel of your operating system.
For example, on Linux: 
[image]

On Windows the process is similar.

JVital2013, 2023-11-17:
Well, we can't have SatDump be the cause of any coronary events. Grab the latest nightly - I just added an button to the decoder to turn the audio on/off during the decode:

[image]

Toggling this setting only affects the current decode. To set the default, change this option on the settings screen:

[image]

Thanks for the feedback!

closed 2023-11-17 by JVital2013